u-blox/ubxlib #224: Unable to connect to AWS following example

davidallenmann, 2024-03-29:
I am trying to follow the example from this document and using the M-Center to send AT commands to the SARA-R510M8S dev board connected to a Windows PC.

I am using an ATT SIM card and am able to send SMS messages, and the ATT control center shows it is connected.

I've gone over the certs and keys and they seem to be correct. I am not sure what 'operation not allowed' means. Is that coming from AWS or from the modem?

Everything is accepted until I get to the command to log in:
 
```
[19:29:41.698] AT+UMQTT=2,"xxxxxx-ats.iot.us-east-1.amazonaws.com",8883
[19:29:41.730] OK
[19:29:43.879] AT+UMQTT=11,1,0
[19:29:43.895] OK
[19:29:45.480] AT+UMQTTC=1
[19:29:45.495] +CME ERROR: operation not allowed
```

At some point the Monitor on the AWS IoT Core dashboard showed that there was a connection made, but I have never had the AT+MQTTC=1 command succeed.

Any thoughts on what else to try?

RobMeades, 2024-03-30:
Hi there, and sorry you're having trouble with this.

`+CME ERROR: operation not allowed` is coming from the module, so it is most likely that it either (a) does not like something in the security profile or (b) does not think it has a connection.  The latter could happen if you are entering the AT commands manually and have not entered the `AT+UPSDx` commands that the `ubxlib` code would normally issue to associate the cellular PDP context established by the module with a "profile" inside the module that the internal MQTT client can use; depending on the u-blox module type, slightly different u-blox-specific `AT+UPSDx` commands have to be issued to let the module know that you are happy for the internal clients to use a PDP context that has been created.

Here is an example of a working sequence run against a SARA-R510M8S module:

```
23:45:56,889 AT+CGATT?
23:45:56,919 +CGATT: 1
23:45:56,949 OK
23:45:56,960 AT+CGACT?
23:45:56,996 +CGACT: 1,1
23:45:57,026 OK
23:45:57,037 AT+UPSD=0,0,0
23:45:57,076 OK
23:45:57,087 AT+UPSD=0,100,1
23:45:57,136 OK
23:45:57,147 AT+UPSDA=0,3
23:45:57,196 OK
23:45:57,227 +UUPSDA: 0,"192.168.3.10"
23:45:57,322 AT+CGACT?
23:45:57,352 +CGACT: 1,1
23:45:57,382 OK
23:46:01,991 AT+UMQTT=2,"ubxlib.com"
23:46:02,046 OK
23:46:02,147 AT+UMQTT=0,"357862090057839"
23:46:02,190 OK
23:46:02,406 AT+UMQTT=11,1,0
23:46:02,435 OK
23:46:03,032 AT+UMQTTC=1
23:46:03,060 OK
23:46:03,360 +UUMQTTC: 1,1
```

davidallenmann, 2024-03-30:
Thanks for the quick response. 
From the m-center console it looks like the modem is reading the PSD profile for att correctly. After AT+UPSD=0,100,1 I get an error when I try to activate it (AT+UPSD=0,3).

In m-center it is showing the Profile as an external PSD profile. I am not sure how the AT_UPSD=0,100,1 knows how to map an external vs internal profile.

```
[13:46:43.267] AT+CGDCONT?
[13:46:43.282] 
[13:46:43.282] AT+CGDCONT?
[13:46:43.282] 
[13:46:43.282] +CGDCONT: 1,"IP","m2m.com.attz.xxxx.xxxx0.gprs","10.236.xxx.xxx",0,0,0,2,0,0,0,0,0,0
[13:46:43.313] 
[13:46:43.313] OK
[13:46:56.951] 
[13:46:56.951] AT+UPSD=0,100,1
[13:46:56.967] 
[13:46:56.967] AT+UPSD=0,100,1
[13:46:56.967] 
[13:46:56.967] OK
[13:47:01.945] 
[13:47:01.945] AT+UPSD=0,3
[13:47:01.976] 
[13:47:01.976] AT+UPSD=0,3
[13:47:01.976] 
[13:47:01.976] ERROR
```

RobMeades, 2024-03-30:
I think you want `AT+UPSDA=0,3`, not `AT+UPSD=0,3`...?

davidallenmann, 2024-04-01:
Same 'error' with UPSDA. 
```
[09:09:05.352] OK
[09:09:15.788] 
[09:09:15.788] AT+UPSD=0,100,1
[09:09:15.804] 
[09:09:15.804] 
[09:09:15.804] OK
[09:09:18.428] 
[09:09:18.428] AT+UPSDA=0,3
[09:09:18.443] 
[09:09:18.443] 
[09:09:18.443] ERROR
```

RobMeades, 2024-04-01:
Could you post the entire AT sequence, from power-on?  If `AT+UPSDA=0,3` is returning `ERROR` then something before this point must be going wrong, e.g. the context is not activated or the context numbering is wrong or there is something not quite right about the security profile.  Note that it is worth doing `AT+UPSD=0,0,0` before doing `AT+UPSD=0,100,1` (that is what the `ubxlib` code does) to reset profile 0 to defaults first.

davidallenmann, 2024-04-01:
From M-Center
1. Refresh Info 
2. Refresh (PSD Profile)
3. From AT Terminal entered UPSD commands

```
[16:15:19.601] CONNECT|COM63|115200|Data bits:8|Parity:none|Stop bits:1|Flow ctrl:hardware
[16:15:19.706] AT
[16:15:19.721] 
[16:15:19.721] AT
[16:15:19.721] 
[16:15:19.721] OK
[16:15:19.814] 
[16:15:19.814] ATE0
[16:15:19.830] 
[16:15:19.830] ATE0
[16:15:19.830] 
[16:15:19.830] OK
[16:15:19.907] 
[16:15:19.907] AT+CMEE=2
[16:15:19.923] 
[16:15:19.923] 
[16:15:19.923] OK
[16:15:19.985] 
[16:15:19.985] AT+CGMI
[16:15:20.015] 
[16:15:20.015] 
[16:15:20.015] u-blox
[16:15:20.015] 
[16:15:20.015] OK
[16:15:20.126] 
[16:15:20.126] AT+CGMM
[16:15:20.173] 
[16:15:20.173] 
[16:15:20.173] SARA-R510M8S
[16:15:20.173] 
[16:15:20.173] OK
[16:15:20.282] 
[16:15:20.282] ATI0
[16:15:20.312] 
[16:15:20.312] 
[16:15:20.312] SARA-R510M8S-00B-03
[16:15:20.312] 
[16:15:20.312] OK
[16:15:20.422] 
[16:15:20.422] AT+CGMR
[16:15:20.453] 
[16:15:20.453] 
[16:15:20.453] 02.10
[16:15:20.453] 
[16:15:20.453] OK
[16:15:20.562] 
[16:15:20.562] ATI9
[16:15:20.593] 
[16:15:20.593] 
[16:15:20.593] 02.10,A00.01
[16:15:20.593] 
[16:15:20.593] OK
[16:15:20.702] 
[16:15:20.702] AT+CGSN
[16:15:20.733] 
[16:15:20.733] 
[16:15:20.733] 357862090326242
[16:15:20.733] 
[16:15:20.733] OK
[16:16:25.621] 
[16:16:25.621] AT+COPS?
[16:16:25.794] 
[16:16:25.794] 
[16:16:25.794] +COPS: 0,0,"AT&T",7
[16:16:25.794] 
[16:16:25.794] OK
[16:16:25.888] 
[16:16:25.888] AT+CSQ
[16:16:25.934] 
[16:16:25.934] 
[16:16:25.934] +CSQ: 12,1
[16:16:25.934] 
[16:16:25.934] OK
[16:16:26.027] 
[16:16:26.027] AT+CEREG?
[16:16:26.043] 
[16:16:26.043] 
[16:16:26.043] +CEREG: 0,1
[16:16:26.043] 
[16:16:26.043] OK
[16:16:29.629] 
[16:16:29.629] AT+UPSD=0,1
[16:16:29.675] 
[16:16:29.675] 
[16:16:29.675] +UPSD: 0,1,""
[16:16:29.675] 
[16:16:29.675] OK
[16:16:29.769] 
[16:16:29.769] AT+UPSD=0,2
[16:16:29.784] 
[16:16:29.784] 
[16:16:29.784] +CME ERROR: operation not supported
[16:16:29.846] 
[16:16:29.846] AT+UPSND=0,0
[16:16:29.862] 
[16:16:29.862] 
[16:16:29.862] +UPSND: 0,0,""
[16:16:29.862] 
[16:16:29.862] OK
[16:16:29.970] 
[16:16:29.970] AT+UPSD=1,1
[16:16:29.986] 
[16:16:29.986] 
[16:16:29.986] +UPSD: 1,1,""
[16:16:29.986] 
[16:16:29.986] OK
[16:16:30.095] 
[16:16:30.095] AT+UPSD=1,2
[16:16:30.125] 
[16:16:30.125] 
[16:16:30.125] +CME ERROR: operation not supported
[16:16:30.188] 
[16:16:30.188] AT+UPSND=1,0
[16:16:30.235] 
[16:16:30.235] 
[16:16:30.235] +UPSND: 1,0,""
[16:16:30.235] 
[16:16:30.235] OK
[16:16:30.329] 
[16:16:30.329] AT+UPSD=2,1
[16:16:30.345] 
[16:16:30.345] 
[16:16:30.345] +UPSD: 2,1,""
[16:16:30.345] 
[16:16:30.345] OK
[16:16:30.454] 
[16:16:30.454] AT+UPSD=2,2
[16:16:30.470] 
[16:16:30.470] 
[16:16:30.470] +CME ERROR: operation not supported
[16:16:30.532] 
[16:16:30.532] AT+UPSND=2,0
[16:16:30.548] 
[16:16:30.548] 
[16:16:30.548] +UPSND: 2,0,""
[16:16:30.548] 
[16:16:30.548] OK
[16:16:30.642] 
[16:16:30.642] AT+UPSD=3,1
[16:16:30.673] 
[16:16:30.673] 
[16:16:30.673] +UPSD: 3,1,""
[16:16:30.673] 
[16:16:30.673] OK
[16:16:30.765] 
[16:16:30.765] AT+UPSD=3,2
[16:16:30.812] 
[16:16:30.812] 
[16:16:30.812] +CME ERROR: operation not supported
[16:16:30.875] 
[16:16:30.875] AT+UPSND=3,0
[16:16:30.891] 
[16:16:30.891] 
[16:16:30.891] +UPSND: 3,0,""
[16:16:30.906] 
[16:16:30.906] OK
[16:16:30.985] 
[16:16:30.985] AT+UPSD=4,1
[16:16:31.000] 
[16:16:31.000] 
[16:16:31.000] +UPSD: 4,1,""
[16:16:31.000] 
[16:16:31.000] OK
[16:16:31.110] 
[16:16:31.110] AT+UPSD=4,2
[16:16:31.126] 
[16:16:31.126] 
[16:16:31.126] +CME ERROR: operation not supported
[16:16:31.189] 
[16:16:31.189] AT+UPSND=4,0
[16:16:31.204] 
[16:16:31.204] 
[16:16:31.204] +UPSND: 4,0,""
[16:16:31.204] 
[16:16:31.204] OK
[16:16:31.314] 
[16:16:31.314] AT+UPSD=5,1
[16:16:31.329] 
[16:16:31.329] 
[16:16:31.329] +UPSD: 5,1,""
[16:16:31.329] 
[16:16:31.329] OK
[16:16:31.423] 
[16:16:31.423] AT+UPSD=5,2
[16:16:31.453] 
[16:16:31.453] 
[16:16:31.453] +CME ERROR: operation not supported
[16:16:31.515] 
[16:16:31.515] AT+UPSND=5,0
[16:16:31.531] 
[16:16:31.531] 
[16:16:31.531] +UPSND: 5,0,""
[16:16:31.547] 
[16:16:31.547] OK
[16:16:31.625] 
[16:16:31.625] AT+UPSD=6,1
[16:16:31.640] 
[16:16:31.640] 
[16:16:31.640] +UPSD: 6,1,""
[16:16:31.640] 
[16:16:31.640] OK
[16:16:31.751] 
[16:16:31.751] AT+UPSD=6,2
[16:16:31.767] 
[16:16:31.767] 
[16:16:31.767] +CME ERROR: operation not supported
[16:16:31.829] 
[16:16:31.829] AT+UPSND=6,0
[16:16:31.845] 
[16:16:31.845] 
[16:16:31.845] +UPSND: 6,0,""
[16:16:31.845] 
[16:16:31.845] OK
[16:16:31.955] 
[16:16:31.955] AT+CGDCONT=?
[16:16:31.970] 
[16:16:31.970] 
[16:16:31.970] +CGDCONT: (0-11),"IP",,,(0-2),(0-4),(0,1),(0,3),(0,1),(0,1),(0,1),(0,1),(0,1),(0,1)
[16:16:31.970] 
[16:16:31.970] +CGDCONT: (0-11),"IPV6",,,(0-2),(0-4),(0,1),(0,3),(0,1),(0,1),(0,1),(0,1),(0,1),(0,1)
[16:16:31.985] 
[16:16:31.985] +CGDCONT: (0-11),"IPV4V6",,,(0-2),(0-4),(0,1),(0,3),(0,1),(0,1),(0,1),(0,1),(0,1),(0,1)
[16:16:31.985] 
[16:16:31.985] OK
[16:16:32.142] 
[16:16:32.142] AT+CGDCONT?
[16:16:32.173] 
[16:16:32.173] 
[16:16:32.173] +CGDCONT: 1,"IP","m2m.com.attz.mnc170.xxxxxx.gprs","10.242.xxx.xx",0,0,0,2,0,0,0,0,0,0
[16:16:32.173] 
[16:16:32.173] OK
[16:16:54.034] 
[16:16:54.034] AT+UPSD=0,0,0
[16:16:54.049] 
[16:16:54.049] 
[16:16:54.049] OK
[16:16:58.527] 
[16:16:58.527] AT+UPSD=0,100,1
[16:16:58.574] 
[16:16:58.574] 
[16:16:58.574] OK
[16:18:28.205] 
[16:18:28.205] AT+UPSDA=0,3
[16:18:28.221] 
[16:18:28.221] 
[16:18:28.221] ERROR

```

davidallenmann, 2024-04-01:
[image]

RobMeades, 2024-04-01:
As far as I can tell you have context 0 activated, which is the default context that kind of "comes with" an LTE registration.  The `ubxlib` code never uses that, it uses context 1, so that you can set the APN etc., and as far as I can tell you have not activated context 1.  This is up to you. of course; since you are entering these commands manually you can do as you wish, so I guess you could probably do:

```
AT+UPSD=0,0,0
OK
AT+UPSD=0,100,0
OK
AT+UPSDA=0,3
OK
```

...and then wait for the `+UUPSDA` to come back.

You say at the start of your original issue "I am trying to follow the example from this document..."; which document are you referring to?  This site is about the `ubxlib` code, rather than about the documentation of AT sequences, so I'm not sure what you are trying to follow.

RobMeades, 2024-05-10:
Do you need any more help with this or can we close the issue?

RobMeades, 2024-05-16:
I'm going to assume you have what you need now and close this issue: if there is more to discuss please feel free to re-open it.

closed 2024-05-16 by RobMeades

SeanKudrna, 2024-08-20:
> Hi there, and sorry you're having trouble with this.
> 
> `+CME ERROR: operation not allowed` is coming from the module, so it is most likely that it either (a) does not like something in the security profile or (b) does not think it has a connection.
> 
> The latter could happen if you are entering the AT commands manually and have not entered the `AT+UPSDx` commands that the `ubxlib` code would normally issue to associate the cellular PDP context established by the module with a "profile" inside the module that the internal MQTT client can use; depending on the u-blox module type, slightly different u-blox-specific `AT+UPSDx` commands have to be issued to let the module know that you are happy for the internal clients to use a PDP context that has been created.
> 
> Here is an example of a working sequence run against a SARA-R510M8S module (taken from a `ubxlib` regression test run):
> 
> ```
> 23:45:56,889 AT+CGATT?
> 23:45:56,919 +CGATT: 1
> 23:45:56,949 OK
> 23:45:56,960 AT+CGACT?
> 23:45:56,996 +CGACT: 1,1
> 23:45:57,026 OK
> 23:45:57,037 AT+UPSD=0,0,0
> 23:45:57,076 OK
> 23:45:57,087 AT+UPSD=0,100,1
> 23:45:57,136 OK
> 23:45:57,147 AT+UPSDA=0,3
> 23:45:57,196 OK
> 23:45:57,227 +UUPSDA: 0,"192.168.3.10"
> 23:46:01,991 AT+UMQTT=2,"ubxlib.com"
> 23:46:02,046 OK
> 23:46:02,406 AT+UMQTT=11,1,0
> 23:46:02,435 OK
> 23:46:03,032 AT+UMQTTC=1
> 23:46:03,060 OK
> 23:46:03,360 +UUMQTTC: 1,1
> ```

I had the same issue as the OP adding this pdp profile setup resolved my issue.