u-blox/ubxlib #226: SAM-M10Q Current Draw Issue 3.3V, ~1mA instead of ~10mA

nautalert, 2024-04-09:
Firmware Version: ROM SPG 5.10 (7b202e)

EXT_INT is set to HIGH.
RESET is set to LOW.

I2C setup with the following settings:
uGnssPwrSetFlag(gnssHandle, 1UL << U_GNSS_PWR_FLAG_EXTINT_WAKE_ENABLE);
uGnssPwrSetFlag(gnssHandle, 1UL << U_GNSS_PWR_FLAG_EPHEMERIS_WAKE_ENABLE);
uGnssPwrSetFlag(gnssHandle, 1UL << U_GNSS_PWR_FLAG_EXTINT_BACKUP_ENABLE);

Open issue with UBXLIB with error setting Power Saving Modes so it is defaulting to none.

Just calling uLocationGet(gnssHandle, U_LOCATION_TYPE_GNSS,  NULL, NULL, &location, NULL);
until a fix is obtained. During that time, the SAM-M10Q is only drawing approximately 1mA. Long time to first fix. Once a fix is obtained, future fixes are faster but not as fast as they should be.

We would expect to see ~10mA at 3.3V but are only seeing ~1mA at 3.3V. It will eventually get a fix but takes a long time.

Any idea why it's not drawing full power during acquisition?

RobMeades, 2024-04-09:
Hi: I'm afraid I'll have to plead "I'm just the MCU-side SW guy" here.  I would offer to consult the right person internally but you know from #125 how quick that is, and in this case HW is concerned, which won't make things any quicker.

That said, I will add this to the internal query related to #125 to see if it triggers any thoughts.

The usual way of dealing with issue of this nature is to go back through your supplier: where do you obtain your GNSS chips?

nautalert, 2024-04-09:
Will reach out to our distributor as well from the hardware side. Does the software setup listed above seem correct? All GNSS are enabled by default, correct? Meaning it's not just GPS and we have to add GLONASS and others via the library.

RobMeades, 2024-04-09:
From the [integration manual](https://content.u-blox.com/sites/default/files/documents/SAM-M10Q_IntegrationManual_UBX-22020019.pdf) (section 2.1.2):

> SAM-M10Q supports concurrent reception of four major GNSS constellations using the GPS L1C/A, Galileo E1, BeiDou B1C, and GLONASS L1OF signals. The default configuration is concurrent reception of GPS, Galileo, GLONASS, and BeiDou B1C with QZSS and SBAS enabled.

So that looks like at least one flavour of each system.

RobMeades, 2024-05-10:
@nautalert: we've resolved the [other issue](https://github.com/u-blox/ubxlib/issues/125), is there more we need to do from a `ubxlib` perspective on this issue, or can we close this?

nautalert, 2024-05-10:
Go ahead and close it. We are working through our distributor on it. Thank you.

closed 2024-05-10 by RobMeades