u-blox/ubxlib #227: Zephyr device tree BLE NUS in nRF52832

maleroy, 2024-04-23:
Hello,

I am trying to have a BLE NUS connection between an F9P and my phone (central) via an nRF52832 (peripheral). I am trying to get a minimal code based on [the corresponding XPLR example](https://github.com/u-blox/ubxlib_examples_xplr_iot/blob/master/examples/ble_nus/src/main.c) to run but believe I am running into some device tree issues (which I see from previous reported issues to be a headache source 😅).

My understanding is that seeing that the code I'm implementing is flashed to the nRF52832, my overlay should contain the `open-cpu` tag. What's more, in order to be in accordance with the Zephyr philosophy, I should rely on
```
errorCode = uDeviceOpen(NULL, &gDeviceHandle);
```
as opposed to
```
uDeviceGetDefaults(U_DEVICE_TYPE_SHORT_RANGE_OPEN_CPU, &gDeviceCfg);
errorCode = uDeviceOpen(&gDeviceCfg, &gDeviceHandle);
```
though whatever I do I can't seem to get past an `U_ERROR_COMMON_NOT_INITIALISED` `errorCode`.

My overlay file is as follows:
```
 / {
    cfg-device-short-range {
        compatible = "u-blox,ubxlib-device-short-range";
        status = "okay";
        module-type = "U_SHORT_RANGE_MODULE_TYPE_INTERNAL";
        open-cpu;
        network = <&label_cfg_network_ble_peripheral>;
       };
    
    label_cfg_network_ble_peripheral: cfg-network-ble-peripheral {
        compatible = "u-blox,ubxlib-network-ble";
        status = "okay";
        role = "U_BLE_CFG_ROLE_PERIPHERAL";
    };
};
```

My `main.c` can be found [here](https://github.com/u-blox/ubxlib/files/15075576/main.txt) and my `prj.conf` [here](https://github.com/u-blox/ubxlib/files/15075551/prj.txt).

Via the debugger (I'm using an [nRF52 DK](https://www.nordicsemi.com/Products/Development-hardware/nRF52-DK)), I've narrowed down the bringing up of the network failure to be from [the device instance pointer saying the context pointer](https://github.com/u-blox/ubxlib/blob/v1.4.0/common/short_range/src/u_short_range_private.c#L170) is `0x0 <nvs_flash_al_wrt>`. Am I missing something obvious here? Apologies if it is trivial, it is my first experience with nRF/Zephyr.

Many thanks in advance for your help!

P.S.: development environment is VS Code + nRF Connect for VS Code v2024.3.25 + nRF toolchain v2.5.0 and ubxlib v1.4.0

RobMeades, 2024-04-23:
Thanks for posting and sorry you're having trouble with this.

Do you have the debug log output from `ubxlib`, as far as it gets, when this is run?  I ask because, when it processes the device configuration it picks up from the device tree, it prints out a line beginning with `U_PORT_BOARD_CFG` which tells you what is has done.

Thing is, though, the error code `U_ERROR_COMMON_NOT_INITIALISED` sounds like it isn't even getting that far.  If this is the error code returned by `uDeviceOpen()`, it sems more likely that you are missing a conditional compilation flag somehow.

Could you post the output from a clean build so that we can see which files are being pulled in?  Specifically, it would good to see the line `"UBXLIB_FEATURES will be...`.

maleroy, 2024-04-23:
Hi Rob,

Thanks for your comments and help. I should have been more explicit in my initial message, the `U_ERROR_COMMON_NOT_INITIALISED` is raised by the `uNetworkInterfaceUp` call, not the `uDeviceOpen` one; apologies for that!

Here is the output of a pristine build, which shows `gnss` and `short_range` features (just like I intended via a local environment variable):
```
 *  Executing task: nRF Connect: Build [pristine]: ubxlib_ble/build (active) 

Building ubxlib_ble
west build --build-dir /home/mleroy/Documents/nrf_projects/ubxlib_ble/build /home/mleroy/Documents/nrf_projects/ubxlib_ble --pristine --board nrf52dk_nrf52832 --no-sysbuild -- -DNCS_TOOLCHAIN_VERSION=NONE -DCACHED_CONF_FILE=/home/mleroy/Documents/nrf_projects/ubxlib_ble/ubxlib/port/platform/zephyr/default.conf;/home/mleroy/Documents/nrf_projects/ubxlib_ble/prj.conf -DDTC_OVERLAY_FILE=/home/mleroy/Documents/nrf_projects/ubxlib_ble/nrf52dk_nrf52832.overlay

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/mleroy/Documents/nrf_projects/ubxlib_ble
-- CMake version: 3.20.5
-- Found Python3: /home/mleroy/ncs/toolchains/7795df4459/usr/local/bin/python3.8 (found suitable version "3.8.2", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/mleroy/.cache/zephyr
-- Zephyr version: 3.4.99 (/home/mleroy/ncs/v2.5.0/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: nrf52dk_nrf52832
-- Found host-tools: zephyr 0.16.1 (/home/mleroy/ncs/toolchains/7795df4459/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (/home/mleroy/ncs/toolchains/7795df4459/opt/zephyr-sdk)
-- Found Dtc: /home/mleroy/ncs/toolchains/7795df4459/usr/bin/dtc (found suitable version "1.4.7", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/mleroy/ncs/v2.5.0/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: /home/mleroy/Documents/nrf_projects/ubxlib_ble/nrf52dk_nrf52832.overlay
-- Generated zephyr.dts: /home/mleroy/Documents/nrf_projects/ubxlib_ble/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/mleroy/Documents/nrf_projects/ubxlib_ble/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/mleroy/Documents/nrf_projects/ubxlib_ble/build/zephyr/dts.cmake
Parsing /home/mleroy/ncs/v2.5.0/zephyr/Kconfig
Loaded configuration '/home/mleroy/ncs/v2.5.0/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration '/home/mleroy/Documents/nrf_projects/ubxlib_ble/ubxlib/port/platform/zephyr/default.conf'
Merged configuration '/home/mleroy/Documents/nrf_projects/ubxlib_ble/prj.conf'
Configuration saved to '/home/mleroy/Documents/nrf_projects/ubxlib_ble/build/zephyr/.config'
Kconfig header saved to '/home/mleroy/Documents/nrf_projects/ubxlib_ble/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/mleroy/ncs/toolchains/7795df4459/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mleroy/ncs/toolchains/7795df4459/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
UBXLIB_FEATURES will be "gnss;short_range"
CMake Warning at /home/mleroy/ncs/v2.5.0/zephyr/CMakeLists.txt:1958 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /home/mleroy/Documents/nrf_projects/ubxlib_ble/build
-- west build: building application
[1/360] Preparing syscall dependency handling

[3/360] Generating include/generated/version.h
-- Zephyr version: 3.4.99 (/home/mleroy/ncs/v2.5.0/zephyr), build: v3.4.99-ncs1
[143/360] Building C object CMakeFiles/app.dir/ubxlib/port/platform/zephyr/src/u_port_board_cfg.c.obj
../ubxlib/port/platform/zephyr/src/u_port_board_cfg.c:837:26: warning: 'gpDeviceCfgWifiNetworkListApIpAddress' defined but not used [-Wunused-const-variable=]
  837 | static const char *const gpDeviceCfgWifiNetworkListApIpAddress[][2] = {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../ubxlib/port/platform/zephyr/src/u_port_board_cfg.c:826:26: warning: 'gpDeviceCfgWifiNetworkListApPassPhrase' defined but not used [-Wunused-const-variable=]
  826 | static const char *const gpDeviceCfgWifiNetworkListApPassPhrase[][2] = {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../ubxlib/port/platform/zephyr/src/u_port_board_cfg.c:815:22: warning: 'gDeviceCfgWifiNetworkListApAuthentication' defined but not used [-Wunused-const-variable=]
  815 | static const int32_t gDeviceCfgWifiNetworkListApAuthentication[][2] = {
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../ubxlib/port/platform/zephyr/src/u_port_board_cfg.c:804:26: warning: 'gpDeviceCfgWifiNetworkListApSsid' defined but not used [-Wunused-const-variable=]
  804 | static const char *const gpDeviceCfgWifiNetworkListApSsid[][2] = {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../ubxlib/port/platform/zephyr/src/u_port_board_cfg.c:793:26: warning: 'gpDeviceCfgWifiNetworkListMode' defined but not used [-Wunused-const-variable=]
  793 | static const char *const gpDeviceCfgWifiNetworkListMode[][2] = {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
In terms of console output, here it is when I comment the `uPortLogOff` call in the `main.c` I attached in my previous message:
```
*** Booting nRF Connect SDK v2.5.0 ***

Initiating the module...
U_PORT_BOARD_CFG: using SHORT-RANGE device "cfg-device-short-range" from the device tree, module-type 0, open-cpu.
Starting BLE...
U_PORT_BOARD_CFG: using BLE network configuration associated with SHORT-RANGE device "cfg-device-short-range" from the device tree, role 2.
* Failed to bring up the network: -2
```

Thanks again for your help!

RobMeades, 2024-04-23:
Thanks for the swift response.  Sorry about the warnings, we try to avoid conditional compilation flags inside `ubxlib` but it is very much the Zephyr way and so in this particular case we end up with a combinatorial nightmare.  Anyway...

As you say, you have `UBXLIB_FEATURES` `gnss` and `short_range`, which seems right.  The print from your `main.c` "`Starting BLE...`" is only emitted if `uDeviceOpen()` return 0, so that's good.

However: the second parameter to [uNetworkInterfaceUp()](https://github.com/u-blox/ubxlib/blob/79e808eb8be3095b4f94efcffba97515ba1870bd/common/network/api/u_network.h#L172) should be the network type, which in this case would be [U_NETWORK_TYPE_BLE](https://github.com/u-blox/ubxlib/blob/79e808eb8be3095b4f94efcffba97515ba1870bd/common/network/api/u_network_type.h#L64); you are calling it with `U_BLE_CFG_ROLE_PERIPHERAL` instead, which will turn out to be a value of 2, which corresponds to network type `U_NETWORK_TYPE_CELL`; that will indeed come back as not initialised since you don't have cellular in your build...?

RobMeades, 2024-04-24:
Apologies, I was getting `role` and `type` mixed up, let me look again...

maleroy, 2024-04-24:
Sure, thank you for your prompt reply!

Side note, as I wrote my message above it made me think that I could try adding a dummy Wifi `phandle`, i.e. making the overlay file as follows:
```
 / {
    cfg-device-short-range {
        compatible = "u-blox,ubxlib-device-short-range";
        status = "okay";
        module-type = "U_SHORT_RANGE_MODULE_TYPE_INTERNAL";
        open-cpu;
        network = <&label_cfg_network_ble_peripheral &label_cfg_network_wifi_client_home>;
       };
    
    label_cfg_network_wifi_client_home: cfg_network_wifi_client_home {
        compatible = "u-blox,ubxlib-network-wifi";
        mode = "U_WIFI_MODE_NONE";  // valid according to https://github.com/u-blox/ubxlib/blob/v1.4.0/port/platform/zephyr/dts/bindings/u-blox%2Cubxlib-network-wifi.yaml#L36
    };

    label_cfg_network_ble_peripheral: cfg-network-ble-peripheral {
        compatible = "u-blox,ubxlib-network-ble";
        status = "okay";
        role = "U_BLE_CFG_ROLE_PERIPHERAL";
    };
};
```
and albeit the build warnings regarding the unused variables are gone, the failure remains.

plerup, 2024-04-24:
Hi,
I'm sorry but Ubxlib currently doesn't support any BLE Gatt operations for open-cpu (internal) mode. The only thing available for now is the U-blox service SPS.
The example for the XPLR-IOT1 uses the BLE functionality of the connected Nina W15 and not the host Nora B1.

So I'm afraid the only option for you at this point is to implement this using the Zephyr apis. An example can be found here: https://github.com/junqingzou/Zephyr_NUS_Sample

BR
/Peter Lerup

maleroy, 2024-04-24:
Oh that's a shame, I thought and hoped it could work... So be it, I'll try with an alternative. Thank you both for your help!

closed 2024-04-24 by maleroy