u-blox/ubxlib #229: ESP-IDF can not initialise port

KjartanOli, 2024-04-26:
I've followed the instructions in `port/platform/esp-idf/mcu/esp32/README` and my test application compiles successfully.  However it immidiately crashes on boot with the following being logged to the serial port:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15532
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (29) boot: ESP-IDF v5.2 2nd stage bootloader
I (29) boot: compile time Apr 26 2024 16:35:27
I (30) boot: Multicore bootloader
I (34) boot: chip revision: v1.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=082d0h ( 33488) map
I (114) esp_image: segment 1: paddr=000182f8 vaddr=3ffb0000 size=0221ch (  8732) load
I (118) esp_image: segment 2: paddr=0001a51c vaddr=40080000 size=05afch ( 23292) load
I (131) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=13a44h ( 80452) map
I (159) esp_image: segment 4: paddr=00033a6c vaddr=40085afc size=0723ch ( 29244) load
I (177) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (189) cpu_start: Multicore app
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 160000000 Hz
I (197) cpu_start: Application information:
I (200) cpu_start: Project name:     ubxlib-test
I (206) cpu_start: App version:      1
I (210) cpu_start: Compile time:     Apr 26 2024 16:35:12
I (216) cpu_start: ELF file SHA256:  ee235219b...
I (222) cpu_start: ESP-IDF:          v5.2
I (226) cpu_start: Min chip rev:     v0.0
I (231) cpu_start: Max chip rev:     v3.99 
I (236) cpu_start: Chip rev:         v1.0
I (241) heap_init: Initializing. RAM available for dynamic allocation:
I (248) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (254) heap_init: At 3FFB2CD8 len 0002D328 (180 KiB): DRAM
I (260) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (266) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (273) heap_init: At 4008CD38 len 000132C8 (76 KiB): IRAM
I (280) spi_flash: detected chip: gd
I (283) spi_flash: flash io: dio
W (287) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (300) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (312) main_task: Started on CPU0
I (322) main_task: Calling app_main()

assert failed: prvInitialiseNewTask tasks.c:1061 (uxPriority < ( 25 ))

Backtrace: 0x40081646:0x3ffb4d40 0x40085819:0x3ffb4d60 0x4008b805:0x3ffb4d80 0x400873eb:0x3ffb4ea0 0x40088c64:0x3ffb4ed0 0x400d526e:0x3ffb4f10 0x400d5185:0x3ffb4f40 0x400d4f8a:0x3ffb4f70 0x400d4eb1:0x3ffb4f90 0x400d34e7:0x3ffb4fb0 0x400e327c:0x3ffb4fd0 0x400861a5:0x3ffb5000

ELF file SHA256: ee235219b
```

This happens with as minimal a program as:
```c
#include <stdio.h>

// Bring in all of the ubxlib public header files
#include "ubxlib.h"

// Bring in the application settings
#include "u_cfg_app_platform_specific.h"

void app_main(void)
{
	uPortInit();
}
```

I'm using ESP-IDF 5.2 with a TTGO LORA32 board connected to a ublox ZED-F9P via I2C.  If any further information or testing is required I will be happy to provide it.

RobMeades, 2024-04-26:
Hi, and thanks for posting.  ESP-IDF has recently add this assert, you probably need https://github.com/u-blox/ubxlib/commit/2380fe51699407a6fdf526be542abc5d94ff18fa to fix it (or just take head revision).

KjartanOli, 2024-04-26:
That seems to have done it.  Thank you.

closed 2024-04-26 by KjartanOli