u-blox/ubxlib #231: Unable to activate PDP context

jraats, 2024-05-07:
Hi [RobMeades](https://github.com/RobMeades),
After updating the ubxlib to the latest version (master) I'm unable to activate the PDP context. The problem is that the library doesn't read the actual status (the PDP context is in fact activated).

The problem is created in commit https://github.com/u-blox/ubxlib/commit/ac08677d4570aa46632317c28a4b114ad645f47d were the `+CGACT` status is not fully read.

The old implementation was:
```
uAtClientLockExtend(atHandle);
uAtClientTimeoutSet(atHandle,
					pInstance->pModule->responseMaxWaitMs);
uAtClientCommandStart(atHandle, "AT+CGACT?");
uAtClientCommandStop(atHandle);
ours = false;
for (size_t y = 0; (y < U_CELL_NET_MAX_NUM_CONTEXTS) && !ours; y++) {
	uAtClientResponseStart(atHandle, "+CGACT:");
	// Check if this is our context ID
	if (uAtClientReadInt(atHandle) == contextId) {
		ours = true;
		// If it is, 1 means activated
		activated = (uAtClientReadInt(atHandle) == 1);
	}
}
uAtClientResponseStop(atHandle);
```

which requests the `CGACT` status and reads for `U_CELL_NET_MAX_NUM_CONTEXTS` times the status.

The new implementation is:
```
uAtClientLockExtend(atHandle);
uAtClientTimeoutSet(atHandle,
					pInstance->pModule->responseMaxWaitMs);
uAtClientCommandStart(atHandle, "AT+CGACT?");
uAtClientCommandStop(atHandle);
ours = false;
for (size_t y = 0; (y < maxNumContexts) && !ours; y++) {
	uAtClientResponseStart(atHandle, "+CGACT:");
	// Check if this is our context ID
	if (uAtClientReadInt(atHandle) == contextId) {
		ours = true;
		// If it is, 1 means activated
		activated = (uAtClientReadInt(atHandle) == 1);
	}
}
uAtClientResponseStop(atHandle);
```

which requests the `CGACT` status and reads for `maxNumContexts` (=1) times the status.

The PDP context to activate is 1, but the for loop will stop after processing the first response and (wrongly) returning that the PDP context is not activated.

After changing https://github.com/u-blox/ubxlib/blob/c25c7b30f921c344d7506ebc4bd0eef392a4adc0/cell/src/u_cell_net.c#L1695 from 
```
for (size_t y = 0; (y < maxNumContexts) && !ours; y++) {
```
to
```
for (size_t y = 0; (y < U_CELL_NET_MAX_NUM_CONTEXTS) && !ours; y++) {
```

everything is working as intended.

Can you look at this bug and apply a fix for this?

Thanks in advance.

RobMeades, 2024-05-07:
Hi, and sorry about this.  The change in the loop limit was just an optimisation but what I can't understand is why we don't see the same problem here.  Which module type are you using?  I guess that in your case the module must be returning a `+CGACT: 0` before the "wanted" `+CGACT: 1`?

jraats, 2024-05-07:
Hi,
Hereby the logging with AT commands:
```
AT+COPS?

+COPS: 0,0,"NL KPN",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 2,0

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 2,0

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 2,0

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 2,0

OK
AT+CGACT=1,1

OK
AT+CGACT?

+CGACT: 0,0

+CGACT: 1,1

+CGACT: 2,0

OK
AT+CGACT=1,1

OK
U_CELL_NET: unable to activate a PDP context, is APN "mobileinternet.tele2.se" correct?
```

As you can see the `AT+CGACT?` replies with 3 different PDP statusses and only the first one (`+CGACT: 0,0`) is processed.
The first `+CGACT: 0,0` -> not activated
The second one `+CGACT: 1,1` -> activated
The tirth one `+CGACT: 2,0` -> not activated.

I'm using the `SARA-R510M8S-00B` modem.

RobMeades, 2024-05-07:
Fascinating: not a behaviour we see here, must be network dependent.  Fix (as you have proposed, remove the optimisation) is under test.

jraats, 2024-05-07:
If I can help you with additional information like modem firmware version ect. Don't hesitate to ask. It is strange that my modem is behaving odd.

jraats, 2024-05-07:
I'm currently using MNO profile 90. I used profile 100 (europe), but I needed to support Taiwan, so I switched to 90 (Global).
Do you want me to test with different profiles? Which kind of profiles do you want me to test? I'm living in the Netherlands so all europe profiles are fine for testing.

philwareublox, 2024-05-07:
Thanks. Global Profile 90 should only have one PDP Context defined. Could you show the AT log of the "+CGDCONT?" query?
I suspect this is showing up three contexts, but the profile shows only one. 

Can you confirm what firmware your R510M8S-00B module has? v2.xx?

Also if you try to set MNO Profile 100, and then go back to MNO Profile 90, do you still see the three contexts?

RobMeades, 2024-05-08:
@jraats: fix in commit ed77521aa73d4a0e68cf560851c30f0d2252c601.

syedqaisar, 2024-05-23:
> Hi [RobMeades](https://github.com/RobMeades), After updating the ubxlib to the latest version (master) I'm unable to activate the PDP context. The problem is that the library doesn't read the actual status (the PDP context is in fact activated).
> 
> The problem is created in commit [ac08677](https://github.com/u-blox/ubxlib/commit/ac08677d4570aa46632317c28a4b114ad645f47d) were the `+CGACT` status is not fully read.
> 
> The old implementation was:
> 
> ```
> uAtClientLockExtend(atHandle);
> uAtClientTimeoutSet(atHandle,
> 					pInstance->pModule->responseMaxWaitMs);
> uAtClientCommandStart(atHandle, "AT+CGACT?");
> uAtClientCommandStop(atHandle);
> ours = false;
> for (size_t y = 0; (y < U_CELL_NET_MAX_NUM_CONTEXTS) && !ours; y++) {
> 	uAtClientResponseStart(atHandle, "+CGACT:");
> 	// Check if this is our context ID
> 	if (uAtClientReadInt(atHandle) == contextId) {
> 		ours = true;
> 		// If it is, 1 means activated
> 		activated = (uAtClientReadInt(atHandle) == 1);
> 	}
> }
> uAtClientResponseStop(atHandle);
> ```
> 
> which requests the `CGACT` status and reads for `U_CELL_NET_MAX_NUM_CONTEXTS` times the status.
> 
> The new implementation is:
> 
> ```
> uAtClientLockExtend(atHandle);
> uAtClientTimeoutSet(atHandle,
> 					pInstance->pModule->responseMaxWaitMs);
> uAtClientCommandStart(atHandle, "AT+CGACT?");
> uAtClientCommandStop(atHandle);
> ours = false;
> for (size_t y = 0; (y < maxNumContexts) && !ours; y++) {
> 	uAtClientResponseStart(atHandle, "+CGACT:");
> 	// Check if this is our context ID
> 	if (uAtClientReadInt(atHandle) == contextId) {
> 		ours = true;
> 		// If it is, 1 means activated
> 		activated = (uAtClientReadInt(atHandle) == 1);
> 	}
> }
> uAtClientResponseStop(atHandle);
> ```
> 
> which requests the `CGACT` status and reads for `maxNumContexts` (=1) times the status.
> 
> The PDP context to activate is 1, but the for loop will stop after processing the first response and (wrongly) returning that the PDP context is not activated.
> 
> After changing
> 
> https://github.com/u-blox/ubxlib/blob/c25c7b30f921c344d7506ebc4bd0eef392a4adc0/cell/src/u_cell_net.c#L1695
> 
> from
> ```
> for (size_t y = 0; (y < maxNumContexts) && !ours; y++) {
> ```
> 
> to
> 
> ```
> for (size_t y = 0; (y < U_CELL_NET_MAX_NUM_CONTEXTS) && !ours; y++) {
> ```
> 
> everything is working as intended.
> 
> Can you look at this bug and apply a fix for this?
> 
> Thanks in advance.

Hello jraat.
i am using Ublox Sara r422 for cellular through tcp. and i want to ubxlib but i am unable to use it, can you help me.
thank you.

RobMeades, 2024-05-28:
@jraats: I am going to close this one now as I think we are done: please feel free to re-open it (or open a new issue) if there is more to discuss.

closed 2024-05-28 by RobMeades