u-blox/ubxlib #235: Refresh radio parameters delay

eeFLis, 2024-05-14:
Hi there

Wouldn't it make sense to just add a delay with the delta to the last function call instead of 500ms each time?
https://github.com/u-blox/ubxlib/blob/7f70e9073ffdc7b23374a892682d7166af233f4b/cell/src/u_cell_info.c#L686C1-L686C41

RobMeades, 2024-05-14:
Hi eeFLis: having discussed this with @philwareublox, suggestion is that we make the delay a conditionally-compiled value, implemented as a `#ifndef`/`#define`/`#endif` kinda thing and add a note to:

https://github.com/u-blox/ubxlib/blob/7f70e9073ffdc7b23374a892682d7166af233f4b/cell/api/u_cell_info.h#L69-L79

...pointing it out and suggesting that an application where power/timing is important (your case) might override it to 0 and manage any timing in the application instead.

Would that work?

eeFLis, 2024-05-14:
Yes, that works for us.
Out of curiosity, do you see a problem if the function is called in a loop without the delay?
As the response from the module is awaited, the module should not be overtasked.

RobMeades, 2024-05-14:
Good, change should be here in an hour or two.

> Out of curiosity, do you see a problem if the function is called in a loop without the delay?

I added that delay because I found in the past that a module didn't like being hit with the radio refresh-related commands frequently.  However, I can't recall now which module that was and it may, of course, be something that has since been fixed, hence my [unhelpful] answer is "no, unless you see one" :-).

RobMeades, 2024-05-14:
The test failures below are instance 14 doing its MGA/SER failure (Giovanni tells me that they have added filtering in their service back-end now and so the problem should hopefully go away for us sometime this week), the Windows instance (23) failing to get back all of the data it sent in the test `deviceSerial()`, the M9 GNSS module on the HPG board of instance 25 getting one fewer RRLP messages than we think it should in test `gnssMsgReceiveNonBlocking()`, the SARA-R422 module on instance 22 failing to perform an HTTP POST (we time-out after a 60 second wait) in `exampleHttpClient()`, the LENA-R8 module on instance 29 not getting the echo back of a secured DTLS packet in `exampleSocketsDtls()` and the `wifiStationConnectWrongPassphrase()` on instance 30, using uconnectExpress 2, being unable to connect to the Wifi network in a test which deliberately does NOT connect to the wifi network (unh?).  I have seen the latter failure once before and raised it with Peter but Peter retires tomorrow so I will raise it as a Jira ticket instead.

None of these random failures are likely to be associated with making the cellular radio parameter refresh delay configurable, hence merging.

RobMeades, 2024-05-14:
Fix can be found in 87a0a2c07b0ca85e08a209215a38bc70bf219cd4, will close this now: please feel free to re-open if there is more to discuss,

closed 2024-05-14 by RobMeades