u-blox/ubxlib #238: Using ubxlib to interface two ublox peripherals

jamal-mouline, 2024-05-22:
Hello,
This is more of a general high level question than an issue

Suppose the host MCU (for example a Nordic DK) is wired to two ublox peripherals over two distinct UARTs
A cellular modem (for example a SARA-R5)
and a GNSS receiver (for example the MIA-M10Q)

Would it be possible to establish two parallel instances, one for each peripheral?

I would imagine the init calls would be unique/common to both interfaces: uPortInit and uDeviceInit

Then distinction needs to be made at the open via uDeviceOpen to obtain two separate handles
the first DeviceCfg shall have a type of U_DEVICE_TYPE_CELL
the second DeviceCfg shall have a type of U_DEVICE_TYPE_GNSS

Please let me know if this is something that has already been tested
Thank you

RobMeades, 2024-05-22:
Hi Jamal.  Provided the UART HW blocks are distinct there should be no issue; this is something we test.

RobMeades, 2024-05-28:
Hi Jamal: I'm going to close this one now, please feel free to re-open if there is more to discuss.

closed 2024-05-28 by RobMeades

jamal-mouline, 2024-05-28:
Hi Rob,
Likely to revisit this one in the near future...
Thanks,