u-blox/ubxlib #240: Ublox MAX-M10: Getting  errror -5  for uGnssInfoGetVersions and -2 for uGnssPosGet

cid92, 2024-05-24:
Hi there,

I am trying to get nrf52840 to communicate with the Ublox module via I2C. 
The device tree is configure as follows:
```
/{
	gps {
		status = "okay";
		compatible = "u-blox,ubxlib-device-gnss";
		transport-type = "i2c0";
		i2c-already-open;
		i2c-address = <0x42>;
		i2c-max-segment-size = <256>;
		module-type = "U_GNSS_MODULE_TYPE_M10";

	};
};

&i2c0 {
	status = "okay";
	pinctrl-0 = <&i2c0_default>;
	pinctrl-1 = <&i2c0_sleep>;
	pinctrl-names = "default", "sleep";
	compatible = "nordic,nrf-twim";
}; 
```
Running the following in the main:
```
int main()
{
    // Toggle the GPS reset pin
    gpio_pin_configure_dt(&gpsExtInt, GPIO_OUTPUT_INACTIVE);
    k_sleep(K_MSEC(10));
    gpio_pin_set_dt(&gpsExtInt, 0);
    k_sleep(K_MSEC(500));
    uDeviceHandle_t ubx_device_handle = NULL;
    int32_t latitudeX1e7;
    int32_t longitudeX1e7;

    int err;
    if (0 != (err = uPortInit())) {
        LOG_ERR("uPortInit failed %d", err);
    }

    if (0 != (err = uPortI2cInit())) {
        LOG_ERR("uPortI2cInit failed %d", err);

    }

    if (0 != (err = uDeviceInit())) {
        LOG_ERR("uDeviceInit failed %d", err);

    }

    if (0 != (err = uDeviceOpen(NULL, &ubx_device_handle))) {
        LOG_ERR("uDeviceOpen failed %d", err);
    }

    uGnssVersionType_t version;

    if (0 == (err = uGnssInfoGetVersions(&ubx_device_handle, &version))) {
        LOG_INF("GNSS ver: %s", version.ver);
        LOG_INF("GNSS fw: %s", version.fw);
        LOG_INF("GNSS hw: %s", version.hw);
        LOG_INF("GNSS prot: %s", version.prot);
        LOG_INF("GNSS mod: %s", version.mod);
    }
    else
    {
        LOG_ERR("uGnssInfoGetVersions %d", err);
    }

    if (0  == (err = uGnssPosGet(&ubx_device_handle, &latitudeX1e7, &longitudeX1e7,
                    NULL, NULL, NULL, NULL, NULL, NULL))) {
        printf("I am here: https://maps.google.com/?q=%3.7f,%3.7f\n",
                ((double) latitudeX1e7) / 10000000,
                ((double) longitudeX1e7) / 10000000); 
    }
    else {
         LOG_ERR("uGnssPosGet %d", err);
    }
	return 0;

}
````
The log outcome:
```
*** Booting nRF Connect SDK 3758bcbfa5cd ***
U_PORT_BOARD_CFG: using GNSS device "gps" from the device tree, module-type 2 with pin-enable-power -1 (0xffffffff), pin-data-ready -1 (0xffffffff)...
U_PORT_BOARD_CFG: ...GNSS on I2C 0, i2c-address 0x42, i2c-clock-hertz 100000, i2c-max-segment-size 256, i2c-already-open.
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 1c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
[00:00:02.201,995] <err> main: uGnssInfoGetVersions -5
[00:00:02.202,026] <err> main: uGnssPosGet -2
```
Am I missing any important step? Was the GNSS initialised successful? It responded with 1C

RobMeades, 2024-05-24:
Hi there, sorry you're having trouble with this and thanks for posting.  Looking at the code, it might be as simple as you need this:

```
uGnssInfoGetVersions(ubx_device_handle, &version);
```

...rather than this:

```
uGnssInfoGetVersions(&ubx_device_handle, &version);
```

cid92, 2024-05-25:
Thank you @RobMeades . That resolve the problem

closed 2024-05-25 by cid92