u-blox/ubxlib #241: Unable to bring up GNSS!

mos216, 2024-05-24:
Helllo, 

I'm trying to test Lena R8 M10 in zephyr platform. so far I could test successfully CELL modem but no luck with GNSS modem. I'm not sure what the issue is but here is some screenshot from the code and logs output. 

my code is based on the example main_loc_gnss.c 

device tree:

`	cfg-device-cellular {
        compatible = "u-blox,ubxlib-device-cellular";
        status = "okay";
        transport-type = "uart2";
		uart-baud-rate = <115200>;
        module-type = "U_CELL_MODULE_TYPE_LENA_R8";
        pin-pwr-on = <38>;   // 1, 6
        pin-vint = <39>;     // 1, 7
        network = <&label_cfg_network_cellular_1nce &label_cfg_network_gnss_inside>;
    };

    label_cfg_network_cellular_simbase: cfg-network-cellular-simbase {
        compatible = "u-blox,ubxlib-network-cellular";
        status = "okay";
        apn = "simbase";
		timeout-seconds = <400>;
    };
	label_cfg_network_cellular_1nce: cfg-network-cellular-1nce {
        compatible = "u-blox,ubxlib-network-cellular";
        status = "okay";
        apn = "iot.1nce.net";
    };
	
	label_cfg_network_gnss_inside: cfg-network-gnss-inside {
        compatible = "u-blox,ubxlib-network-gnss";
        status = "okay";
        module-type = "U_GNSS_MODULE_TYPE_ANY";
    };`

device configuration 
static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_CELL,
    .deviceCfg = {
        .cfgCell = {
            .moduleType = U_CELL_MODULE_TYPE_LENA_R8,
            .pSimPinCode = NULL, /* SIM pin */
            .pinEnablePower = U_CFG_APP_PIN_CELL_ENABLE_POWER,
            .pinPwrOn = 38,
            .pinVInt = 39,
            .pinDtrPowerSaving = U_CFG_APP_PIN_CELL_DTR
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
    .transportCfg = {
        .cfgUart = {
            .uart = 2,
            .baudRate = U_CELL_UART_BAUD_RATE,
            .pinTxd = 40,
            .pinRxd = 41,
            .pinCts = 43,
            .pinRts = 42,
            .pPrefix = NULL
        },
    },
};
// NETWORK configuration for GNSS
static const uNetworkCfgGnss_t gNetworkCfg = {
    .type = U_NETWORK_TYPE_GNSS,
    .moduleType = U_GNSS_MODULE_TYPE_M10,
    .devicePinPwr = U_CFG_APP_CELL_PIN_GNSS_POWER, // The pins of the *cellular* *module* that are connected
    .devicePinDataReady = U_CFG_APP_CELL_PIN_GNSS_DATA_READY // to the GNSS chip's power and Data Ready lines
};

main function 

 uDeviceHandle_t devHandle = NULL;
    uLocation_t location;
    int32_t returnCode;

    // Set an out of range value so that we can test it later
    location.timeUtc = -1;

    // Initialise the APIs we will need
    uPortInit();
    uDeviceInit();

    // Open the cellular device
    returnCode = uDeviceOpen(&gDeviceCfg, &devHandle);
    LOG_INF("Opened cellular device with return code %d.\n", returnCode);

    if (returnCode == 0) {
        LOG_INF("Bringing up GNSS...\n");
        if (uNetworkInterfaceUp(devHandle, U_NETWORK_TYPE_GNSS,
                                &gNetworkCfg) == 0) 

            // Now get location
            if (uLocationGet(devHandle, U_LOCATION_TYPE_GNSS,
                             NULL, NULL, &location, NULL) == 0) {
                printLocation(location.latitudeX1e7, location.longitudeX1e7);
            } else {
                LOG_INF("Unable to get a location fix!\n");
            }

            // When finished with the GNSS network layer
            LOG_INF("Taking down GNSS...\n");
            uNetworkInterfaceDown(devHandle, U_NETWORK_TYPE_GNSS);
        } else {
            LOG_INF("Unable to bring up GNSS!\n");
        }

        // Close the device
        if (uDeviceClose(devHandle, true) != 0) {
            // Device has not responded to power off request, just release resources
            uDeviceClose(devHandle, false);
        }
    } else {
        LOG_INF("Unable to bring up the cellular device!\n");
    }

    // Tidy up
    uDeviceDeinit();
    uPortDeinit();
    LOG_INF("Done.\n");

Logs output 

*** Booting nRF Connect SDK v3.5.99-ncs1 ***
[00:00:00.423,583] <inf> ubx_board_cfg: U_PORT_BOARD_CFG: using CELLULAR device "cfg-device-cellular" from the device tree, module-type 8 on UART 2, uart-baud-rate 115200 with pin-enable-power -1 (0xffffffff), pin-pwr-on 38 (0x26), pin-vint 39  (0x27), pin-dtr-power-saving -1  (0xffffffff).

[00:00:00.440,216] <inf> ubx_ell_pwr: U_CELL_PWR: powering on, module is already on.

[00:00:11.525,482] <inf> main: Opened cellular device with return code 0.

[00:00:11.525,543] <inf> main: Bringing up GNSS...

[00:00:11.525,634] <inf> ubx_board_cfg: U_PORT_BOARD_CFG: using GNSS network configuration associated with CELLULAR device "cfg-device-cellular" from the device tree, GNSS module-type 3, device-pin-pwr -1, device-pin-data-ready -1.

[00:00:22.467,071] <inf> ubx_gnss: U_GNSS: initialising with ENABLE_POWER pin 
[00:00:22.467,102] <inf> ubx_gnss: not connected
[00:00:22.467,132] <inf> ubx
[00:00:44.800,994] <inf> main: Unable to bring up GNSS!

[00:00:44.801,055] <inf> ubx_ell_pwr: U_CELL_PWR: powering off with AT command.
[00:01:32.513,702] <inf> main: Done.

RobMeades, 2024-05-24:
Hi, thanks for posting and sorry you're having trouble with this.  Could you let us know what the [return value](https://github.com/u-blox/ubxlib/blob/master/DEBUG_HINTS.md#reading-error-return-codes) from `uNetworkInterfaceUp()` for GNSS is?

mos216, 2024-05-24:
Hello,
it is -10  (U_ERROR_COMMON_DEVICE_ERROR)

RobMeades, 2024-05-24:
Thanks; I'm not seeing the `ubxlib` AT log output in the logs you pasted in above, it should be [on by default](https://github.com/u-blox/ubxlib/blob/b645c901a44ecc2aec125aaacdd5e9a0511e27f9/common/device/src/u_device_private_cell.c#L161), pushed out over the same stream as you get the `U_CELL_PWR: powering on, module is already on` prints.

AT communications with the module _must_ be happening or you wouldn't get as far as `Opened cellular device with return code 0`, and you don't seem to be switching it off, so I'm puzzled at to why it is not visible.  During the `uDeviceOpen()` call it should appear on the log stream something like this (taken from one of our LENA-R8 test runs):

```
U_CELL_PWR: powering on.
U_AT_CLIENT_0-1: possible URC data readable 1, already buffered 0.
[00]U_AT_CLIENT_0-1: URC checking done.
AT
U_AT_CLIENT_0-1: AT error -10.
AT
U_AT_CLIENT_0-1: AT error -10.
AT
U_AT_CLIENT_0-1: AT error -10.
AT
U_AT_CLIENT_0-1: AT error -10.
AT
U_AT_CLIENT_0-1: timeout.
U_AT_CLIENT_0-1: AT error -10.
U_AT_CLIENT_0-1: AT error -10.
U_AT_CLIENT_0-1: stop tag not found.
AT
U_AT_CLIENT_0-1: AT error -10.
AT
OK
U_CELL_PWR: powering on, module is alive.
ATE0
OK
AT+CMEE=2
OK
AT+UDCONF=1,0
OK
ATI9
02.00,A01.40
OK
AT&C1
OK
AT&D0
OK
AT&K3
OK
```

Since LENA-R8 does not support access to the internal GNSS device via CMUX, so `ubxlib` will be talking to it via the `AT+UGUBX` AT commands and so I'd like to see what those are doing.

mos216, 2024-05-24:
Here you is my actual logs output 

*** Booting nRF Connect SDK v3.5.99-ncs1 ***
U_PORT_BOARD_CFG: using CELLULAR device "cfg-device-cellular" from the device tree, module-type 8 on UART 2, uart-baud-rate 115200 with pin-enable-power -1 (0xffffffff), pin-pwr-on 38 (0x26), pin-vint 39  (0x27), pin-dtr-power-saving -1  (0xffffffff).
U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin 39 (0x27) (and is 1 when module is on).
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT

+UUSTATUS: READY
AT
AT

OK
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

02.00,A01.40

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=4,1300

OK
AT+CPSMS?

+CPSMS: 0,,,"10100100","00100100"

OK
AT+UGPRF?

+UGPRF:2,0,""

OK
AT+CFUN=4

OK
U_PORT_BOARD_CFG: using GNSS network configuration associated with CELLULAR device "cfg-device-cellular" from the device tree, GNSS module-type 3, device-pin-pwr -1, device-pin-data-ready -1.
AT+CMUX=0,0,,128

OK
ATE0

OK
AT+CMEE=2

OK
[00:00:08.056,915] <inf> main: Opened cellular device with return code 0.

[00:00:08.056,945] <inf> main: Bringing up GNSS...

U_GNSS: initialising with ENABLE_POWER pin not connected, transport type AT.
AT

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGIND=1

OK
AT+UGPS?

+UGPS: 0,0,0

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGPS=1,15,127

+CME ERROR: Operation not supported
AT

OK
AT+UGPS=1,15,127

+CME ERROR: Operation not supported
AT

OK
AT+UGPS=1,15,127

+CME ERROR: Operation not supported
U_CELL_PWR: powering off with AT command.
AT

OK
AT+UPSV=0

OK
AT+CPWROFF

+CPOF: MS OFF

OK
[00:00:46.116,516] <inf> main: Unable to bring up GNSS -10!

[00:00:48.530,273] <inf> main: Done.

RobMeades, 2024-05-24:
Ah, that's better, thanks.  The issue is this:

```
AT+UGPS=1,15,127
+CME ERROR: Operation not supported
```

The last parameter to `AT+UGPS` tells the cellular module which GNSS system types to enable in the built-in GNSS device.  `ubxlib` asks for all of them and _most_ of our modules simply set the ones they can and ignore ones they can't (sending a `+UGIND` URC to say what it has done).  For some reason LENA-R8 doesn't do this and instead rejects the entire command and doesn't switch GNSS on.

To make this work for you, you will need to define [U_CELL_LOC_GNSS_SYSTEM_TYPES](https://github.com/u-blox/ubxlib/blob/b645c901a44ecc2aec125aaacdd5e9a0511e27f9/cell/api/u_cell_loc.h#L139) when building `ubxlib` and set it to something that LENA-R8 is happy with, maybe just `1` for now, to get it working.

I will make a note to find a more organised way to make LENA-R8 happy.

mos216, 2024-05-24:
Hi Rob, 
By modifing U_CELL_LOC_GNSS_SYSTEM_TYPES no changes happens on AT command , it sends still 127 , I had to change U_GNSS_PWR_SYSTEM_TYPES to get it sending 1 instead of 127 

it still not working but has now different error 

+UUSTATUS: READY

OK
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

OK
AT&C1

02.00,A01.40

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=4,1300

OK
AT+CPSMS?

OK
AT+UGPRF?

+CPSMS: 0,,,"10100100","00100100"

OK
AT+CFUN=4

+UGPRF:2,0,""

OK
U_PORT_BOARD_CFG: using GNSS network configuration associated with CELLULAR device "cfg-device-cellular" from the device tree, GNSS module-type 3, device-pin-pwr -1, device-pin-data-ready -1.

OK
AT+CMUX=0,0,,128

OK
ATE0

OK
AT+CMEE=2

OK
[00:00:12.508,911] <inf> main: Opened cellular device with return code 0.

[00:00:12.508,972] <inf> main: Bringing up GNSS...

U_GNSS: initialising with ENABLE_POWER pin not connected, transport type AT.
AT

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGIND=1

OK
AT+UGPS?

+UGPS: 0,0,0

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGPS=1,15,1

OK
AT+UGUBX="B5620A0400000E34"
U_GNSS: sent UBX command b5 62 0a 04 00 00 0e 34.

+UGUBX: "no message"

OK
U_GNSS_PWR: could not identify the module type.
U_CELL_PWR: powering off with AT command.
AT+UPSV=0

OK
AT+CPWROFF

+CPOF: MS OFF

OK
[00:00:19.555,480] <inf> main: Unable to bring up GNSS -24!

[00:00:21.917,327] <inf> main: Done.

RobMeades, 2024-05-24:
Ah, sorry for misdirecting you.

```
U_GNSS: sent UBX command b5 62 0a 04 00 00 0e 34.
+UGUBX: "no message"
OK
```

`ubxlib` is sending the UBX command `0x0a 0x04`, `UBX-MON-VER`, to find out the module type, which the GNSS device really is supposed to respond to, `no message` indicating that the cellular module has not got a response from it.  Hmph.

You could try changing `U_GNSS_MODULE_TYPE_ANY` in your `cfg-network-gnss-inside` to be `U_GNSS_MODULE_TYPE_M10`; `ubxlib` will then not ask the GNSS device what type it is and move on to the next thing, we could see if it responds to that?

mos216, 2024-05-24:
aah oke indeed this error message is no more displayed but now get still no location fix 

U_GNSS: initializing with ENABLE_POWER pin not connected, transport type AT.
AT

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGIND=1

OK
AT+UGPS?

+UGPS: 0,0,0

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGPS=1,15,1

OK
AT+UGUBX="B562010700000819"
U_GNSS: sent UBX command b5 62 01 07 00 00 08 19.

+UGUBX: "no message"

OK
AT+UGPS=0

+UUGIND: 0,1

OK
U_CELL_PWR: powering off with AT command.
AT+UPSV=0

OK
AT+CPWROFF

+CPOF: MS OFF

OK
[00:00:18.420,166] <inf> main: Unable to get a location fix!

[00:00:18.420,196] <inf> main: Taking down GNSS...

[00:00:21.336,761] <inf> main: Done.

RobMeades, 2024-05-24:
OK, so now the GNSS chip is not responding to the UBX command `0x01 0x07`, `UBX-NAV-PVT`, the command which retrieves the position fix from it.  Eh?

Can you give me some idea of the time intervals between the commands/responses?  Compiling `ubxlib` with `U_AT_CLIENT_PRINT_WITH_TIMESTAMP` is an easy way to do that.  Just wondering if LENA-R8 needs extra time for the GNSS device to power on or some such?  In the meantime I will enquire internally with the people who know LENA-R8.

mos216, 2024-05-24:
actually when I run it in debug mode then it gets the location fixed but when I run it in normal mode, it is not able to get the location fixed. it seems as u said , that needs some interval after powering gnss modem on. 
so I added some delay before calling uLocationGet and then it starts to work

RobMeades, 2024-05-24:
Interesting: my internal contact confirms that the procedure is correct.  How much delay do you think seems to be required?

BTW, what HW are you using?  Apparently pin 99 of the module needs to be supplied with 1.8V for the GNSS device to be powered, but I guess that is being supplied in your case?

mos216, 2024-05-24:
for time-delay I just started with 5 sec , I didn't have chance to fine-tune it yet. But I hope to fix it without delay. 
for the hardware I use the evk-8 of ublox (https://www.u-blox.com/en/product/evk-r8) I just connect this board to external mcu (nrf5340 dk) using Tx, Rx. 

Here is the output logs without delay 

U_PORT_BOARD_CFG: using GNSS network configuration associated with CELLULAR device "cfg-device-cellular" from the device tree, GNSS module-type 2, device-pin-pwr -1, device-pin-data-ready -1.
0000/01/01 00:00:11.764: AT+CMUX=0,0,,128
0000/01/01 00:00:11.820: 
0000/01/01 00:00:11.822: OK
0000/01/01 00:00:12.460: ATE0
0000/01/01 00:00:12.514: 
0000/01/01 00:00:12.516: OK
0000/01/01 00:00:12.540: AT+CMEE=2
0000/01/01 00:00:12.595: 
0000/01/01 00:00:12.598: OK
[00:00:12.254,730] <inf> main: Opened cellular device with return code 0.

[00:00:12.254,760] <inf> main: Bringing up GNSS...

U_GNSS: initialising with ENABLE_POWER pin not connected, transport type AT.
0000/01/01 00:00:17.720: AT
0000/01/01 00:00:17.773: 
0000/01/01 00:00:17.775: OK
0000/01/01 00:00:17.778: ATI
0000/01/01 00:00:17.832: 
0000/01/01 00:00:17.834: LENA-R8001M10-00C-00
0000/01/01 00:00:17.838: 
0000/01/01 00:00:17.841: OK
0000/01/01 00:00:17.865: AT+UGIND=1
0000/01/01 00:00:17.919: 
0000/01/01 00:00:17.921: OK
0000/01/01 00:00:17.945: AT+UGPS?
0000/01/01 00:00:17.999: 
0000/01/01 00:00:18.001: +UGPS: 0,0,0
0000/01/01 00:00:18.004: 
0000/01/01 00:00:18.007: OK
0000/01/01 00:00:18.030: ATI
0000/01/01 00:00:18.083: 
0000/01/01 00:00:18.086: LENA-R8001M10-00C-00
0000/01/01 00:00:18.090: 
0000/01/01 00:00:18.092: OK
0000/01/01 00:00:18.596: AT+UGPS=1,15,1
0000/01/01 00:00:18.651: 
0000/01/01 00:00:18.653: OK
0000/01/01 00:00:18.677: AT+UGUBX="B562010700000819"
U_GNSS: sent UBX command b5 62 01 07 00 00 08 19.
0000/01/01 00:00:18.729: 
0000/01/01 00:00:18.731: +UGUBX: "no message"
0000/01/01 00:00:18.735: 
0000/01/01 00:00:18.738: OK
0000/01/01 00:00:19.241: AT+UGPS=0
0000/01/01 00:00:19.255: 
0000/01/01 00:00:19.257: +UUGIND: 0,1
0000/01/01 00:00:19.291: 
0000/01/01 00:00:19.294: OK
U_CELL_PWR: powering off with AT command.
0000/01/01 00:00:19.401: AT+UPSV=0
0000/01/01 00:00:19.455: 
0000/01/01 00:00:19.457: OK
0000/01/01 00:00:19.481: AT+CPWROFF
0000/01/01 00:00:19.555: 
0000/01/01 00:00:19.558: +CPOF: MS OFF
0000/01/01 00:00:19.561: 
0000/01/01 00:00:19.564: OK
[00:00:19.260,833] <inf> main: Unable to get a location fix -11 

[00:00:19.260,864] <inf> main: Taking down GNSS...

[00:00:22.196,899] <inf> main: Done.

RobMeades, 2024-05-28:
> is it possible to run the gnss module without need to cell-module, since the gnss module has its own power supply?

From the [integration manual](https://content.u-blox.com/sites/default/files/documents/LENA-R8_SysIntegrManual_UBX-22015376.pdf), section 1.5.3.4:

> The GNSS system integrated in the LENA-R8001M10 module can be operated fully independently from the cellular system integrated in the same module, but, as shown in Figure 2, the two systems are internally interconnected by dedicated I2C interface, allowing access to the GNSS system directly via the cellular system, which relays control messages to the GNSS system.

[image]

So I guess that's a "yes".

RobMeades, 2024-05-28:
On your original question, I have pushed to a [preview branch](https://github.com/u-blox/ubxlib/tree/preview_fix_cell_gnss_lena_r8_rmea) a proposed fix for the LENA-R8 system types and the LENA-R8 GNSS-power-on-delay LENA-R8 issues: would it be possible for you to try it?

If so, please:

(a) put the module type back to `U_GNSS_MODULE_TYPE_ANY` in your `cfg-network-gnss-inside`, leaving your override of the GNSS system types in place, and see if the GNSS device is now powered-up and recognised correctly,
(b) as a second step, also try removing your override of the GNSS system types, see if the revised setting in the new code fixes it.

Should either (a) or (b) fail, please post an AT log from power-on here and I'll get on it.

RobMeades, 2024-05-28:
I have updated the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_fix_cell_gnss_lena_r8_rmea) to:

1) rebase it on 759020036fdcff164934c697e2780f7c102b20cd (the fix for #242),
2) revise the fix for setting the GNSS system types in the `AT+UGPS` command in a way that works for LENA-R8: I was puzzled as to why you had to change `U_GNSS_PWR_SYSTEM_TYPES` instead of `U_CELL_LOC_GNSS_SYSTEM_TYPES`; this was because the mechanism for setting cellular-module-specific GNSS system types only worked for cellular modules that supported CMUX-access to the built-in GNSS chip, it did not work for the LENA-R8xxxM10 case, which is the only cellular module where the old `AT+UGUBX` mechanism is the sole way to access the built-in GNSS chip.

To summarise:

- the built-in GNSS module should, I hope, now always switch on: if you have time please give the updated [preview branch](https://github.com/u-blox/ubxlib/tree/preview_fix_cell_gnss_lena_r8_rmea) a try,
- the issue seen in the "option (a)" log you pasted above is, unfortunately, not something that `ubxlib` can do anything about: it can be minimised by setting the assistance types to none (i.e. override [U_GNSS_AIDING_TYPES](https://github.com/u-blox/ubxlib/blob/b645c901a44ecc2aec125aaacdd5e9a0511e27f9/gnss/src/u_gnss_pwr.c#L74) to zero) and by calling `uLocationGet()` in a loop until you do receive a location (or give up trying).

mos216, 2024-05-30:
Hi Rob,
Many thanks for the update now it works indeed without need to set U_GNSS_PWR_SYSTEM_TYPE externally. only the issue as you described still with getting the location fixed, it still more time than expected. My intend finally is not to use GNSS as part of CELL device but as an independent GNSS device to satisfy our application requirement. I will give it try as independent gnss module and come back to you in case there are issues

RobMeades, 2024-05-30:
> I will give it try as independent gnss module and come back to you in case there are issues

That makes sense: you should be able to just use it with `uDeviceOpen()` as a UART-connected GNSS device and it _should_ work.  Let us know if it does not for any reason.  Will close this issue now.

closed 2024-05-30 by RobMeades

RobMeades, 2024-06-06:
The fixes that were in the [preview branch](https://github.com/u-blox/ubxlib/tree/preview_fix_cell_gnss_lena_r8_rmea) are now on `master` here in commits 34f820ee0d9cc36ee9bbc894ed2bff4057ee1892 and d896fc63ae2bc04b69ef47f9b30e7d3eab11bb88.  I will delete the preview branch some time next week.