u-blox/ubxlib #243: esp32c6 i2c m10

jimakos96, 2024-05-28:
Hi 
I am using esp-idf to program the esp32c6 with the sparkfun sam-m10q using I2C . 
My configuration using the ubxlib is the following : 

`  uDeviceCfg_t deviceConfig ;
    uDeviceHandle_t devHandle = NULL;
    int32_t returnCode;
    int32_t guardCount = 0;
    MyContext_t context = {0};

    
    deviceConfig.deviceType = U_DEVICE_TYPE_GNSS;
    deviceConfig.deviceCfg.cfgGnss.moduleType = U_GNSS_MODULE_TYPE_M10;
    deviceConfig.deviceCfg.cfgGnss.pinDataReady = -1;
    deviceConfig.deviceCfg.cfgGnss.pinEnablePower = U_CFG_APP_PIN_GNSS_ENABLE_POWER;
    deviceConfig.deviceCfg.cfgGnss.i2cAddress = 0x42;
    deviceConfig.transportType = U_DEVICE_TRANSPORT_TYPE_I2C;
    deviceConfig.transportCfg.cfgI2c.i2c = 1;
    deviceConfig.transportCfg.cfgI2c.pinScl = 18;
    deviceConfig.transportCfg.cfgI2c.pinSda = 19;

    // Initialise the APIs we will need
    uPortInit();
    uPortI2cInit(); // You only need this if an I2C interface is used
    uDeviceInit();
    // Open the device
    returnCode = uDeviceOpen(&deviceConfig, &devHandle);
    printf("Opened device with return code %ld.\n", returnCode);
`

I am getting the error code -5  ,which i think has to do with the parameters in my setup. Could you  tell me what may cause this error and how to fix it ?

RobMeades, 2024-05-28:
Hi and thanks for posting, sorry you're having trouble with this.

From the code you have posted above it looks as though your `deviceConfig` is not a `const` type, which the compiler would naturally zero for you, hence you likely need to set it to 0 first, i.e.:

`uDeviceCfg_t deviceConfig = {0};`

...otherwise the fields that you are not populating in the structure could contain anything, which could cause `U_ERROR_COMMON_INVALID_PARAMETER`.

Give this a try and let us know how it goes.

RobMeades, 2024-05-28:
> assert failed: periph_module_enable periph_ctrl.c:58 (periph < PERIPH_MODULE_MAX)

This is not a `ubxlib` file, not something I have ever seen before I'm afraid.  Which version of ESP-IDF are you using? We test with version 5.0.3 and we have [customers who are using 5.2](#229) successfully with I2C.

You will likely need to find out what the values of `periph` and `PERIPH_MODULE_MAX` are and where the value for `periph` is coming from.

jimakos96, 2024-05-28:
i am using the ESP-IDF v5.2.1-dirty

RobMeades, 2024-05-28:
Checking the [ESP-IDF pages for ESP32C6, it says](https://docs.espressif.com/projects/esp-idf/en/release-v5.2/esp32c6/api-reference/peripherals/i2c.html):

> ESP32-C6 has 1 I2C controller (also called port), responsible for handling communication on the I2C bus. A single I2C controller can be a master or a slave.

ESP-IDF usually numbers ports from zero, so it might be that you only have I2C port 0, so you might need:

```
deviceConfig.transportCfg.cfgI2c.i2c = 0;
```