u-blox/ubxlib #244: uCellInfoGetTimeUtc - AT+CCLK sometimes returns invalid date

arnoutdekimo, 2024-05-28:
Hi,

I' pretty sure the issue does not comes from ubxlib itself, but just wanted to report that during testing, I noticed that -sometimes- uCellInfoGetTimeUtc returned unexpected values. From the test reference code and datasheet, I assumed the module would return correct time (e.g. through fetching this from the cellular network after association)

[image]

[image]

I let a few devices, all using a SARA-R4 chipset, run and after cellular association fetch the time using this API. Almost always this returns a valid response, but -sometimes- I get a consistent reponse around the year 2080.. which appears to be the factory default settings for the clock on the sara r4:

[image]

Unfortunately the datasheet does not specify where the network-based clock update comes from exactly, which guarantees are given on correctness, or any official way to check that the time has been updated somehow externally. I wanted to just make a ticket so perhaps in the future some finds it and gets the heads-up for this function.

Again, I don't see an issue with the ubxlib parsing, so feel free to close this issue immediately:)

Kind regards,
Arnout

RobMeades, 2024-05-28:
Hi Arnout: we have seen this behaviour when the network doesn't support time values at all, hence [the note on uCellInfoGetTimeUtc()](https://github.com/u-blox/ubxlib/blob/b645c901a44ecc2aec125aaacdd5e9a0511e27f9/cell/api/u_cell_info.h#L336-L344) about doing range checking, but doing it _sometimes_ is a bit weird, unless maybe the module had lost and regained service around then, or maybe was caught right in the act of changing cell and in a kind of no-mans land?

arnoutdekimo, 2024-05-28:
Hi Rob,

Ah great, there is already a note :) 
I'll indeed do the range checking, and worst case will need to do some NTP implementation myself.

Thanks for your swift reply!

RobMeades, 2024-05-28:
NP, thanks for posting; the information may be useful to others.  I will close this now.