u-blox/ubxlib #245: esp idf i2c legacy driver issue

jimakos96, 2024-05-30:
Hello 

I can see on the  port/platform/esp-idf/src/u_port_i2c.c that the legacy i2c driver is used for esp-idf , so for now only the legacy i2c driver is used , is there any plan on updating to the new version of driver (i2c_master.h) , because the legacy driver will be deprecated in the future .

RobMeades, 2024-05-30:
Hi there: it is indeed on the TODO list; we will probably combine it with moving forward to ESP-IDF 5.2.  We can't drop support for the old I2C driver of course, so it will make the ESP-IDF port-layer code a bit messy, and it will need to be done under a compilation switch so as not to break existing users.

Suggest we leave this issue open and then we can updated it when done.

RobMeades, 2024-06-11:
I've just begun implementing against the new ESP-IDF I2C API, however it seems to be missing one vital feature: it does not permit I2C stop bits to be suppressed.  In the old API one had full control over this; we need it in order to determine the number of bytes available in the GNSS device (write 0xFD followed by no stop bit to the GNSS device, read two bytes and that gives you the amount of data it has buffered).  Unless I've missed something...?

I have raised a feature request on ESP-IDF: https://github.com/espressif/esp-idf/issues/13952

RobMeades, 2024-06-13:
FYI, the ESP-IDF I2C API _does_ have that feature, in fact [i2c_master_transmit_receive()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2c.html#_CPPv427i2c_master_transmit_receive23i2c_master_dev_handle_tPK7uint8_t6size_tP7uint8_t6size_ti) does exactly that, it just doesn't say that it does that in the API description.

I'm on to the next issue, which is that it doesn't seem possible to implement our [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) API on top of the new ESP-IDF I2C API.  I have asked (https://github.com/espressif/esp-idf/issues/13968) if Espressif can see a way to do it.

RobMeades, 2024-06-13:
Hmmm, I'm now hitting a couple more issues which seem to be known and fixed on ESP-IDF `main` but not yet ported to a release branch:

https://github.com/espressif/esp-idf/issues/13962
https://github.com/espressif/esp-idf/issues/13922

I think we might wait a little while and let it settle down...

RobMeades, 2024-07-04:
FYI, Espressif have now provided an API that allows [uPortI2cAdopt()](https://github.com/u-blox/ubxlib/blob/ba0a0f17cbd3028dfa2097decccd48458e08b021/port/api/u_port_i2c.h#L122) to be implemented, see https://github.com/espressif/esp-idf/commit/992d8bc5f20556adb62571d55a1517a9848d4b51.

Once this, and hopefully the two fixed issues mentioned above, have made it into a tagged release we will try again.

RobMeades, 2024-07-25:
FYI, it looks like ESP-IDF are gearing up for a 5.3 release; it is holiday season here now so likely we can try this in September when 5.3 is released.

synologic, 2024-08-16:
5.3 is released :)

Beer is guaranteed for this port to be available soon ;-)

RobMeades, 2024-08-28:
B*gger: looks like the [feature we needed](https://github.com/espressif/esp-idf/issues/13968), even thought it was [committed to ESP-IDF master](https://github.com/espressif/esp-idf/commit/992d8bc5f20556adb62571d55a1517a9848d4b51) two months ago, isn't in the ESP-IDF 5.3 release, it stops on the revision of [i2c_master.c](https://github.com/espressif/esp-idf/commits/master/components/esp_driver_i2c/i2c_master.c) just before.  [I have asked](https://github.com/espressif/esp-idf/issues/13968#issuecomment-2314903466) when it will make it into an official release.

In the mean time I will try testing with [ESP-IDF master](https://github.com/espressif/esp-idf); we might then choose to test with a specific commit hash ahead of v5.3.

synologic, 2024-08-29:
So i'm guessing we should stick to the old API and disable the warning at the boot :)))

RobMeades, 2024-08-29:
I guess it depends how quickly you need to get working: you will see from the issue I raised on ESP-IDF that someone else had raised what _looks_ like the same issue (talking to a u-blox GNSS chip) back at the start of July, [updated just last week](https://github.com/espressif/esp-idf/issues/14129#issuecomment-2295882059), to which *mythbuster5* responded that they would take a look, but then a few hours ago the *espressif-bot* marked that issue as "done" with resolution "NA", which seems a bit odd.

What is the warning you get at boot?  Some form of run-time deprecation thingy?

RobMeades, 2024-08-30:
Update: looks like the issue that was [raised earlier](https://github.com/espressif/esp-idf/issues/14129) on ESP-IDF concerning talking to a u-blox GNSS chip using the new I2C API has, in fact, been fixed on ESP-IDF `master` in commit https://github.com/espressif/esp-idf/commit/8aee667873143a4f2c029d21a4bfff358d287db4.  I've just done a very quick test with that included and things are looking good, so don't lose hope!

I will need to do some more testing but, if things continue to look good, I will post a preview branch here with the changes to use the new ESP-IDF I2C API in it so that you can try it if you wish, while I get it put through regression test/review here.

RobMeades, 2024-09-02:
I have pushed [preview_esp_idf_i2c_new_api_rmea](https://github.com/u-blox/ubxlib/tree/preview_esp_idf_i2c_new_api_rmea) here: this uses the new ESP-IDF I2C API by default if you compile it against ESP-IDF `master` (and of course you need commit https://github.com/espressif/esp-idf/commit/8aee667873143a4f2c029d21a4bfff358d287db4 or later) and passes our regression testing.

Once Espressif have included the fixes that are in commits https://github.com/espressif/esp-idf/commit/992d8bc5f20556adb62571d55a1517a9848d4b51 and https://github.com/espressif/esp-idf/commit/8aee667873143a4f2c029d21a4bfff358d287db4 in a release branch we will update `ubxlib` `master` to user the new ESP-IDF I2C API for that ESP-IDF release and later; we will let you know when that is done.  We will delete the preview branch sometime after that.

closed 2024-11-11 by manemannen