u-blox/ubxlib #247: best way to declare multiple uDeviceCfg_t in zephyr

mos216, 2024-06-05:
Hello, 
I noticed that uDeviceCfg_t has more properties than the one in the device tree. for example, pSimPinCode can be set only by the uDeviceCfg_t  structure but not by the device tree. what is the best way to combine properties of uDeviceCfg_t  structure with the device-tree when uDeviceOpen is called since we must pass NULL to uDeviceOpen if device interface is declared in device-tree?

Thank you in advance!

RobMeades, 2024-06-05:
Hi: all of the device and network configuration items that have an effect in the [uDeviceCfg_t](https://github.com/u-blox/ubxlib/blob/759020036fdcff164934c697e2780f7c102b20cd/common/device/api/u_device.h#L477), [uNetworkCfgCell_t](https://github.com/u-blox/ubxlib/blob/759020036fdcff164934c697e2780f7c102b20cd/common/network/api/u_network_config_cell.h#L55), [uNetworkCfgGnss_t](https://github.com/u-blox/ubxlib/blob/759020036fdcff164934c697e2780f7c102b20cd/common/network/api/u_network_config_gnss.h#L55), [uNetworkCfgWifi_t](https://github.com/u-blox/ubxlib/blob/759020036fdcff164934c697e2780f7c102b20cd/common/network/api/u_network_config_wifi.h#L73) and [uNetworkCfgBle_t](https://github.com/u-blox/ubxlib/blob/759020036fdcff164934c697e2780f7c102b20cd/common/network/api/u_network_config_ble.h#L55) structures should be settable through the device tree.  The SIM PIN code is not there only because it has to always be set to `NULL`; we added it to the `uDeviceCfg_t` structure when we first created it years ago because it seemed logical but `ubxlib` has never supported setting a SIM PIN (the SIM PIN does not offer any security to speak of as it ends up being sent transparently over a serial port every time the cellular module is powered on).

So it _should_ be possible to do everything you need to do through the device tree; let us know if not.

RobMeades, 2024-06-14:
@mos216: are you OK to close this one now?

mos216, 2024-06-15:
Yes sure! thanks again!

closed 2024-06-15 by RobMeades