u-blox/ubxlib #249: Unable to start continuous location!

mos216, 2024-06-06:
Hello,

Currently, trying to get the sample (loc_cell_gnss_continues working but so far no luck. Here below the output log

+UUSTATUS: READY
AT
AT

OK
U_CELL_PWR: powering on, module is already on.
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

02.00,A01.40

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=4,1300

OK
AT+UGPRF?

+UGPRF:2,0,""

OK
AT+CFUN=4

OK
Opened cellular device with return code 0.
Bringing up GNSS...
U_PORT_BOARD_CFG: using GNSS network configuration associated with CELLULAR device "cfg-device-cellular" from the device tree, GNSS module-type 2, device-pin-pwr -1, device-pin-data-ready -1.
AT+CMUX=0,0,,128

OK
ATE0

OK
AT+CMEE=2

OK
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type AT.
ATI

LENA-R8001M10-00C-00

OK
AT+UGIND=1

OK
AT+UGPS?

+UGPS: 0,0,0

OK
ATI

LENA-R8001M10-00C-00

OK
AT+UGPS=1,15,295

OK
Starting continuous location.
Unable to start continuous location! [-4]Taking down GNSS...

+UUGIND: 0,295
AT+UGPS?

+UGPS: 1,0,0

OK
AT+UGPS=0

OK
U_CELL_PWR: powering off with AT command.
AT+UPSV=0

OK
AT+CPWROFF

+CPOF: MS OFF

OK

+UUGIND: 1,9

+UUGIND: 8,9

RobMeades, 2024-06-06:
Apologies, I had forgotten the conversation we had only a week ago in #241 where it was clear that you are using LENA-R8 and trying to address the M10 GNSS chip inside the LENA-R8 module.

As indicated in my [most recent post](https://github.com/u-blox/ubxlib/issues/248#issuecomment-2151617718) on #248, and as I think you have already concluded, for LENA-R8 it will be much simpler for you to access the internal GNSS chip via the dedicated/separate `tx_gnss`/`rx_gnss` pins provided on the module; while you _could_ access the internal GNSS chip via the cellular device you would get into the problems we saw in #241 with `UPOS` interference and continuous location would, unfortunately, never work: it relies on a stream of data from the GNSS device, it can never work with the polled `AT+UGUBX` mechanism, which is the only way that LENA-R8 allows the AT (i.e. cellular) interface access to the internal GNSS chip.  I will add a note to the [main_loc_gnss_cell_continuous.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_cell_continuous.c) example that it will not work for a LENA-R8xxxM10 module to avoid further confusion.

So, to use continuous location with a GNSS chip connected via the `tx_gnss`/`rx_gnss` pins you would connect-up another UART of your MCU to those pins and then follow the [main_loc_gnss_continuous.c](https://github.com/u-blox/ubxlib/blob/master/example/location/main_loc_gnss_continuous.c) example, i.e. the pattern for a dedicated GNSS device, not a GNSS chip inside a cellular module (i.e. we are ignoring the fact that it is inside the cellular module, it is just a GNSS device as far as `ubxlib` is concerned, the LENA-R8 HW is apparently designed to work that way).  The device configuration you specify would be for a GNSS device and, in your Zephyr device configuration for the GNSS device, you would tell `ubxlib` to use the UART HW block that is connected to the `tx_gnss`/`rx_gnss` pins, baud rate as appropriate for the M10 device (which I think is probably 9600, the default, anyway, but please check).

Apologies that this is so confusing, it is just a peculiarity of LENA-R8 I'm afraid.

RobMeades, 2024-06-06:
Documentation updated in commit addf595e277de3cd47ac5b239c7d5ced8bb29346.

mos216, 2024-06-08:
Hi Rob, thanks for explanation.

I tried to manipulate the gnss modem using tx_gnss/rx_gnss directly but no response. I see data on the tx_gnss of the modem but it doesn't response to the mcu host. it keeps sending positioning data after hot fix frequently seems like. 

U_PORT_BOARD_CFG: using GNSS device "cfg-device-gnss" from the device tree, module-type 2 with pin-enable-power 38 (0x26), pin-data-ready -1 (0xffffffff)...<CR><LF>
U_PORT_BOARD_CFG: ...GNSS on UART 1, uart-baud-rate 9600.<CR><LF>
U_GNSS: initialising with ENABLE_POWER pin 38 (0x26), set to 1 to power on GNSS, transport type UART.<CR><LF>
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.<CR><LF>
Opened device with return code -8.<CR><LF>
Unable to bring up the device!<CR><LF>

Do I need to enable something using AT command to get this one work. and is there a way to check which baudrate this device is working. I tried multiple baudrates but still no success.

mos216, 2024-06-09:
Hi Rob,

sorry is my bad. I didn't notice the default baud-rate in the integration manual. 

But do we need to remove this line (uNetworkInterfaceUp) from Gnss initialization ? bc now with a standalone gnss, Cell and Gnss have their initialization functions, so ubxlib will have two instances?

Another question, since in Zephyr we are using dts , how can we specify which device (GNSS or CELL) has to be opened when calling uDeviceOpen, since the first parameter is null.

RobMeades, 2024-06-09:
> But do we need to remove this line (uNetworkInterfaceUp) from Gnss initialization ? bc now with a standalone gnss, Cell and Gnss have their separate initialization functions, so ubxlib will have two instances, right?

Apologies, I did not explain well: what I meant was to remove the `uNetworkInterfaceUp()` line for GNSS where the _cellular_ handle is being passed in as `devHandle`.  You _should_ keep the `uNetworkInterfaceUp()` line for GNSS where the _GNSS_ handle is being passed in.

> Another question, since in Zephyr we are using dts , how can we specify which device (GNSS or CELL) has to be opened when calling uDeviceOpen, since the first parameter is null.

See fourth bullet point [here](https://github.com/u-blox/ubxlib/tree/master/port/platform/zephyr#specifying-device-and-network-configuration-through-the-device-tree): if you have both a cellular and a GNSS device in your device tree then you do need to have a `uDeviceCfg_t` structure that you pass to `uDeviceOpen()` but you only need to populate the `uDeviceType_t` member of that structure; with `U_DEVICE_TYPE_CELL` for the `uDeviceCfg_t` structure you pass when opening the cellular device and `U_DEVICE_TYPE_GNSS` for the `uDeviceCfg_t` structure you pass when opening the GNSS device.

Then the `ubxlib` code is able to tell which is which.  A bit of a waste I know but it should be a `const` structure, so only a small waste of flash.

mos216, 2024-06-11:
Hi Rob, thanks for the info and explanation. 
I built a sample where first uDeviceOpen for CELL  and after that initializing mqtt stuff and subscribing to a topic. In the same sample, I initialized GNSS as a separate modem, starting again uDeviceOpen for GNSS (in the same sample), and after that calling uLocationGetContinuousStart for permanent looking for location fix. so afterwards we have a one sample where CeLL is subscribing to mqtt topic and gnss is working continuously for getlocation. However, when a message is published to the topic where cell is subscribing to, a message is received successfully but for somehow hardfault error occurs and crash the whole application. but when I stop gnss from looking for location, then everything works fine. Actually debugging this issue is little bit complex in zephyr bc a lot of routins participating behind scene and I'm not so much families how zephyr stack is implemented in ubxlib but there is a relation or some shared resources when two devices with different types are initialized in same firmware. 
Could you guide me or give some hints for this issue? 

here the coredump data

[00:00:35.997,680] <err> os: ***** USAGE FAULT *****
[00:00:35.997,711] <err> os:   Illegal load of EXC_RETURN into PC
[00:00:35.997,741] <err> os: r0/a1:  0x00000002  r1/a2:  0x00050ddb  r2/a3:  0xfffffffe
[00:00:35.997,741] <err> os: r3/a4:  0x0099b000 r12/ip:  0x00054fb0 r14/lr:  0x00054fb1
[00:00:35.997,772] <err> os:  xpsr:  0x00027600
[00:00:35.997,772] <err> os: s[ 0]:  0x000277c8  s[ 1]:  0x81000000  s[ 2]:  0x00000002  s[ 3]:  0x0004bf2d
[00:00:35.997,802] <err> os: s[ 4]:  0x00000002  s[ 5]:  0x00052917  s[ 6]:  0x00000002  s[ 7]:  0x00050ddb
[00:00:35.997,802] <err> os: s[ 8]:  0x00000002  s[ 9]:  0x00050687  s[10]:  0x0000000a  s[11]:  0x00053d44

mos216, 2024-06-11:
Hi Rob, after debugging it, I figured out that the stack size of the GNSS thread was small in the main file. after increasing the stack size everything started to work fine so far

but thank you for the explanation about using .map file , it was helpful

RobMeades, 2024-06-11:
Excellent!

RobMeades, 2024-06-14:
I will close this one now since this part, at least, seems to be working for you.  Please feel free to re-open if that is not the case.

closed 2024-06-14 by RobMeades