u-blox/ubxlib #250: Compiling & building

danzadok, 2024-06-11:
Is there any instructions document or script how to build? 
I am trying to build for Windows. Do you have any MS Solution/Project file to share?
It seems that modules are using internal header files of other modules. For instance the internal header files in the folder port\platform\common\debug_utils\src are used by other modules. Is there any INCLUDE convention for compiling and building?
Where can I find the list of the compilation directives like U_CFG_DISABLE_TEST_AUTOMATION, that should be configured?

RobMeades, 2024-06-11:
Hi there, and thanks for posting.  We provide build instructions for the `runner` application which executes all of our tests and examples, the Windows version can be found [here](https://github.com/u-blox/ubxlib/blob/master/port/platform/windows/mcu/win32/README.md).  You will see that for Windows we provide CMake files that can be imported into VSCode.  Of course, ultimately you don't want to build our `runner` application, you want to build your own application, but this should show you how things are put things together and then you can [adapt](https://github.com/u-blox/ubxlib/blob/master/README.md#examples) that to your own needs.

> It seems that modules are using internal header files of other modules.

Our naming convention for header files can be found [here](https://github.com/u-blox/ubxlib/blob/master/DEVELOPMENT_PRINCIPLES.md#be-careful-what-you-expose), but you shouldn't need to worry about any of that; as far as the application is concerned just `#include <ubxlib.h>` and you will get all of the public header files, the ones in the `api` directories; an application should not need to refer to header files in a `src` (or `test`) directory.

> Where can I find the list of the compilation directives like U_CFG_DISABLE_TEST_AUTOMATION, that should be configured?

You should not need to define anything: that kind of stuff is only used internally when we are testing.  From an application perspective, with a very small number of exceptions ([geofencing](https://github.com/u-blox/ubxlib/blob/master/common/geofence/README.md) and [cellular PPP](https://github.com/u-blox/ubxlib/blob/master/cell/README.md#ppp-level-integration-with-a-platform)), we deliberately [avoid use of compilation flags](https://github.com/u-blox/ubxlib/blob/master/DEVELOPMENT_PRINCIPLES.md#avoid-compile-time-dependencies-within-ubxlib).  Now if you are building our examples then you will see that the examples files do include macros, things like `U_CFG_TEST_CELL_MODULE_TYPE` and `U_CFG_APP_PIN_CELL_TXD` etc.; these are documented in the `README.md` for each set of examples (e.g. see [here](https://github.com/u-blox/ubxlib/tree/master/example/sockets) for the sockets one), but this is just to make the examples easy to build, none of those values are built into `ubxlib` itself, in your application you can just put a number, `ubxlib` does not care.

RobMeades, 2024-06-25:
@danzadok: is it OK to close this now?

danzadok, 2024-06-26:
Yes , Thanks

closed 2024-06-26 by RobMeades