u-blox/ubxlib #252: Build for STM32 CUBE

danzadok, 2024-06-13:
I have found this repo https://github.com/u-blox/ubxlib_evk_applications, that is a good starting point for Windows and Linux using ubxlib.
But I have difficulty to build the ubxlib code for STM32 using the CUBE IDE.
There is a makefile named stm32f4.mk that should be used for building for such a mcu, but I failed to import it into the IDE.
Do you have instructions how to build for STM32?

Thanks

RobMeades, 2024-06-13:
> There is a `makefile` named `stm32f4.mk` that should be used for building for such a mcu, but I failed to import it into the IDE.
> Do you have instructions how to build for STM32?

We provide instructions for how to compile our `runner` build for ST here:

https://github.com/u-blox/ubxlib/tree/master/port/platform/stm32cube/mcu/stm32f4

For the ST Eclipse-based IDE, you need a `Makefile` (`stm32f4.mk` is included into a `Makefile`); you can import the `runner` `Makefile` as a "makefile project", then build and run it in Eclipse

Of course, as indicated in #250, ultimately you don't want to build our `runner` application, you want to build your own application, but this should show you how things are put together.

philwareublox, 2024-06-13:
Hi @danzadok ,

This is our Windows or Linux application example for UBXLIB. This is not set up to be simply configured for STM or any other embedded systems. 
This is only for Windows or Linux.

If you have found a STM.mk file, I should remove this :)

The content of the Common and Tasks folder, as well as the cellular tracker app folder should work on any system, but the main.c is built for the OS, Windows or Linux.

regards,
Phil.

RobMeades, 2024-06-25:
@danzadok: is it OK to close this now?

danzadok, 2024-06-26:
Yes , Thanks

closed 2024-06-26 by RobMeades