u-blox/ubxlib #257: Data Ready PIN

michaelboeding, 2024-06-23:
Is the data ready pin usable? Meaning if I hooked this up in hardware for a MAXM10S would it notify me via an interrupt when new GPS data is available?

```c++
  //the gps device 
  uDeviceCfg_t gDeviceCfg = {
      .deviceType = U_DEVICE_TYPE_GNSS,
      .deviceCfg = {
          .cfgGnss = {
              .moduleType = U_GNSS_MODULE_TYPE_M10,
              .pinEnablePower = U_CFG_APP_PIN_GNSS_ENABLE_POWER,
              .pinDataReady = -1,
          },
      },
      .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
      .transportCfg = {
          .cfgUart = {
              .uart = UART_NUM,
              .baudRate = BAUD_RATE,
              .pinTxd = TX_PIN,
              .pinRxd = RX_PIN,
              .pinCts = CTS_PIN,
              .pinRts = RTS_PIN,
              .pPrefix = NULL
          },
      },
  };`
        
       ```

RobMeades, 2024-06-24:
HI Michael: we added that field for forwards compatibility only, unfortunately there is currently no functionality behind it.  This is partly because I wasn't sure of the usage scenario: when `ubxlib` has asked the GNSS device for something it would likely just wait for an answer without monitoring a pin, if the `ubxlib` code is streaming stuff from the GNSS device likewise it would probably just be active all of the time.  I guess that the main usage of such a pin would be for an MCU that wants to power-save while waiting, which would be done outside of `ubxlib`, so we could, as you say, maybe provide a callback in interrupt-context when the GNSS device sets its Data Ready pin, and then the application code could do something.

Can you see an use-case where `ubxlib` _itself_ would take note of the pin?  I had been thinking of streaming as a kind of continuous high-speed thing but I guess there is no good reason for that, a stream could be intermittent, in which case we might have the UART/I2C/SPI receive gate itself on the GNSS device's Data Ready pin when doing streaming location, i.e. wait for a Data Ready interrupt, read from the GNSS device until empty and then return to waiting on a Data Ready pin interrupt.  Probably we could have the GNSS UART/I2C/SPI receive code wait [with a timeout] on a semaphore which is set in the interrupt, then whatever natural power-saving capability is configured for the RTOS would take over during the wait.

Would that make sense?  And then if this works I guess we might was well apply it even to the "`ubxlib` has asked for something and is waiting for a response" case, though I'd need to check if the GNSS device's Data Ready pin works generically in all cases rather than being specific to the availability of a location fix (it is not a pin I have so far looked at in any detail).

RobMeades, 2024-06-24:
Had a look into the code now and this seems quite do-able: when do you need it by?

michaelboeding, 2024-06-24:
Hey Rob, I think my use case is more of the initial gps fix and whenever I send an initial gps request. It's not urgent but would be nice to have. I can get by with my application using the current methods, but it would be nice to have in maybe a month? And I guess the MAXM10S doesn't have geofences built into the module firmware but if it did we could use that pin for various other Wakeup Sources like that.

RobMeades, 2024-06-25:
OK, I think it can be done transparently to the application, hope to get it done in a week or two, distractions permitting.  Will update this issue when I have something for you to try.

cturvey, 2024-07-23:
Does it support UART or just SPI and DDC/I2C?
Also remember this is a PIO# not a pin number on the module
You can free them by disabling interfaces, ie CFG-I2C-ENABLED=0 frees PIO# for SCL/SDA
Review UBX-MON-HW(3) to see current usage/assignments
Some of the product docs have pin to PIO# relationship documented, others do not.
```
0x10510003 CFG-I2C-ENABLED

0x10A20001 CFG-TXREADY-ENABLED          0      FALSE
0x10A20002 CFG-TXREADY-POLARITY         0      FALSE
0x20A20003 CFG-TXREADY-PIN             00          0
0x30A20004 CFG-TXREADY-THRESHOLD     0000          0
0x20A20005 CFG-TXREADY-INTERFACE       00          0
```

RobMeades, 2024-07-23:
Hi @cturvey: all three, and yes, we call the disables for interfaces that would otherwise conflict with the choice of pin.

Reason for asking what PIO @michaelboeding has chosen is that I can't test all of them so would like to at least eye-ball-confirm that what I have should work.  Thankfully the MAX-M10S data sheet does list the PIOs.

cturvey, 2024-07-23:
For UARTs one might consider the TIMEPULSE (1PPS), now there's some latency there, but you can control the pulse width such that the falling edge is an approximation. The UART also isn't flow controlled at any level, so the data just comes out, and you can catch the RXNE or FIFO related signals on your MCU

RobMeades, 2024-07-23:
> The UART also isn't flow controlled at any level, so the data just comes out, 

Indeed, it can be done in other ways for UART.

cturvey, 2024-07-23:
Yes, but sending query form commands for solutions can have significant latency, potentially over a second in particularly ill-timed interactions.

One might connect the TX pin to the MCU's GPIO/EXTI such that it wakes, and have a short, sacrificial message, that you can lose in the mean time whilst the MCU wakes and comes to it's senses. Some MCU have an LPUART that can be viable independently of the MCU viability.

RobMeades, 2024-07-23:
@michaelboeding: the preview branch is passing testing on ESP-IDF so I've pushed it out here in case you have chance to try it out: https://github.com/u-blox/ubxlib/tree/preview_feature_gnss_data_ready_rmea.

All you need to do is connect a pin of your MCU to a PIO of the GNSS device such as TIMEPULSE or EXTINT or one of the unused comms lines and then, in the structure you pass to `uDeviceOpen()` for the GNSS device, populate the `pinDataReady` field with the MCU pin number and the `devicePioDataReady` field with the PIO number of the GNSS device (see table 10 of the [MAX-M10S data sheet](https://content.u-blox.com/sites/default/files/MAX-M10S_DataSheet_UBX-20035208.pdf) for the PIO numbers).  With that done, while waiting for a message, the `ubxlib` code will stop on a semaphore until the GNSS device sends TX-Ready high, allowing the RTOS to sleep, or you to do other things, etc.

TX-Ready only works if you are connected to the GNSS device over SPI or I2C (that is a limitation of the GNSS device).

I will be away for all of August so any feedback this week is welcomed, otherwise take your time.

RobMeades, 2024-07-24:
@michaelboeding: turns out I've managed to complete testing on our other platforms also so this is now good to go from our end.  I have put it in for review, hope to get it merged before I leave on hols mid next week.  Feedback still welcomed until the end of this week probably.

michaelboeding, 2024-07-24:
Hey Rob - sorry for the delay I will test this weekend!

RobMeades, 2024-07-29:
@michaelboeding: since I have now received the relevant approvals and it has passed testing, the feature is now on `master ` here, see commit 2896142106fe596fb289748c68ba4afd8c1e717b.  Feedback still welcomed of course, though it would probably be best to test `master` rather than the preview branch, which I will delete when I return from holiday at the very end of August.

closed 2024-11-11 by manemannen