u-blox/ubxlib #258: A good practice proposol

danzadok, 2024-06-23:
Hello,
First, I'd like to apologize if my proposal sounds arrogant.  
I think it would be better if the 'Porting' code would be separated into two categories: OS & HW

RobMeades, 2024-06-23:
That would be possible but it wouldn't necessarily save a great deal: we have now removed nRF5-SDK so the only two platforms using FreeRTOS (i.e. where there is a duplication in the `ubxlib` code) are ESP-IDF and native STM32, and since ESP-IDF have branched FreeRTOS we have to build with their version anyway, though it might be that we don't use any of their modified code, I'd have to check.

Otherwise Windows, Linux and Zephyr are all distinct of course, and these days i gather that there is ESP-IDF support within Zephyr, as there is STM32 support, so likely in the future there may just be Windows, Linux and Zephyr.

RobMeades, 2024-07-05:
Anyway, we appreciate the suggestion but the change in the code tree this would cause, versus the gain, and considering that the underlying FreeRTOS code is different for each platform in any case, means we will likely leave things as they are.  Should another platform that uses FreeRTOS come along we will likely do as you suggest, though from what we can see everyone is moving in a Zephyr-wards direction.

closed 2024-07-05 by RobMeades