u-blox/ubxlib #259: Cellular Callback for Signal Changes

michaelboeding, 2024-06-24:
I think it would be useful to have some simple callbacks for cellular connections that respond to changes in signal strength, using straightforward categories like Good, Fair, Weak, and Poor. From what I understand, the type of connection (e.g., 2G, CAT-M1) affects how current signal strength categories are interpreted, with each transport type having different implications. I've tried to construct my own system, but I'm not an expert on the meanings of specific signal metrics such as RSRP dBm, RSRQ dB, RSSI dBm, RxQual, and SNR dB. It would be helpful to have notifications about changes in these signals, such as 'degraded signal due to X', etc.

RobMeades, 2024-06-25:
Hi Michael: always going to be a difficult topic this.  The module does not emit notifications of this nature, so something would have to poll for the information, and I'd probably suggest that the application does this.

Then the data has to be interpreted, and I think what you're saying is u-blox is probably best placed to provide a useful interpretation, which is probably true.  Maybe what we're talking about here is some kind of helper function, and an example to go with it?  The helper function would be called periodically, and maybe also on a given set of triggers (registration, connection, etc.) by the application and would have a [short] memory that would allow it to be able to judge "degraded" kind of outcomes.

We will discuss internally.

RobMeades, 2024-07-24:
@philwareublox: just a reminder that we said you might look into embodying the advice we document [somewhere] into code that might help with this problem.