u-blox/ubxlib #260: Enable debug log

jimakos96, 2024-06-26:
Hello 
I want somehow to enable the logs so i can see more satellite information like constellations .
Is there a setting to enable to make it happen ?

 Thanks for your attention.

RobMeades, 2024-06-26:
Hi there: there is no generic debug command for GNSS stuff, you would need to switch on the `UBX` messages you would like to see in the GNSS device and then read them.

We had an issue of this nature raised in https://github.com/u-blox/ubxlib/issues/205.  I'm not sure which GNSS device you're using but the discussion in that issue was about an M10 device (M9 and M10 devices have an entirely different configuration interface to the M8 devices).

In that particular issue, mabroens wanted to display all of the information received from the GNSS device in U-Centre so we added a Python script, described [here](https://github.com/u-blox/ubxlib/tree/master/gnss/api), which will allow you to do that also, i.e. if you enable and read the messages you would like using `ubxlib` you can leave your device running and execute the Python script in parallel; it will parse the `UBX` messages out of the `ubxlib` log stream and funnel them, live, into U-Centre.

Irrespective of whether you want that U-Centre bit, you still need to write the code to do the switching-on and reading of the GNSS messages.  The best thing to do is to follow the [dec_main.c](https://github.com/u-blox/ubxlib/blob/d4d58a55d5c7525009bde0be966db0ad4d29f692/example/gnss/dec_main.c#L267) example.  You will see that the example switches on only a single `UBX` message, `0x0107` (`UBX-NAV-PVT`).  You could, for instance, use `0x01FF` to switch on all `UBX-NAV` messages, or you could do `U_GNSS_CFG_SET_VAL_RAM()` (for an M9 or M10 device) or `uGnssCfgSetMsgRate()` (for an M8 device) and `uGnssMsgReceiveStart()` multiple times (you can use the same callback function) for a random set of messages of interest to you.  I would probably not do `0xFFFF`, which would switch on all `UBX` messages, as you would likely run out of bandwidth on the comms link to the GNSS device.

The [dec_main.c](https://github.com/u-blox/ubxlib/blob/d4d58a55d5c7525009bde0be966db0ad4d29f692/example/gnss/dec_main.c#L197) example calls `pUGnssDecAlloc()` on the messages it receives, which you probably don't need to do as `pUGnssDecAlloc()` only knows about a couple of `UBX` messages (`UBX-NAV-PVT` and `UBX-NAV-HPPOSLLH`), but you can read the messages and process them as you wish, or, as I say, you could funnel them into U-Center for interpretation using the Python script.

RobMeades, 2024-07-23:
Is there more to discuss on this or are you happy to close the issue?

RobMeades, 2024-07-30:
I will close this now: please feel free to re-open if there is more to discuss.

closed 2024-07-30 by RobMeades