u-blox/ubxlib #261: MQTT Clean Session Support for LARA-R6801D Module

codeph112, 2024-07-04:
When using the LARA-R6801D module, the MQTT clean session feature is not supported. As a result, messages published with QoS 1 are not received by clients upon reconnection if they were offline during the publish.

philwareublox, 2024-07-04:
Hi,

Unfortuantely the LARA-R6 module currently does not support setting the CleanSession option and will always clean the session. I don't know when there will be new firmware with this feature.

philwareublox, 2024-07-05:
What is your timescales for this project? Are you working with a distributor?

codeph112, 2024-07-08:
Dear @philwareublox,

We are encountering a critical issue with the LARA-R6801D module regarding the MQTT clean session feature. In our low-power project, the device sends a maximum of 24 frames per day, and the GSM modem remains off for the rest of the time. Many device settings, from firmware upgrades to peripheral control, are managed by notifications/control messages sent from the control panel. Without the ability to set the clean session to false, we cannot receive the last control messages when the GSM modem turns off and then back on.

It is crucial for our project to set the clean session to false to ensure we receive all pending messages upon reconnection. We are nearing the completion of our project, and this feature is mandatory for its success. As seen in the application note, the LARA-R6801D supports the clean session feature in MQTT Anywhere (AT+UMQTTSN=10,1 OK Set the MQTT-SN clean session). However, it appears that this feature is disabled for standard MQTT.

[AppNote: LARA-R6 Internet Applications Development Guide](https://content.u-blox.com/sites/default/files/documents/LARA-R6-Internet-applications-development-guide_AppNote_UBX-22001854.pdf)

This issue is very urgent as we are working on a government railway project with tight deadlines. We need this feature enabled within a week to meet our delivery schedule.

Please assist us in enabling the clean session feature for standard MQTT on the LARA-R6801D module as soon as possible.

philwareublox, 2024-07-08:
Hi @codeph112 

MQTT CleanSession has been requested from other customers too and we have it in our backlog, but at the moment we don't have a schedule for a new LARA-R6 firmware release. 

I understand your issue/requirement, but for now I can only suggest to adapt the application to send a "Hello" message to a topic which causes the cloud service to send down the message. 

What company are you purchasing the modules from? Could you request through them this feature, which would then create a support ticket with us, raising the requirement for this feature with us on our suport system. If you send me an email (phil.ware@u-blox.com) with your company/project and I will see if we have this already in our SalesForce system. I can then raise/push this issue/ internally today.

Regards,
Phil.

philwareublox, 2024-07-09:
Hi @codeph112 ,

We see your account in our system and are talking internally about this issue. 
Please contact your Field Application's Engineer for further information.

Regards,
Phil.

closed 2024-07-09 by codeph112