u-blox/ubxlib #262: Connecting NEO-6M to ESP32

jnschnkr, 2024-07-07:
I have a NEO-6M module and want to connect it with an ESP32S3 board. I managed to get the position with the nmea_parser example inside the ESP-IDF but wanted to use the ubxlib now. My code is mostly copied from the gnss example. I've tried everything I can find but I'm unable to make it work and couldn't find more information on the errors I get.

```
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "esp_system.h"
#include "driver/uart.h"
#include "ubxlib.h"

#define GNSS_RX_PIN 5
#define GNSS_TX_PIN 6

// Optionally, define a user context which you can use to pass
// information to a callback that does not have a dedicated
// void * callback parameter (this is the case with
// uGnssPosGetStreamedStart()).
typedef struct {
    char *pSomething;
    int32_t somethingElse;
} MyContext_t;

// Count of the number of position fixes received
static size_t gPositionCount = 0;

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_GNSS,
    .deviceCfg = {
        .cfgGnss = {
            .moduleType = U_GNSS_MODULE_TYPE_M9,
            .pinEnablePower = -1,
            .pinDataReady = -1, // Not used
            // There is an additional field here:
            // "i2cAddress", which we do NOT set,
            // we allow the compiler to set it to 0
            // and all will be fine. You may set the
            // field to the I2C address of your GNSS
            // device if you have modified the I2C
            // address of your GNSS device to something
            // other than the default value of 0x42,
            // for example:
            // .i2cAddress = 0x43
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
    .transportCfg = {
        .cfgUart = {
            .uart = UART_NUM_1,
            .baudRate = U_GNSS_UART_BAUD_RATE, /* Use 0 to try all possible baud rates
                                                  and find the correct one. */
            .pinTxd = GNSS_TX_PIN,  // Use -1 if on Zephyr or Linux or Windows
            .pinRxd = GNSS_RX_PIN,  // Use -1 if on Zephyr or Linux or Windows
            .pinCts = -1,  // Use -1 if on Zephyr
            .pinRts = -1,  // Use -1 if on Zephyr
            .pPrefix = NULL
        },
    },
};

static char latLongToBits(int32_t thingX1e7,
[... 141 lines cut ...]
Unable to open GNSS!
Done.
```

Configurations `U_GNSS_MODULE_TYPE_M9` and `U_GNSS_MODULE_TYPE_M10` allow me to open the device with returnCode 0 but I get returnCode -1025 when starting the stream. The full output is as follows:

```
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type UART.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3f 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 41 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
Opened device with return code 0.
Starting position stream.
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: got Nack for 0x06 0x8b.
U_GNSS: sent command b5 62 06 8a 10 00 01 03 00 00 01 00 21 30 e8 03 02 00 21 30 01 00 35 90.
U_GNSS: decoded UBX response 0x05 0x00: 06 8a [body 2 byte(s)].
Start stream with return code -1025.
Unable to start position stream!
Done.
```

Thanks for any inputs on how I could solve my problem.

cturvey, 2024-07-07:
The Generation 6 devices (2019) have been End-of-Life for many years, it's probably unrealistic to expect support now and moving forward.
UBXLIB might be overkill if you're just skimming NMEA/UBX for a GPS Only receiver.

RobMeades, 2024-07-08:
Hi, and thanks for posting.  I'm afraid Clive is correct: you'll see from [u_errror_common.h](https://github.com/u-blox/ubxlib/blob/master/common/error/api/u_error_common.h) that code -24 means [U_ERROR_COMMON_UNKNOWN_MODULE_TYPE](https://github.com/u-blox/ubxlib/blob/dcd6db3200e47f5d00dff40b6054fda55ecb0452/common/error/api/u_error_common.h#L79C5-L79C40); `ubxlib` support for GNSS modules begins with the M8 series, the M6 GNSS module doesn't support/understand most of the `UBX` messages that `ubxlib` will send.

Sorry!

jnschnkr, 2024-07-08:
Thanks for your help anyways. At least, I know now that it's not supported and that I don't need to try any longer.

closed 2024-07-08 by jnschnkr