u-blox/ubxlib #263: PlatformIO PackageException

maleroy, 2024-07-11:
Hello,

I've been developing an application with a ZED-F9P connected to an nRF52832. My platform of choice is PlatformIO (core is 6.1.5).

Development was going fine until I did a full clean instead of a clean of my project. Now when trying to integrate ubxlib, the console states the following:
```
 *  Executing task: platformio run 

Processing nrf52_dk (platform: nordicnrf52; framework: zephyr; board: nrf52_dk)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/u-blox/ubxlib @ 1.4.0
git version 2.34.1
Cloning into '/home/mleroy/.platformio/.cache/tmp/pkg-installing-dtq9d5z8'...
remote: Enumerating objects: 1240, done.
remote: Counting objects: 100% (1240/1240), done.
remote: Compressing objects: 100% (1039/1039), done.
remote: Total 1240 (delta 345), reused 670 (delta 149), pack-reused 0
Receiving objects: 100% (1240/1240), 6.32 MiB | 11.02 MiB/s, done.
Resolving deltas: 100% (345/345), done.
Submodule 'common/geofence/geographiclib' (https://github.com/geographiclib/geographiclib) registered for path 'common/geofence/geographiclib'
Submodule 'common/short_range/src/gen2/ucxclient' (https://github.com/u-blox/ucxclient.git) registered for path 'common/short_range/src/gen2/ucxclient'
Cloning into '/home/mleroy/.platformio/.cache/tmp/pkg-installing-dtq9d5z8/common/geofence/geographiclib'...
remote: Enumerating objects: 30501, done.        
remote: Counting objects: 100% (3634/3634), done.        
remote: Compressing objects: 100% (398/398), done.        
remote: Total 30501 (delta 3340), reused 3387 (delta 3236), pack-reused 26867        
Receiving objects: 100% (30501/30501), 13.79 MiB | 10.47 MiB/s, done.
Resolving deltas: 100% (24598/24598), done.
Cloning into '/home/mleroy/.platformio/.cache/tmp/pkg-installing-dtq9d5z8/common/short_range/src/gen2/ucxclient'...
remote: Enumerating objects: 412, done.        
remote: Counting objects: 100% (174/174), done.        
remote: Compressing objects: 100% (99/99), done.        
remote: Total 412 (delta 87), reused 111 (delta 72), pack-reused 238        
Receiving objects: 100% (412/412), 259.81 KiB | 3.25 MiB/s, done.
Resolving deltas: 100% (221/221), done.
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 19 (delta 10), reused 16 (delta 10), pack-reused 0
Unpacking objects: 100% (19/19), 144.32 KiB | 1.80 MiB/s, done.
From https://github.com/geographiclib/geographiclib
 * branch              b3bfff214c9639cbb9ffb1338dc19c2d10e5269e -> FETCH_HEAD
Submodule path 'common/geofence/geographiclib': checked out 'b3bfff214c9639cbb9ffb1338dc19c2d10e5269e'
Submodule path 'common/short_range/src/gen2/ucxclient': checked out '508768f0b566368149aa22502558e5481b853b2e'
PackageException: Package version 1.5.0+sha.0209dcf doesn't satisfy requirements 1.4.0 based on PackageMetadata <type=library name=ubxlib version=1.5.0+sha.0209dcf spec={'owner': None, 'id': None, 'name': 'ubxlib', 'requirements': '1.4.0', 'uri': 'git+https://github.com/u-blox/ubxlib'}

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

I've tried with ubxlib v1.3.1 but the same error occurs.

Was something broken?

Thanks in advance for your help!

maleroy, 2024-07-11:
It seems [0209dcf](https://github.com/u-blox/ubxlib/commit/0209dcf03ebe4d54a6b601f9f7a2741dbf56c785) is the perpetrator; forking the repo and putting `1.4.0` back in `library.json` allows the import to successfully happen.

RobMeades, 2024-07-11:
Hi, thanks for posting and sorry you're having trouble with this.  My guess, and it is only a guess, is that PlatformIO does some sort of version matching thing and, though I've updated the `library.json` it picks up to 1.5.0 I haven't published 1.5.0 to PlatformIO yet because I'm waiting for our release tag to go down.

However, there is no reason to wait actually, it is purely an administrative process, so let me do the publishing to PlatformIO now...

RobMeades, 2024-07-11:
Hokay, `ubxlib` 1.5.0 has now been published to PlatformIO:

[image]

Has that helped?

maleroy, 2024-07-15:
Hi Rob,

Thanks for the quick action and for your patience whilst waiting for my reply.

Unfortunately, it doesn't seem to fix it. Going back to trying to include u-blox/ubxlib @ 1.4.0 still raises a PackageException :/

RobMeades, 2024-07-15:
Hmph.  Maybe it needs the `ubxlib` 1.5.0 release tag to go down also?  Our product manager should be doing that any moment now (ping @hkro :-)).

hkro, 2024-07-16:
I just released 1.5.0. Sorry for the delay. I assume that solves it.

RobMeades, 2024-07-24:
@maleroy: did adding the 1.5.0 tag resolve your PackageException issue?

closed 2024-11-11 by manemannen