u-blox/ubxlib #264: Config baudrate for NEO-F10 GNSS module using uGnssCfgValSet()

Vanbk, 2024-07-19:
I am using the UART for communicating with the F10 module. However, the low default baudrate (38400) causes "lost of data". Thus, I am trying to set the higher baudrate using this function: 

returnCode = uGnssCfgValSet(device, U_GNSS_CFG_VAL_KEY_ITEM_UART1_BAUDRATE_U4, 115200, U_GNSS_CFG_VAL_TRANSACTION_NONE, U_GNSS_CFG_VAL_LAYER_RAM|U_GNSS_CFG_VAL_LAYER_FLASH); 

The return code is 0 but It seems calling the function is not effective. Could you please tell me if my method is wrong ? if my method is wrong, what procedure should I take to set the baudrate properly? Thank you so much!

RobMeades, 2024-07-19:
Hi, and thanks for transferring your query in here from the u-blox support forum.  Your code looks fine to me: could you run it again with the line `uPortLogOff()` commented out and paste in here the log output that `ubxlib` is sending?  Not sure which MCU you are on but the `ubxlib` logging output will come out of wherever `printf()` goes.

That will confirm the message exchange with the GNSS device.

Vanbk, 2024-07-19:
Thank you for your support. I found the problem. The second input parameter should be U_GNSS_CFG_VAL_KEY_ID_UART1_BAUDRATE_U4 instead of U_GNSS_CFG_VAL_KEY_ITEM_UART1_BAUDRATE_U4. the "_KEY_ID_" and "_KEY_ITEM_" mad me confused.

RobMeades, 2024-07-19:
Excellent!  My apologies for the confusing naming., a three character difference in the middle of a very long variable name was not a good idea; we live and learn :-).

cturvey, 2024-07-23:
I'm going to cross-link this here as I think it might be helpful to the broader conversation and spidering.
 https://portal.u-blox.com/s/question/0D5Oj00000LvXlkKAF/does-ubxlib-support-gnss-module-neof10n

It did present the dilemma that the NEO-F10N doesn't export the I2C pins of the F10 / UBX-F10150. I hadn't noticed that when I skimmed the docs/integration guide. USB not support in the IC, and presumable not SPI, as it uses the 4-pins that would normally be RX,TX,SCL,SDA

It is my expectation is that the I2C can be exported via the UART pins if the UART is disabled, and the I2C enabled and remapped.

Remap SCL/SDA on to TX/RX, respectively
```
0x10520005 CFG-UART1-ENABLED = 0
0x10510003 CFG-I2C-ENABLED = 1
0x10510004 CFG-I2C-REMAP = 1
```

RobMeades, 2024-07-30:
I think we are done with this one now, so am closing it, please feel free to re-open if there is more to discuss.