u-blox/ubxlib #265: SIM card detection

jamal-mouline, 2024-07-23:
Hello,

According to the AT manual,
gpio_mode for SIM card detection is 7

the ubxlib uCellGpioConfig() API offers only two modes: input or output (boolean parameter isOutput)

is there another API to configure for SIM card detection

or possibly extend uCellGpioConfig to support mode 7

Thank you

RobMeades, 2024-07-23:
Hi Jamal: does [uCellGpioConfigSpecialFunction()](https://github.com/u-blox/ubxlib/blob/a5f9fc741338e1afb1fd42615d195d4b55fa3521/cell/api/u_cell_gpio.h#L182) not work for you, with parameter [U_CELL_GPIO_SPECIAL_FUNCTION_SIM_CARD_DETECTION](https://github.com/u-blox/ubxlib/blob/a5f9fc741338e1afb1fd42615d195d4b55fa3521/cell/api/u_cell_gpio.h#L105).

jamal-mouline, 2024-07-24:
Hi Rob,
Thanks for the hint Re: uCellGpioConfigSpecialFunction

Indeed my initial grep on it came up empty
Had to run an update on ubxlib, and now I do see it

Will to incorporate it in my app next and see what happen...

Thank you

jamal-mouline, 2024-07-24:
Update:
Was able to integrate the new API
Will test it when the LEXI hardware becomes available
Thank you

Closing ticket...

closed 2024-07-24 by RobMeades