u-blox/ubxlib #268: Ubxlib / Configuration of the M10 Gnss receiver

jamal-mouline, 2024-07-31:
Hello Rob,
Question whether ubxlib has an interface to report status of the MIA M10 Gnss receiver, such as hardware status or I/O subsystem.

M10 interface Manual available here:
https://content.u-blox.com/sites/default/files/u-blox-M10-SPG-5.10_InterfaceDescription_UBX-21035062.pdf?hash=undefined

Specific interest in accessing the I/O pin status as described by command
3.14.4 UBX-MON-HW3 (0x0a 0x37) as documented on page 86 on the interface manual

Several configuration values are defined under u_gnss_cfg_val_key.h
https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_cfg_val_key.h
and this code 
https://github.com/u-blox/ubxlib/blob/master/example/gnss/cfg_val_main.c
provides examples for using seemingly relevant  uGnssCfgValSet and uGnssCfgValGet

But I am not able to find the keyId correlated to UBX-MON class?!?

Some guidance would be greatly appreciated

Thank you

jamal-mouline, 2024-07-31:
Wild guess:
Can I make use of this interface to query UBX-MON-HW3?

      length = uUbxProtocolEncode(**0x0a, 0x37**, NULL, 0, pBuffer);
      if (uGnssMsgSend(pObj->deviceHandle, pBuffer, length) == length)
      {
         messageId.type = U_GNSS_PROTOCOL_UBX;
         messageId.id.ubx = 0x0a37; // This could be any UBX message ID/class
         length = uGnssMsgReceive(pObj->deviceHandle, &messageId, &pBuffer, MY_MESSAGE_BUFFER_LENGTH, 30000, NULL);

cturvey, 2024-07-31:
Probably nothing at a PIO# or field level, the UBX-MON is more a a reporting/monitoring method.

Periodic output would be these:
```
0x20910354 CFG-MSGOUT-UBX_MON_HW3_I2C
0x20910355 CFG-MSGOUT-UBX_MON_HW3_UART1
0x20910358 CFG-MSGOUT-UBX_MON_HW3_SPI
```
Association of the PIO# tends to related to specific peripheral and functions, ie is the I2C enabled, which pair of pins are mapped. Which PIO# to use for TXREADY,  GEOFENCE or Antenna Management.

jamal-mouline, 2024-07-31:
Just want to make clear I understood the feedback
Using a very basic query example:
Say I want a command to send to the MIA 10 to query whether PIO5 is at a high or low level

Is it that there is no interface available to query the state of PIO5?

Please clarify...
Thank you

cturvey, 2024-07-31:
If UBX-MON-HW3 has the information, you should be able to send a query form of the command, ie a packet with a zero length payload. It doesn't involve a configuration key.
If you want a message with this data every couple of seconds, you can pace via the CFG-MSGOUT keys

There are other PIO related messages, but are not part of the public facing documentation.

Perhaps u-blox-M10 SPG 5.10 Interface description, UBX-21035061, NDA required, or FAE / INT level material.

jamal-mouline, 2024-08-01:
Thanks for the feedback
Could you please share a code snippet to send a query form?
It would be for one time poll, I don't need to make it periodic
Thank you

cturvey, 2024-08-01:
I'm not using ubxlib, I'd assume the byte code would look like this, and be sent like this
```
B5 62 0A 37 00 00 41 CD

uint8_t ubx_mon_hw3_query[] = { 0xB5,0x62,0x0A,0x37,0x00,0x00,0x41,0xCD };
uGnssMsgSend(pObj->deviceHandle, ubx_mon_hw3_query, sizeof(ubx_mon_hw3_query));
```

jamal-mouline, 2024-08-01:
Got it,
Looks like ubxlib has an encode API to generate ubx_mon_hw3_query
uUbxProtocolEncode

To retrieve the response content, I see
uGnssMsgReceive

Thanks,

closed 2024-11-11 by manemannen