u-blox/ubxlib #269: SARAR4 Unresponsive

michaelboeding, 2024-08-04:
I sometimes have my SARAR4 module become unresponsive, or almost delayed responses. Is there a command that I can use to check the status of the module and then reboot it ?

m-abubakar, 2024-08-06:
Hi @michaelboeding,
  Sorry you're experiencing this issue. To identify the problem and provide a workaround, could you please let me know which SARA-R4 variant you’re using?

michaelboeding, 2024-08-06:
SARAR422S-01B-00

m-abubakar, 2024-08-07:
Thank you, @michaelboeding, for providing the details. I’ll reach out to the relevant team to gather more information and address this issue.

m-abubakar, 2024-08-08:
Hi @michaelboeding,
   Sorry if we have tried this before, we could update the module to the SARA-R422S-01B-01 firmware to see if that resolves the issue. You can find the instructions for this process on the u-blox website under the "FW Update AppNote" and in "PCN (Product Change Note)" with firmware references.

Here are some useful links:
https://www.u-blox.com/en/product/sara-r4-series
https://content.u-blox.com/sites/default/files/documents/SARA-R42-01B-01_PCN_UBX-23006496.pdf
https://content.u-blox.com/sites/default/files/documents/SARA-R4-LEXI-R4-FW-Update_AppNote_UBX-17049154.pdf
https://content.u-blox.com/system/files/SARA-R422S-01B-01-IP-0124A0110-007K00.zip?hash=V2uzv8JMeGt3-iIp-bxLiqzqc-kwiAlijpTMl75Uyyc

michaelboeding, 2024-08-12:
I've been down that route before and it didn't fix the issue. Is there just a simple command I can send and if I don't get a response in X amount of time I can assume the module needs to be rebooted?

m-abubakar, 2024-08-12:
Hi @michaelboeding,

I’m sorry to hear that the previous approach didn’t resolve the issue. I understand how frustrating that can be. I have forwarded this issue to our concerned team and hope to resolve this issue soon.

Meanwhile, A simple command to check if a module is responsive might be to send a `AT` and check for `OK` response. Other similar commands are: AT+CIMI, AT+CSQ etc.

Let me know if this helps.

Thank you!

closed 2024-11-11 by manemannen