u-blox/ubxlib #270: In order to run the Arduino sample on platformio, ubxlib_test.h is included. Please distribute it.

Hokoshi, 2024-08-05:
The #include <ubxlib_test.h> in port\platform\platformio\arduino\app.iso is missing.

Could you please include it?

Hokoshi, 2024-08-05:
Please show me the contents of the "ubxlib_test.h" file.

m-abubakar, 2024-08-05:
Hi @Hokoshi 
Sorry about that, could you please try adding `u_runner.h` instead and see if this fixes the issue.

Thank you.

m-abubakar, 2024-08-06:
Hi @Hokoshi ,
  Following up on my previous comment, we have discontinued support for the pure Arduino platform([BREAKING CHANGE FOR ARDUINO PLATFORM: removal. (#1170)](https://github.com/u-blox/ubxlib/commit/72552304cdaf73ebe866d3702ab1b8df25a55e23)). However, you can still build ubxlib for Arduino using PlatformIO.
 
To build ubxlib with PlatformIO, add this line to your platformio.ini file
`lib_deps = https://github.com/u-blox/ubxlib`

For detailed instructions, please see the documentation [here](https://github.com/u-blox/ubxlib/tree/master/port/platform/platformio#introduction).
 
Below is a an example code build for Arduino for MQTT with a cellular device. 

```
#include <Arduino.h>
#include <ubxlib.h>
// put function declarations here:

// Bring in the application settings
#include "u_cfg_app_platform_specific.h"

/* ----------------------------------------------------------------
 * COMPILE-TIME MACROS
 * -------------------------------------------------------------- */

// MQTT broker URL: there is no port number on the end of this URL,
// and hence, conventionally, it does not include TLS security.  You
// may make a secure [D]TLS connection on broker.emqx.io instead
// by editing this code to add [D]TLS security (see below) and
// changing MY_BROKER_NAME to have ":8883" on the end.
#define MY_BROKER_NAME "ubxlib.com"

// Cellular configuration.
// Set U_CFG_TEST_CELL_MODULE_TYPE to your module type,
// chosen from the values in cell/api/u_cell_module_type.h
// just for an example we have selected the below one.
#define U_CFG_TEST_CELL_MODULE_TYPE U_CELL_MODULE_TYPE_LEXI_R422

// Note that the pin numbers are those of the MCU: if you
// are using an MCU inside a u-blox module the IO pin numbering
// for the module is likely different to that of the MCU: check
// the data sheet for the module to determine the mapping.

// DEVICE i.e. module/chip configuration: in this case a cellular
// module connected via UART

static const uDeviceCfg_t gDeviceCfg = {
    .deviceType = U_DEVICE_TYPE_CELL,
    .deviceCfg = {
        .cfgCell = {
            .moduleType = U_CFG_TEST_CELL_MODULE_TYPE,
            .pSimPinCode = NULL, /* SIM pin */
            .pinEnablePower = U_CFG_APP_PIN_CELL_ENABLE_POWER,
            .pinPwrOn = U_CFG_APP_PIN_CELL_PWR_ON,
            .pinVInt = U_CFG_APP_PIN_CELL_VINT,
            .pinDtrPowerSaving = U_CFG_APP_PIN_CELL_DTR
        },
    },
    .transportType = U_DEVICE_TRANSPORT_TYPE_UART,
    .transportCfg = {
        .cfgUart = {
            .uart = U_CFG_APP_CELL_UART,
            .baudRate = U_CELL_UART_BAUD_RATE,
            .pinTxd = U_CFG_APP_PIN_CELL_TXD,  // Use -1 if on Zephyr or Linux or Windows
            .pinRxd = U_CFG_APP_PIN_CELL_RXD,  // Use -1 if on Zephyr or Linux or Windows
            .pinCts = U_CFG_APP_PIN_CELL_CTS,  // Use -1 if on Zephyr
            .pinRts = U_CFG_APP_PIN_CELL_RTS,  // Use -1 if on Zephyr
            .pPrefix = NULL
        },
    },
};

// NETWORK configuration for cellular
[... 251 lines cut ...]
      uDeviceClose(devHandle, false);

  } else {
      uPortLog("Unable to bring up the device!\n");
  }

  // Tidy up
  uDeviceDeinit();
  uPortDeinit();

  uPortLog("Done.\n");

  // Stop the compiler warning about tlsSettings being unused
  (void) tlsSettings;

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

More can be found [here.](https://github.com/u-blox/ubxlib/tree/master/example)
 
In order to minimize the confusion, the `port\platform\platformio\arduino\app.ino` file is used in our internal automated tests and the `ubxlib_test.h` file is created on runtime. This is just an empty file to satisfy testing requirements. ([see here](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/common/automation/tasks/platformio.py#L128)).

Hokoshi, 2024-08-07:
* { font-size: 13px; font-family: 'MS Pゴシック', sans-serif;}p, ul, ol, blockquote { margin: 0;}a { color: #0064c8; text-decoration: none;}a:hover { color: #0057af; text-decoration: underline;}a:active { color: #004c98;}

Very thanks abubakar.

I could build your code.

However, I can't think of what my next step should be.

I'm trying to connect NORA-W3 to SONY's SPRESENSE, but SPRESENSE only has an Arduino environment, and ubxlib doesn't have an example of wifi Arduino. If Arduino support is going to disappear in the future, I might have to give up.

Hiroaki

m-abubakar, 2024-08-07:
Apologies for not being more clear of what I meant earlier. 

Support for the Arduino IDE has been phased out in favor of PlatformIO. This means that we’ll be using PlatformIO for all Arduino-related tasks, which, in my opinion, provides a better overall IDE experience.

For the Wi-Fi example, you might find that the [sockets](https://github.com/u-blox/ubxlib/tree/master/example/sockets) example could be helpful for your needs.

Information about the helpful examples ubxlib provides could be found on its [README.md](https://github.com/u-blox/ubxlib/tree/master#examples)

You can also go through the [README here](https://github.com/u-blox/ubxlib/tree/master/wifi) to learn more about what ubxlib provides with respect to wifi.

Thank you.

closed 2024-09-23 by m-abubakar

Hokoshi, 2024-09-23:
Ok Thanks.