u-blox/ubxlib #271: Help needed with integrating ubxlib into ESP-IDF with VS Code

nasihnazer, 2024-08-21:
I'm working on a project where I need to integrate the ubxlib library with the ESP-IDF framework using VS Code. I'm encountering some challenges and would appreciate any guidance or advice.

Project Details:
- Framework: ESP-IDF version 4.4.1
- IDE: VS Code
- Library: ubxlib (GNSS library from u-blox)
- Configuration Files: Located in the components directory of my ESP-IDF project.

Issues Faced
- Difficulty in configuring the ubxlib library with ESP-IDF.
- Issues with build configuration and linking.
- Encountering specific errors or unexpected behavior during the build process.

Steps Taken:
- Followed the integration instructions provided in the ubxlib documentation.
- Configured the ubxlib components directory as required.
- Ensured that all dependencies are correctly included in the project.

m-abubakar, 2024-08-26:
Hi @nasihnazer,

I’m sorry to hear you’re experiencing these issues. While ubxlib has been updated to a slightly newer version (ESP-IDF 5.2.2), it may still be compatible with the older version of ESP-IDF that you're using.

Could you please share the specific errors you’re encountering? If possible, any output from the build process or details about your configuration would be helpful.

nasihnazer, 2024-08-26:
im want to use ubxlib for gnss but im not able to integrate it with esp idf, when im intergrating it using the steps mentioned in this ubxlib/port/platform/esp-idf/mcu/esp32 at master · u-blox/ubxlib (github.com)<https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32>, but when im building it esp-idf shows that ubxlib is not take as component and cant be used, is there any step by step to integrate ubxlib in esp-idf to use gnss

nasihnazer, 2024-08-27:
Hi @m-abubakar,
I want to use ubxlib for gnss but I'm not able to integrate it with esp-idf, when I'm integrating it using the [steps ](https://github.com/u-blox/ubxlib/tree/master/port/platform/esp-idf/mcu/esp32)mentioned in documentation, but when I'm building it in  esp-idf shows that ubxlib is not take as component and can't be used. is there any step by step to integrate ubxlib in esp-idf to use gnss

m-abubakar, 2024-08-27:
Sorry to see that you have followed the steps as you mentioned [here](https://github.com/u-blox/ubxlib/issues/194#issuecomment-2306387153) and could not build successfully. Here I am just simplifying the steps for you for VS Code. 
 

- Create a new "Hello world" project in VSCode(ESP-IDF). 
- Replace the `CMakeLists.txt` file in your project with the one in the [port/platform/esp-idf/mcu/esp32/runner/CMakeLists.txt](https://github.com/u-blox/ubxlib/blob/master/port/platform/esp-idf/mcu/esp32/runner/CMakeLists.txt).
- Now open this `CMakeLists.txt` file and make the following changes ([I have attached a final sample file](https://github.com/user-attachments/files/16760114/CMakeLists.txt))
- Change the path to 'components' directory  as per your ubxlib source path 
-  Remove the ubxlib_runner entries.

```CMake
set(EXTRA_COMPONENT_DIRS "path-to-ubxlib/port/platform/esp-idf/mcu/esp32/components")
                         "${CMAKE_CURRENT_LIST_DIR}/main")

set(COMPONENTS "driver" "esptool_py" "unity" "ubxlib" "main" "heap" ${GEODESIC_COMPONENT})
```

- Remove the TEST_COMPONENTS line, the following one
```CMake
set(TEST_COMPONENTS "replace-me"  CACHE STRING "Component to test")
```

Now just need `#include "ubxlib.h"`  in your main and you can use any ubxlib API in your application.

- Build the project once to see there are no errors.

Let me know if you face any issues, Thank you.

nasihnazer, 2024-08-27:
Thank you @m-abubakar i will try it and let you know if i face any issues, i have doubt should i clone this [ubxlib](https://github.com/u-blox/ubxlib.git) to components of my project ?

m-abubakar, 2024-08-27:
Hi @nasihnazer,
  
It looks like a few dependencies might be missing. I am curious to know if you have downloaded the sub modules too using `git submodule update --init --recursive` as described [here](https://github.com/u-blox/ubxlib#the-hard-way-down-in-the-detail). This could help resolve the issue.

nasihnazer, 2024-08-27:
Hi @m-abubakar,
I'm able to build it without any issues, now im trying to use the [pos_main](https://github.com/u-blox/ubxlib/blob/master/example/gnss/pos_main.c) from gnss and this is the out I'm getting, should i change anything to use the example code

log:-
I (0) cpu_start: App cpu up.
I (248) cpu_start: Pro cpu start user code
I (248) cpu_start: cpu freq: 160000000
I (248) cpu_start: Application information:
I (252) cpu_start: Project name:     gpstest3
I (257) cpu_start: App version:      1
I (262) cpu_start: Compile time:     Aug 27 2024 16:57:37
I (268) cpu_start: ELF file SHA256:  ea03885eda1ed229...
I (274) cpu_start: ESP-IDF:          v4.4.1
I (279) heap_init: Initializing. RAM available for dynamic allocation:
I (286) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (292) heap_init: At 3FFB31A8 len 0002CE58 (179 KiB): DRAM
I (298) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (304) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (311) heap_init: At 4008D9B8 len 00012648 (73 KiB): IRAM
I (318) spi_flash: detected chip: generic
I (322) spi_flash: flash io: dio
I (327) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
U_GNSS: initialising with ENABLE_POWER pin not connected, transport type I2C.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
U_GNSS: sent command b5 62 06 04 04 00 00 00 09 00 17 76.
U_GNSS: sent command b5 62 0a 06 00 00 10 3a.
U_GNSS: decoded UBX response 0x0a 0x06: 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 [body 120 byte(s)].
Opened device with return code 0.
Starting position stream.
U_GNSS: sent command b5 62 06 8b 08 00 00 00 00 00 ff ff 21 00 b8 cc.
U_GNSS: decoded UBX response 0x06 0x8b: 01 00 00 00 03 00 21 20 01 01 00 21 30 e8 03 02 00 21 30 01 00 [body 21 byte(s)].
U_GNSS: sent command b5 62 06 01 02 00 01 07 11 3a.
U_GNSS: decoded UBX response 0x06 0x01: 01 07 00 00 00 00 00 00 [body 8 byte(s)].
U_GNSS: sent command b5 62 06 01 08 00 01 07 01 00 00 00 00 00 18 e2.
U_GNSS: decoded UBX response 0x05 0x00: 06 01 [body 2 byte(s)].
Unable to start position stream!
Done.

closed 2024-11-11 by manemannen