u-blox/ubxlib #272: Support for pass through AT commands with ubxlib

jamal-mouline, 2024-08-22:
Hi,
Question about pass through AT command within ubxlib

Once the modem is powered up, does ubxlib offer a method to support pass through commands?

This is useful in lab certification environment.

Thank you
Jamal

m-abubakar, 2024-08-26:
Hi @jamal-mouline,

Thank you for your intriguing query. Could you please provide a bit more detail about what you're trying to accomplish? From what I understand, you're looking for a method to send and receive AT commands directly, bypassing the AT Client in ubxlib. It seems you’re aiming to forward text commands directly and receive the output as text, possibly via a callback?

Thank you.

jamal-mouline, 2024-09-05:
Hi @m-abubakar 
Let's hold on this issue a bit as we may have a hardware design with direct access to the modem AT interface through USB, bypassing the host controller completely
To your question, we're trying to support 3GPP command for lab pre-certifications
Thank you

closed 2024-09-23 by m-abubakar