u-blox/ubxlib #274: Sending Chunks Instead of Large Buffer in uGnssMgaResponseSend()

nautalert, 2024-08-26:
Trying to limit memory usage since the buffer required for offline data can be rather large. Is there a way to send it in smaller chunks instead of one large buffer?

mazgch, 2024-08-26:
You could split up the MGA buffer by parsing out the UBX messages and then send each message separately. You will find tens if messages depending on the request Assist Now Online request.

If you just chunk the buffer up, make sure not to interleave other messages between multiple chunks.

RobMeades, 2024-08-28:
Hi, and thanks for posting.  The MGA data is just a literal stream of discrete UBX messages that the GNSS chip understands, so you can parse the MGA data to find the messages in it and then send each message down one at a time, or whatever you wish, to fit with your buffer limitations, just make sure that you send whole UBX messages.

See the function `uUbxProtocolDecode()` for how to do this:

https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/ubx_protocol/api/u_ubx_protocol.h#L154-L226

RobMeades, 2024-09-02:
@nautalert: does the above help you at all?

nautalert, 2024-09-02:
So just sending the commands via uGnssMsgSend() is all we need to do?

RobMeades, 2024-09-02:
Yes, once you have chopped-up the data downloaded from the MGA server into discrete (whole) UBX messages, send them all down (in order), one after the other, and the GNSS chip will act upon them.  From memory each message is relatively short (maybe 50 or 100 bytes), which is a lot better than many kbytes,

nautalert, 2024-09-02:
Yes, 84 bytes to be exact. :) Thanks for the info. Is this what is happening on the backend? Because this method seems to take longer.

RobMeades, 2024-09-02:
Yes, kind of.  [uGnssMsgSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_msg.h#L212) will always check for an acknowledgment from the GNSS chip, each time.

[uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L562) includes a [flowControl](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L204) parameter which allows you to chose how much acking you need from the GNSS chip:

- `U_GNSS_MGA_FLOW_CONTROL_SIMPLE` is the same as requiring an ack for each message, just like using [uGnssMsgSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_msg.h#L212),
- `U_GNSS_MGA_FLOW_CONTROL_SMART` sends only as many UBX messages as are known to fit within the RAM buffer of our GNSS chips _without_ asking for any acks and only then start waiting for acks,
- `U_GNSS_MGA_FLOW_CONTROL_WAIT` waits for no acks from the GNSS chip at all and just adds a [U_GNSS_MGA_INTER_MESSAGE_DELAY_MS](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L111) (10 ms) delay between each one.

The comment above [uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L562) warns that you must send a complete MGA response to the GNSS chip but that is only because the GNSS chip does require complete UBX messages, so you can't just randomly chop the MGA response up.  If you have parsed the MGA response into discrete (whole) UBX messages, you could use [uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L562) to send down [a set of] those UBX messages with, say `U_GNSS_MGA_FLOW_CONTROL_WAIT`, if you think that would be faster than waiting for an ack for each one.

RobMeades, 2024-09-02:
Sorry, no, I'm talking rubbish, [uGnssMsgSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_msg.h#L212) does NOT wait for an ack at all, since it has no idea what you are sending (UBX message or NMEA message or whatever), so it should be like using [uGnssMgaResponseSend()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/gnss/api/u_gnss_mga.h#L562) without even the 10 ms delay.

closed 2024-09-06 by RobMeades