u-blox/ubxlib #276: stm32u5 CubeIDE + CubeMX project integration issue of ubxlib without test runner

frastrello, 2024-08-29:
Hi,

I'm new with the `ubxlib` library and I know but I'm not expert with FreeRTOS, but my aim is to integrate `ubxlib` in my project without using the test runner and makefile project (considering all the warning I found in the library).

I have a NUCLEO-U5A5ZJ-Q (not the NUCLEO-U575ZI-Q) board connected to a SARA-R520M10 EVK with USART2, using FreeRTOS in STM32CubIDE and an empty project with CubeMX.

So, I proceeded with these steps (I try to put them in chronologic order):
1 - I added Free RTOS with CubeMX configurator to my empty project for a NUCLEO-U5A5ZJ-Q with 1 task: `MyTask()`
2 - I copied the library in my project folder
3 - I selected one example from the example folder: `lte_cfg`.
4 - I added this `#define U_CFG_TEST_CELL_MODULE_TYPE U_CELL_MODULE_TYPE_SARA_R52` in one configuration file
5 - **Important**: I copied the include, variables and the main code in `app_freertos.c`: `uPortInit()` and `uDeviceInit()` in my `MX_FREERTOS_Init()` because I checked inside the `osThreadNew()` function... and the rest of code in `MyTask()`. **Q1: Am I using correctly the example code, `uPortInit()` and `uDeviceInit()`? Or should I modify them, because the error I get is linked to the fact the osKernel is still not launched?**
6 - I deleted the example folder, test folders in each directory and all the platform except the [port/platform/stm32cube/src](https://github.com/u-blox/ubxlib/tree/master/port/platform/stm32cube/src)
7 - I deleted and used the files in [port/platform/stm32cube/src](https://github.com/u-blox/ubxlib/tree/master/port/platform/stm32cube/src) according to the internal `README.md` file for my platform
8 - I added all the peripheral with CubeMX requested by all the library submodules (i.e. I2C, SPI, LPUART e USART) , configuring correct HAL or LL driver.
9 - I added all necessary folders to the include paths and all the symbols to my project to compile for a STM32U5 with FreeRTOS (i.e. `U_PORT_STM32_CMSIS_ON_FREERTOS`, `CMSIS_V2`, `U_PORT_STM32_PURE_CMSIS`, `USE_HAL_DRIVER`, `USE_FULL_LL_DRIVER`, `U_CFG_DISABLE_TEST_AUTOMATION`)
10 - I aligned my `FreeRTOSConfig.h` to what I found in [port/platform/stm32cube/mcu/stm32u5/cfg](https://github.com/u-blox/ubxlib/tree/preview_platform_stm32u5_rmea/port/platform/stm32cube/mcu/stm32u5/cfg) using CubeMX configurator and manually adding the rest, then I used the `u_cfg_app_platform_specific.h`, `u_cfg_hw_platform_specific.h` and `u_cfg_os_platform_specific.h` defines aligning the pinout used at least for the USART port and other OS data
11 - I used `heap_useNewlib.c` and excluded `heap1.c` from building
12 - I commented some multiple definition like `_sbrk `implementation in `sysmem.c` and other automatic handler definitions in s`tm32u5xx_it.c`
13 - **Important**: In order to compile I changed symbol S`TM32U575xx` in `STM32U5A5xx`, defined in my project and changed `#include "stm32u575xx.h" `in `#include "stm32u5a5xx.h"`, because my different target. **Q2: I checked my reasources should be ok, but could it be an issue?**

In this way I successfully compile, but during the execution I get this:
[image]

**Q3: can you help me accomplish my goal?**

Thanks and best regards,
Fabio

m-abubakar, 2024-09-03:
Hi Fabio!
Updates are continuously pushed to ubxlib and it is recommended to use the makefile and integrate it into your STM32CubeIDE as adding the files and configuring the project manually may cause difficulties for you to handle it in future.

Once the project is configured properly no warning should be there.  The route you described is a little bit different and we are trying your steps to catch your problem somehow.

frastrello, 2024-09-06:
Hi @m-abubakar , I was integrating the library in an already configured STM32 project, thanks a lot for the feedback.
I found an AT Command library and think I will proceed with it and without ubxlib.
Thanks and best regards.
Fabio

closed 2024-10-10 by m-abubakar