u-blox/ubxlib #278: Connectivity to a SARA-R520M10 from an external MCU nRF52840 DK

jamal-mouline, 2024-09-05:
Hello @RobMeades 

Appreciate support over this issue.
Been working with @Steve.Brown.
He recommended to reach out to you.

We have a uBlox devKit (base board and ADP with SARA-R520M10 mounted on it)
We are trying to connect to it via AT commands over UART1 from an external MCU nRF52840 DK

The base board is powered with a 9V power supply, the switch in on
The shunt on the UART switch on the ADP board has been removed per instructions from Steve
The power to the modem is on (verified by issuing AT commands through USB port J201)

Obtained wiring instructions from this User Guide document, adn verified with Steve
https://content.u-blox.com/sites/default/files/documents/EVK-SARA-R52_UserGuide_UBX-23006467.pdf

Section 5.5 on page 18 describes the pin out for the board to board connectors J103 and J104

RTS                        J104 pin 14
CTS                        J104 pin 13
TXD                        J104 pin 16
RXD                       J104 pin 15
Plus
GND                      J104 pin 10
PWR_ON            J104 pin 30
RESET_N             J103 pin 25
V_INT                 J104 pin 35

On the host controller side, using Zephyr 3.5.99
we have the UART1 TX and RX configured through the device tree as P1.1 and P1.2 respectively
The PWR_ON, RESET, and V_INT are configured on P1.6, P1,7, and P1.8 through the CMakeList.txt as follows
# Compile definitions needed for UBXLIB.
target_compile_definitions(app PRIVATE
   U_CFG_APP_CELL_UART=1
   U_CFG_CELL_MODULE_TYPE=1
   U_CFG_APP_PIN_CELL_PWR_ON=38                            ==> should correspond to P1.6
   U_CFG_CELL_DISABLE_UART_POWER_SAVING=0
   U_CFG_APP_PIN_CELL_RESET=39                                 ==> P1.7
   U_CFG_APP_PIN_CELL_VINT=40                                   ==> P1.8
   )

Additional wire connecting the GND on both ends.

Troubleshooting the power up sequence, we hit an error with u_cell.c / addDevice() when trying to set the PWR_ON pin LOW
uPortGpioSet returns -5
uPortLog copy / paste below
Note the call to uCellPwrOn comes a bit later

uart:~$ modem_ctrl power_up
Cell Modem Power Up
uart:~$ **U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin 40 (0x28) (and is 1 when module is on).
U_CELL: uPortGpioSet() for PWR_ON pin 38 (0x26) returned error code -5.**

QUESTIONS:
1-Any missing hardware configuration on either sides?
2-Through the ADP J201, I read AT+USIO? as 2.2,
Does that need to change?
3-Could there be a port configuration issue with this facility pUPortPrivateGetGpioDevice? (it returns -5 when error)
4-Any other hints appreciated...

PLEASE NOTE:
Previously we used the same Application code / device tree configuration to interface the same exact controller host to a Sparkfun board with the SARA-R410M mounted. The only key difference is that the Sparkfun comes with a compatible Arduino interface that plugs in directly onto the nRF52840 devKit

On that hardware configuration, the call to ubxlib uDeviceOpen() completes error free
--log copy/paste below...

uart:~$ system_ctrl version
Build Revisions:
Zephyr  Kernel: 3 3 99
Build Revision: V00.01.08 (DEBUG)
Build DateTime: Sep  5 2024   14:13:40

uart:~$
uart:~$ modem_ctrl  power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT
uart:~$ AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT
AT
AT
AT

OK

OK
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

L0.0.00.00.05.12,A.02.21

OK
AT&C1

OK
AT&D0

OK
AT+UCGED=5

OK
AT&K0

OK
AT+UPSV=0

OK
AT+CPSMS?

+CME ERROR: SIM failure
AT+CEDRXS?

+CME ERROR: SIM failure
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+CGSN

350857124811117

OK
U_CELL_INFO: IMEI is 350857124811117.
HANLDE 1 ea7beef
[00:01:02.433,746] <inf> CellModem: Modem power up and ready to be configured

[00:01:02.433,776] <inf> CellModem: MODEM_POWERING_UP->MODEM_POWERED_UP
[00:01:02.433,807] <inf> CellModem: uartHandle 1
[00:01:02.583,923] <inf> CellModem: subTopic messages/350857124811117/devicebound/

[00:01:02.583,984] <inf> CellModem: httpPath /350857124811117.html

[00:01:02.584,075] <inf> CellModem: Callback modem message 2 MODEM_MSG_POWER_UP_ACK
[00:01:02.584,136] <inf> Device: k_msgq_get event type 6 state DEVICE_INITIALIZATION
[00:01:02.584,167] <inf> Device: Modem Power Up Complete

RobMeades, 2024-09-06:
HI Jamal.

For [uPortGpioSet()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_gpio.c#L227) on the Zephyr platform to return error code -5 ([U_ERROR_COMMON_INVALID_PARAMETER](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/error/api/u_error_common.h#L60)), as you have seen, [pUPortPrivateGetGpioDevice(pin)](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L478) must have returned `NULL`.  For the [nrf52840dk board](https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/nordic/nrf52840dk), as defined by Zephyr, the possible GPIO ports are `gpio0` and `gpio1`.  Both of these have a default status of `okay` in the `nrf52840dk_nrf52840.dts` file so, unless they have been marked as not OK in an `.overlay` file, that should not be a problem.

[pUPortPrivateGetGpioDevice(pin)](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L478) also calls [uPortPrivateGetGpioPortMaxPins()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L678), to get the value of "32" back in this case, which I'd have thought is unlikely to go wrong.  Of course we test nRF52840 operation, and in fact we happen to use pin 38 in our testing, so there can't be anything fundamental wrong here.

If I were you I'd put some debug prints into the functions above to confirm that [uPortPrivateGetGpioPortMaxPins()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L678) is returning 32, that [pUPortPrivateGetGpioDevice(pin)](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L478) is receiving the value 38 and where in the cascade of `if()`s it is ending up.  Specifically, for the case of pin 38, we are expecting it to end up here:

[image]

`pDev` should be a valid (non-`NULL`) pointer after line 593 for the case where `pin` is 38.

jamal-mouline, 2024-09-06:
Hello Rob,
Appreciate the feedback

I have added several calls to uPortLog inside uPortGpioSet, but I could not get them to print anything. Weird!

Stepping through with the debugger instead, I have verified that
uPortSetGpio is called with 38 and 1

but that  uPortPrivateGetGpioPortMaxPins() is returning -1 <<< PROBLEM

so inside pUPortPrivateGetGpioDevice(), portNo is getting the value -38

therefore pDev remains at NULL

PS
I have also verified in the .dts file that
&gpio1 {
	status = "okay";
	gpio-line-names = "", "D0", "D1", "D2", "D3", "D4", "D5", "D6",
		"D7", "", "D8", "D9", "D10", "D11", "D12", "D13";
};

and that is not altered in the .overlay

PS2
One key difference I remember now between the Sparfun/SARA-R410 and the SARA-R520 projects
for the former I was using nRF Connect 2.4.2
for the latter I am using 2.6.1
No sure if that matters at all, but thought I'd mention it

PS3
I made the change to U_CFG_CELL_MODULE_TYPE at 9 instead of 1. Good catch, thank you.
# Compile definitions needed for UBXLIB.
target_compile_definitions(app PRIVATE
   U_CFG_APP_CELL_UART=1
   U_CFG_CELL_MODULE_TYPE=9
   U_CFG_APP_PIN_CELL_PWR_ON=38
   U_CFG_CELL_DISABLE_UART_POWER_SAVING=0
   U_CFG_APP_PIN_CELL_RESET=39
   U_CFG_APP_PIN_CELL_VINT=40
   )

RobMeades, 2024-09-06:
> I have added several calls to uPortLog inside uPortGpioSet, but I could not get them to print anything. Weird!

Ah, yes, see advice [here](https://github.com/u-blox/ubxlib/blob/master/DEBUG_HINTS.md#adding-your-own-debug-prints), should have mentioned that, sorry.

> but that uPortPrivateGetGpioPortMaxPins() is returning -1 <<< PROBLEM

Oh!  Then we're hitting this case:

https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L712-L719

We test with nRFConnect version 2.5.0, you are using nRFConnect version 2.6.1, but the [gpio_driver_config](https://github.com/zephyrproject-rtos/zephyr/blob/34982bb00b57486aeb83a6cb3bb7433791302aed/include/zephyr/drivers/gpio.h#L686-L697) structure in Zephyr, which is what matters here, remains unchanged:

[image]

If you are able to run under a debugger, all I can suggest is that you single step very carefully through [uPortPrivateGetGpioPortMaxPins()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/port/platform/zephyr/src/u_port_private.c#L678C9-L678C41), the first time it is called, to see how it is managing to not get  value of 32 for the number of pins per port.

jamal-mouline, 2024-09-06:
Oh! Then we're hitting this case:
 if (gGpioNumPinsPerPort == 0) { 
     gGpioNumPinsPerPort = -1; 
 } 

Yes, definitely the case, the debugger stops at line 718

The comment about then the Zephyr platform is broken),
is worrisome

From the debugger, it's hard to single step through that function uPortPrivateGetGpioPortMaxPins() as if some of that code is optimized out

I am going to try to Manage west workspace back to 2.5.0 to match your setting,
and re-test...

Not sure how much impact it will have given your comment about [gpio_driver_config] being unchanged, but maybe worth a try...

jamal-mouline, 2024-09-06:
@RobMeades 

That was my best guess as well, gGpioNumPinsPerPort  comes out at -1 from this loop inside uPortPrivateGetGpioPortMaxPins()

        for (int32_t x = 0; (x < 64) && (gGpioNumPinsPerPort < 0); x++) {
            if ((pCfg->port_pin_mask & (1UL << x)) == 0) {
                gGpioNumPinsPerPort = x;
            }
        }

**Now an interesting update:**

It took forever for West Update to downgrade to 2.5.0 (thinking next time, I would just delete the entire Zephyr folder first)

[image]

But the issue with uPortGpioSet() / PWR_ON line has self resolved!

The power up sequence has moved ahead inside uCellPwrPrivateOn()
It is now stumbling into getting a response for AT command --log below

uart:~$ modem_ctrl power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin 40 (0x28) (and is 1 when module is on).
U_CELL_PWR: powering on.
AT
AT
AT
[00:06:58.180,053] <inf> CellModem: Opening a modem device instance...
uart:~$ AT
AT
AT
...
...

AT
AT
AT
U_CELL_PWR: powering on, module is alive.
ATE0
ATE0
ATE0
[00:08:12.090,240] <err> CellModem: Failed to initiate the modem: -257

jamal-mouline, 2024-09-06:
I have already tried both combination of RX TX lines

on the uBlox devKit
TXD J104 pin 16
RXD J104 pin 15

on the Nordic DK / host controller
UART1 TX and RX P1.1 (Arduino D0) and P1.2 (Arduino D1) respectively

Wondering now if I am missing a key configuration on the uBlox devKit?!?

Interestingly, I can communicate with the uBlox devKit through the USB port on the main board
--but not through the USB port J201 that is on the ADP daughter board!?

[14:09:09:171] at
[14:09:09:415] OK
[14:09:09:811] at
[14:09:09:977] OK
[14:09:10:491] ati
[14:09:11:252] SARA-R520M10-02B-00
[14:09:11:252] 
[14:09:11:252] OK
[14:09:19:156] at+CGSN
[14:09:21:412] 350428920015806
[14:09:21:413] 
[14:09:21:413] OK
[14:09:25:002] at+CGMR
[14:09:26:916] 05.09
[14:09:26:916] 
[14:09:26:916] OK
[14:09:48:420] at+CMEE?
[14:09:52:915] +CMEE: 0
[14:09:52:915] 
[14:09:52:915] OK
[14:10:00:116] ati9
[14:10:01:299] 05.09,A00.01
[14:10:01:305] 
[14:10:01:305] OK
[14:10:12:307] at+ucged?
[14:10:15:764] +UCGED: 2
[14:10:15:764] 6,2,310,260
[14:10:15:783] 750,2,50,50,7bde,105950d,96,00000000,ffff,ff,51,1
8,14.87,1,255,255,50,255,255,0,255,255,0,0
[14:10:15:784] 
[14:10:15:785] OK
[14:10:26:073] at+upsv?
[14:10:28:740] +UPSV: 0
[14:10:28:740] 
[14:10:28:741] OK
[14:10:39:054] at+CGSN
[14:10:41:475] 350428920015806
[14:10:41:491] 
[14:10:41:492] OK
[14:13:03:986] at+usio?
[14:13:06:600] +USIO: 2,*2
[14:13:06:620] 
[14:13:06:621] OK

RobMeades, 2024-09-06:
> But the issue with uPortGpioSet() / PWR_ON line has self resolved!

Interesting, we should try updating to nRFConnect 2.6.x ourselves to find out what it is doing.

> It is now stumbling into getting a response for AT command

OK, so now you need to monitor HW to find out what is going on as something, somewhere is not connected quite right.  Usual things to check:

- the MCU UART Tx line must be connected to the TXD line of the EVK, not the RXD line, this is the [DCE convention](https://www.maxlinear.com/Files/Documents/sipex/Literature/DTE_or_DCE.pdf)), similarly MCU UART Rx line to the RXD line of the EVK,
- the EVK is 1V8 logic and the nRF52840 board is 3V3 logic so you must have a level shifter in the way (e.g. https://www.sparkfun.com/products/12009) for both the UART RX and UART TX lines (you can power the 1V8 side from the VINT output pin on the ADP),
- these bi-directional level shifters will NOT work with UART flow control lines, hence you can only work _without_ UART HW flow control; you will need to ground the RTS and DTR lines on the cellular side for it to talk,
- note that *no* level shifter is required for the `PWR_ON` connection as that is just pull-down and will just float up to whatever level it likes,
- `ubxlib` has pulled the `PWR_ON` pin low, check if the `VINT` output of the module goes high; this would indicate that the module is, indeed, on, but the UART comms are what is failing,
- you need to have the DIP switches in the right position to use the ADP UART lines instead of the USB port on the base-board, like this:
[image]
- the jumper on the ADP of the SARA-R52 board needs to be disconnected (i.e. not on the UART _or_ on the USB connection, just disconnected).

jamal-mouline, 2024-09-06:
OK, a lot of good items to check for, but I believe the key one I am missing is the level shifter 3V3 to 1V8
Plus ground DTS and DTR

Otherwise the jumper on the ADP is disconnected

regarding the DIP switches,
I have them in the ON position, with the exception of DB9, this is in the OFF position, with the associated green light on.
Does that match what is on the picture

Thank you

RobMeades, 2024-09-06:
The DIP switch positions we always use (I can't recall now which one is significant and which one is not) is that the DIP switch nearest the main power toggle switch of the entire EVK (let's call this DIP switch 1) has its slider closest to the main power toggle switch (let's call this "up").  The positions of all of the switches are then:

1 Up
2 Down
3 Up
4 Down
5 Down
6 Down

jamal-mouline, 2024-09-06:
@RobMeades Thanks so much for detailed feedback

I take back my comments about the need of level shifters!!

I flipped the DIP switches as recommended above
and more importantly,
connected both RTS and DTR to GND on the ADP J104 connector as recommended in earlier note

Now I can complete the power up and power down sequence successfully (invoking uDeviceOpen and uDeviceClose respectively)

In the power up state, I can measure
the PWR_ON line at 3.4V on J104 / 30
and the V_INT line at 1.8V on J104 / 35
(by the way I had this V_INT line wrongly connected to pin 33 on J104, but it did not seem to matter?!?)

Next week, I shall attempt cellular registration, MQTT, GNSS fix, etc...
but I should have most of the application code ready to validate based on past experiments with the SparkFun and OpenCpu configurations

**In between, I will be very curious about your Team findings with nRF Connect 2.6.x**

Thank you

PS
Below is Host Controller debug console log...

uart:~$ modem_ctrl  power_up
Cell Modem Power Up
uart:~$ U_CELL: initialising with enable power pin not connected, PWR_ON pin 38 (0x26) (and is toggled from 1 to 0) and VInt pin not connected.
AT
AT
AT
AT
[00:02:35.160,186] <inf> CellModem: Opening a modem device instance...
[00:02:35.160,247] <inf> CellModem: MODEM_POWERED_DOWN->MODEM_POWERING_UP
uart:~$ AT
U_CELL_PWR: powering on.
AT
AT
AT
AT
AT
AT

OK
U_CELL_PWR: powering on, module is alive.
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

05.09,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=0

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"10000101","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 90

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
[00:02:41.734,741] <inf> CellModem: Modem power up and ready to be configured
[00:02:41.734,802] <inf> CellModem: MODEM_POWERING_UP->MODEM_POWERED_UP
[00:02:41.734,832] <inf> CellModem: uartHandle 1
uart:~$ modem_ctrl  power_down
U_CELL_PWR: powering off with AT command.
Cell Modem Power Down
uart:~$ AT+CPWROFF
[00:02:48.192,077] <inf> CellModem: MODEM_POWERED_UP->MODEM_POWERING_DOWN
[00:02:48.192,108] <inf> CellModem: Closing an open device instance...

uart:~$
OK
AT
[00:02:53.894,439] <inf> CellModem: MODEM_POWERING_DOWN->MODEM_POWERED_DOWN
uart:~$

RobMeades, 2024-09-07:
That's excellent news @jamal-mouline, well done!  Of course, you will be driving the 1V8 UART pins of the module with 3V3, which I guess works but probably isn't what we'd recommend.  Since this is just a test jig I guess that if it works, it works, so all is good.

Internally we will make a note to look at the pulling in the latest release of nRFConnect and see what problems that brings with it.  It's a bit strange because we are already using the native Zephyr release 3.7, which I think is later, or at least as late as, the Zephyr version underneath nRFConnect 2.6, but of course we will be using that with other MCUs (ST, NXP), so if something has been changed in a Nordic driver we wouldn't notice.

Will update this issue when we know more.

jamal-mouline, 2024-09-09:
Thanks for the additional feedback @RobMeades 

I will be looking further into the 3.3 to 1.8, but will need to be at the lab with a scope to monitor the UART lines
I still observe some intermittent power up failure depending on the sequence of powering up both DKs

Regarding nRFConnect / Zephyr version, my debug console is reporting this
nRFConnect 2.6.1 is using Zephyr Kernel 3.5.99
nRFConnect  2.5.0 is using Zephyr Kernel 3.4.99
as in 
   uint32_t ver = sys_kernel_version_get();
   LOG_INF("%s %d.%d.%d", SYS_KERNEL_VER, SYS_KERNEL_VER_MAJOR(ver), SYS_KERNEL_VER_MINOR(ver), SYS_KERNEL_VER_PATCHLEVEL(ver));

Further update:
When all good with modem power up,
I am now able to register to AT&T using a "hologram" SIM card
Log from Shell console attached below...
Takes about ~18 second the first time, then just couple second for subsequent registrations...

uart:~$ modem_ctrl
modem_ctrl - Modem Control
Subcommands:
  q_test        :Basic Q Test
  power_up      :Cell Modem Power Up
  network_up    :Cell Modem Network Register
  mqtt_test     :Cell Modem Mqtt Test
  location      :Cell Modem Gnss Location
  network_down  :Cell Modem Network de-Register
  power_down    :Cell Modem Power Down
uart:~$ modem_ctrl   network_up
Cell Modem Register
uart:~$ AT+UMNOPROF?

+UMNOPROF: 90

OK
U_CELL_CFG: MNO profile is 90.
AT+UMNOPROF=100

OK
U_CELL_CFG: MNO profile set to 100.
### MNO profile has been changed from 90 to 100.
### Re-booting the module to apply MNO profile change...
U_CELL_PWR: rebooting.
AT+CFUN=16

OK
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

05.09,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=0

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"00010011","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
AT+URAT?

+URAT: 7

OK
U_CELL_CFG: RATs are:
  rank[0]: 10 (in module terms 7).
  rank[1]: 0 (in module terms -1).
  rank[2]: 0 (in module terms -1).
### RAT[0] is 10 - CAT-M1.
### Unable to read RAT uCellCfgGetRat() return -5
U_CELL_CFG: getting band mask for RAT 10 (in module terms 0).
AT+UBANDMASK?

+UBANDMASK: 0,524421,0,1,524420,0

OK
U_CELL_CFG: band mask for RAT 10 (in module terms 7) is 0x0000000000000000 0000000000080085.
### Band mask for RAT CAT-M1 is 0x0000000000000000 0000000000080085.
### Setting band mask for RAT CAT-M1 to 0x0000000000000000 000000000000181a...
U_CELL_CFG: setting band mask for RAT 10 (in module terms 0) to 0x0000000000000000 000000000000181a.
AT+UBANDMASK=0,6170,0

OK
### Re-booting the module to apply RF band changes...
U_CELL_PWR: rebooting.
[00:04:40.187,469] <inf> CellModem: size of table 3 1
uart:~$ AT+CFUN=16

OK
AT
AT

OK
ATE0
ATE0

OK
AT+CMEE=2

OK
AT+UDCONF=1,0

OK
ATI9

05.09,A00.01

OK
AT&C1

OK
AT&D0

OK
AT&K0

OK
AT+UPSV=0

OK
AT+UPSMR=1

OK
AT+CPSMS?

+CPSMS: 0,,,"00010011","00000011"

OK
AT+UMNOPROF?

+UMNOPROF: 100

OK
AT+UGPRF?

+UGPRF: 2,0,""

OK
AT+CFUN=4

OK
U_CELL_NET: preparing to register/connect...
AT+CREG=2

OK
AT+CGREG=2

OK
AT+CEREG=4

OK
AT+CIMI

234500007895849

OK
U_CELL_NET: user-specified APN is "hologram".
AT+CGDCONT=1,"IP","hologram"

OK
AT+UAUTHREQ=1,0,"",""

OK
U_CELL_NET: setting automatic network selection mode...
AT+COPS?

+COPS: 0

OK
[00:04:59.575,866] <inf> CellModem: Bringing up the network interface...
uart:~$ AT+CFUN=1

OK

...

AT+CGREG?

+CGREG: 2,4

OK
-1: OoC
AT+CEREG?

+CEREG: 4,0

OK
-1: NReg
AT+CREG?

+CREG: 2,0

OK
-1: NReg

+CREG: 7,"2709","0bbce811",7

+CEREG: 5,"2709","0bbce811",7
7: RegS
7: RegR
AT+COPS=3,0

OK
AT+COPS?

+COPS: 0,0,"AT&T",7

OK
AT+CGATT?

+CGATT: 1

OK
AT+CGACT?

+CGACT: 1,1

OK
AT+CEER

+CEER: "EMM cause",0,"No cause information available",fff,fff

OK
U_CELL_NET: connected after 18 second(s).
AT+CGACT?

+CGACT: 1,1

OK
AT+CGPADDR=1

+CGPADDR: 1,"100.69.113.127"

OK
U_CELL_NET: IP address "100.69.113.127".
AT+CGCONTRDP=1

+CGCONTRDP: 1,5,"hologram.mnc050.mcc234.gprs","100.69.113.127.255.0.0.0","100.0.0.1","8.8.8.8","8.8.4.4","","",0,0,1500,0,0

OK
U_CELL_NET: primary DNS address:   "8.8.8.8".
U_CELL_NET: secondary DNS address: "8.8.4.4".
[00:05:18.976,745] <inf> CellModem: MODEM_POWERED_UP->MODEM_NETWORK_REGISTERED
uart:~$

jamal-mouline, 2024-09-09:
All good as well with MQTT Publish to Mosquitto

Mosquitto Console:

AzureAD+JamalMouline@DESKTOP-6N00H8P /cygdrive/c/Toolchains/mosquitto-2.0.8
$ mosquitto_sub  -h    test.mosquitto.org -t    350428920015806  -v
350428920015806 HelloWorld
350428920015806 HelloMqttWorld_SARA-R520M10

Nordic Console:

uart:~$ modem_ctrl   mqtt_test    HelloMqttWorld_SARA-R520M10
Mqtt test HelloMqttWorld_SARA-R520M10
uart:~$ AT+UMQTT=2,"test.mosquitto.org"

OK
AT+CGSN

350428920015806

OK
AT+UMQTT=0,"350428920015806"

OK
U_CELL_MQTT: trying to connect...
AT+UMQTTC=1

OK
U_CELL_MQTT: waiting for response for up to 120 second(s)...

+UUMQTTC: 1,1
U_CELL_MQTT: connected after 1 second(s).
AT+UMQTTC=4,1,"messages/350428920015806/devicebound/"

OK

+UUMQTTC: 4,1,1,"messages/350428920015806/devicebound/"
AT+UMQTTC=9,1,0,"350428920015806",27

>HelloMqttWorld_SARA-R520M10
OK

+UUMQTTC: 9,1
AT

OK
U_CELL_MQTT: trying to disconnect...
AT+UMQTTC=0

OK

+UUMQTTC: 0,1
U_CELL_MQTT: waiting for response for up to 120 second(s)...
[00:08:28.826,324] <inf> CellModem: Disconnect MQTT session
uart:~$ U_CELL_MQTT: disconnected after 1 second(s).
[00:08:29.887,847] <inf> CellModem: Close MQTT Client session
uart:~$

RobMeades, 2024-09-15:
Hi Jamal: so that I have the right context here, you have an EVK with a SARA-R520M10 mounted on it and you are trying to use the M10 GNSS device that is inside the cellular module?

> First question: is this the correct/expected setup?

Yes, that is fine.

> During the location request, a RED LED is turned on the ADP DS160, Is that normal / expected?

I've no idea I'm afraid.  Which red LED is this?

> Both U_CFG_APP_CELL_PIN_GNSS_POWER and U_CFG_APP_CELL_PIN_GNSS_DATA_READY are defaulted to -1

Yes, that is correct.  Those are only required if the GNSS device is external to the cellular module, which is not the case here.  You might need to have the cellular GNSS detach switch [SW304] in the opposite position to the “GNSS” position, to make sure that the other GNSS device on the EVK is not the one that is in play.

From your log, the cellular device is definitely talking to a GNSS chip, you can see the UBX messages being exchanged with it; the GNSS chip is unable to get a location fix but it is able to establish the time so it is seeing satellites, just not enough of them.  I guess you are using the active antenna that likely came with the EVK and it is in a location where it can see open sky?

jamal-mouline, 2024-09-16:
Hi @RobMeades 
What I meant by my first question was related to 
AT+UGPS=1,15,295
OK

Thanks for confirming about 
U_CFG_APP_CELL_PIN_GNSS_POWER and U_CFG_APP_CELL_PIN_GNSS_DATA_READY are defaulted to -1

RE:  I guess you are using the active antenna that likely came with the EVK and it is in a location where it can see open sky?
That is correct

To address your question about the context, yes I am trying to use the M10 GNSS that is inside the cellular module, which itself is populated on the ADP board, which is plugged on the EVK on ports J100 and J101

Will attach a picture shortly, showing also the RED LED...

RobMeades, 2024-09-16:
> AT+UGPS=1,15,295
> OK

Yes, this is fine.

It is worth making sure that the DIP switch labelled "GNSS" is in the "detach" position, just in case that otherwise makes the module use the external GNSS chip on the EVK rather than its internal one.

jamal-mouline, 2024-09-16:
Open Sky View
[image]

jamal-mouline, 2024-09-16:
Will attach a picture shortly, showing also the RED LED...
[image]

Also Verified the GNSS red switch, the fourth from the man power switch is in the DETACH position --also shown on the above pic

RobMeades, 2024-09-16:
I enquired internally about DS160 (apologies, hadn't spotted that you'd mentioned the label DS160 before, I'd searched the EVK document for DS160 and it didn't appear so I assumed it was a mis-type).  Apparently it was mounted by mistake on some ADP boards, hence it not being documented.  It is connected to the `TXD_GNSS` line of the module, so it would make sense that it lights up since the GNSS module will, by default, be sending NMEA messages over that UART line.

How long are you waiting for a fix?  Checking the log of `UBX`, messages, right at the very end, it finds a 2D-only fix, seeing only 3 satellites though.  I've no idea why it should take this long but of course it is starting from absolutely nothing, no knowledge of time at the outset, and no stored ephemeris data.  You might just have to wait longer.

```
                        The "2" here indicates a 2D-only fix | 
              This indicates the number of satellites used in the fix |
                                                             V        V
20 dd 7f 14 e8 07 09 0b 17 1f 2e 37 2e 03 00 00 d2 b3 04 00 00 00 26 00 d0 57 f7 d0 95 93 43 15 89 22 01 00 e5 a7 01 00 2d 12 08 00 28 6c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 9e 1d 5f 01 5e 3d 00 00 12 55 43 00 0f 27 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00
08 e1 7f 14 e8 07 09 0b 17 1f 2f 37 da 01 00 00 b8 ba 04 00 02 00 26 03 ea 57 f9 d0 ac 3f 44 15 5c 26 01 00 c4 ab 01 00 ef ce 03 00 e2 d8 00 00 05 c6 00 00 e5 fc ff ff 00 00 00 00 0c c6 00 00 0c f2 23 02 c2 3d 00 00 cb 37 11 00 b9 01 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00
f0 e4 7f 14 e8 07 09 0b 17 1f 30 37 a2 01 00 00 c1 b9 04 00 02 00 26 03 9b 59 f9 d0 50 52 44 15 1a 27 01 00 82 ac 01 00 3d 62 03 00 8d 6a 01 00 93 cc 00 00 75 02 00 00 d3 fc ff ff 97 cc 00 00 00 0d 01 00 f9 36 00 00 25 d2 0e 00 b9 01 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00
d8 e8 7f 14 e8 07 09 0b 17 1f 31 37 21 01 00 00 cd b8 04 00 02 00 26 03 bf 59 f9 d0 83 64 44 15 86 28 01 00 ee ad 01 00 51 cb 02 00 7c 3e 01 00 07 c7 00 00 e5 fc ff ff e2 ff ff ff 0e c7 00 00 cc f3 23 02 b2 25 00 00 34 e1 0a 00 b9 01 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00
d0 13 80 14 e8 07 09 0b 17 20 00 37 af 00 00 00 b6 b0 04 00 02 00 26 03 81 8a f9 d0 7d bd 45 15 3e f7 00 00 a7 7c 01 00 7e b6 01 00 b3 45 01 00 b1 e0 00 00 4e f4 ff ff 6e 02 00 00 ff e0 00 00 ec c4 20 02 10 0f 00 00 ce 48 07 00 30 01 00 00 5c 40 56 2f 00 00 00 00 00 00 00 00
```

jamal-mouline, 2024-09-16:
Thanks for the additional information on DS160, that makes sense now, so no red flag there, thank you

**Re: How long are you waiting for a fix?**

I don't seem to have control on that parameter, or did I miss it?
Please see a snapshot of my code copy/paste below
is there a way to configure the GNSS chip / wait time before calling uLocationGet()?!?

Tracing down the call to uLocation() to ... to posGet() within ubxlib,
it seems to use a fixed timeout value of 4 minute U_GNSS_POS_TIMEOUT_SECONDS == 240
Right?

Unless you want to hard code that value to twice that much and retry?
Thank you

   int32_t returnCode = U_ERROR_COMMON_SUCCESS;
   // Bring up the GNSS network layer on the cellular device.
   LOG_INF("Bringing up GNSS...");
   returnCode = uNetworkInterfaceUp(pObj->deviceHandle, U_NETWORK_TYPE_GNSS, &gNetworkCfg);
   if (returnCode == U_ERROR_COMMON_SUCCESS)
   {
      LOG_INF("Get a location fix");
      returnCode = uLocationGet(pObj->deviceHandle, U_LOCATION_TYPE_GNSS, NULL, NULL, pLocation, NULL);

RobMeades, 2024-09-16:
You can take control of the timeout by passing a `pKeepGoingCalback` parameter into [uLocationGet()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/location/api/u_location.h#L316), i.e, rather than the last parameter being `NULL`, have it point to a function of the following form:

```
static bool keepGoingCallback(uDeviceHandle_t devHandle)
{
    return true;
}
```

While `keepGoingCallback()` returns `true` [uLocationGet()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/location/api/u_location.h#L316) should wait, as soon as `keepGoingCallback()` returns `false` it will exit.

jamal-mouline, 2024-09-16:
I just tried doubling up the value of U_GNSS_POS_TIMEOUT_SECONDS in the CMake file,
But no impact, still timing out with no fix!

Will try the callback suggestion next...

# Compile definitions needed for UBXLIB.
target_compile_definitions(app PRIVATE
   U_CFG_APP_CELL_UART=1
   U_CFG_CELL_MODULE_TYPE=9 # uCellModuleType_t
   U_CFG_APP_PIN_CELL_PWR_ON=38
   U_CFG_CELL_DISABLE_UART_POWER_SAVING=0
   U_CFG_APP_PIN_CELL_RESET=39
   **U_GNSS_POS_TIMEOUT_SECONDS=480**
   )

RobMeades, 2024-09-16:
While I will help as much as I can, I'm really the wrong person to be talking to about this; there's nothing of significance that `ubxlib` configures in the GNSS chip in order to obtain a fix, the GNSS chip just does its thing.  Only being able to see 3 satellites with an active antenna plugged into the GNSS antenna port of the ADP is just weird: likely you need someone who knows something about the HW at this point...?

jamal-mouline, 2024-09-16:
The callback approach did not make much difference either!

Agreed that missing some HW setup is more likely than a ubxlib issue.
ubxlib worked well when we tried it last in the OpenCpu environement
From what I remember a 4th satellite is needed to get a fix.
Will keep you posted if/when I find something else...
Thank you for all the good suggestions anyway.

PS
In between, I would like to keep this ticket open to get your feedback when regressing ubxlib against a later version of nRFConnect SDK (a later version than 2.5.0), so I may upgrade to the same version as well
Thank you

RobMeades, 2024-09-16:
> From what I remember a 4th satellite is needed to get a fix.

Correct, but you should see _loads_ of satellites: we usually see between 20 and 40 here (with a roof-top antenna); I would expect at least double figures.

> In between, I would like to keep this ticket open to get your feedback when regressing ubxlib against a later version of  RFConnect SDK (a later version than 2.5.0), so I may upgrade to the same version as well

Of course.

jamal-mouline, 2024-09-25:
Hi @RobMeades 

Good news, I am now able to get a fix through ubxlib pretty quickly.
Long story short, I only had to extend my antenna ~5-6 outside my window for better open sky view.
With that, I see about 7-8 satellites. Nothing close to 20 but much better than before.

On a related subject, does ubxlib offer an interface to configure the at+UGPS parameters?
By default it is always sending AT+UGPS=1,15,295

Thanks,

PS
On the subject of nRF Connect / Zephyr association, here is what I got back from Nordic
nRF 2.5.0 > Zephyr 3.4.99 <<< this is the currently working version
nRF 2.6.1 > Zephyr 3.5.99 <<< this version had problems as documented above
nRF 2.7.0 > Zephyr 3.6.99 <<< this is available, planning to try it next...
nRF 2.8.0 > Zephyr 3.7.99 (on the roadmap for early November 24)

RobMeades, 2024-09-25:
Good stuff.

> On a related subject, does ubxlib offer an interface to configure the at+UGPS parameters?
> By default it is always sending AT+UGPS=1,15,295

By default we always set all of the supported system types at the outset.  If you are obtaining location through the Cell Locate API (which will use any attached GNSS chip) you can set the system types used by CellLocate with [uCellLocSetSystem()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/cell/api/u_cell_loc.h#L266).

If you are obtaining position from the GNSS chip, rather than via CellLocate (which I think is your case), then, for an M9 or M10 device, you just need to use the right configuration ID to set or unset the system types using the [uGnssCfg API](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_cfg.h), i.e. one of [U_GNSS_CFG_VAL_KEY_ID_SIGNAL_XXX](https://github.com/u-blox/ubxlib/blob/master/gnss/api/u_gnss_cfg_val_key.h#L3071C9-L3071C37), most simply following the kind of pattern in [this example](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/example/gnss/cfg_val_main.c#L284), so something like:

```
if (U_GNSS_CFG_SET_VAL_RAM(devHandle, U_GNSS_CFG_VAL_KEY_ID_SIGNAL_BDS_ENA_L, 0) == 0) {
    printf("Beidou has been disabled.\n");
} else {
    printf("Unable to disable Beidou!\n");
}
```

jamal-mouline, 2024-09-26:
Hi @RobMeades 
Thanks for the feedback
I initially tried uCellLocSetSystem(), which did not work, so understand the reason now

Following your suggestion, I have inserted few of these type of calls in an attempt to disable some system types
   U_GNSS_CFG_SET_VAL_RAM(pObj->deviceHandle, SIGNAL_GAL_ENA_L, 0);
   U_GNSS_CFG_SET_VAL_RAM(pObj->deviceHandle, SIGNAL_BDS_ENA_L, 0);
   U_GNSS_CFG_SET_VAL_RAM(pObj->deviceHandle, SIGNAL_QZSS_L1S_ENA_L, 0);

just before bringing up GNSS (call to uNetworkInterfaceUp())
but I still end up with the default setup
any idea whether I am missing another step?
thanks,

AT+UGPS=1,15,295
OK

RobMeades, 2024-09-27:
> but I still end up with the default setup

Yes, you will still get `AT+UGPS=1,15,295`, that default will remain unchanged, you just need to put the `U_GNSS_CFG_SET_VAL_RAM...` lines _after_ `uNetworkInterfaceUp()` and that will remove the satellite systems you don't need.

> Also how to force the aiding type parameter?
> I tried to redefine U_CELL_LOC_GNSS_AIDING_TYPES from default to 1, but still ended up with 15

Remember that the aiding types is to do with the way Cell Locate works, and you are not using Cell Locate as far as I am aware, so you can ignore it.  If I've misunderstood and you _are_ using CellLocate then you will see from the code [here](https://github.com/u-blox/ubxlib/blob/master/cell/api/u_cell_loc.h#L139):

```
#ifndef U_CELL_LOC_GNSS_AIDING_TYPES
/** The aiding types to request when switching-on a GNSS
 * chip attached to a cellular module (all of them).
 */
# define U_CELL_LOC_GNSS_AIDING_TYPES 15
#endif
```

...that if you define `U_CELL_LOC_GNSS_AIDING_TYPES` when you build the `ubxlib` code to be, say, 1, then that definition will override the one in `u_cell_loc.h`.  How are you #defining the value of `U_CELL_LOC_GNSS_AIDING_TYPES` in your build system?

jamal-mouline, 2024-09-27:
Awesome!
I do see now the impact in the log of inserting those suggested Macros between uNetworkInterfaceUp and uLocationGet
In bold in the log snapshot below
Also seems to reduce the number of used satellites

AT+UGPS?
+UGPS: 0
OK
AT+UGPS=1,15,295
OK
**U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 22 00 31 10 00 fe 97.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 0d 00 31 10 00 e9 2e.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 0f 00 31 10 00 eb 38.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].
U_GNSS: sent command b5 62 06 8a 09 00 01 01 00 00 14 00 31 10 00 f0 51.
U_GNSS: decoded UBX response 0x05 0x01: 06 8a [body 2 byte(s)].**

[00:03:06.599,792] <inf> CellModem: Get a location fix

jamal-mouline, 2024-09-27:
Now that I understand better how to configure the system types, I decided to stay with the default setting as is
--but good to know in case I need to revisit in the future for some sort of optimization

Regarding your earlier question
Re: How are you #defining the value of U_CELL_LOC_GNSS_AIDING_TYPES in your build system?

I was re-defining it in the CMakeList.txt file
But now that I understand CellLocate is not used in the SARA-520M10, I also left it to its default mask value at 15, as you suggested

jamal-mouline, 2024-09-27:
Next week, I plan to re-evaluate the modem power up sequence using the latest available nRF 2.7.0 > Zephyr 3.6.99
and keep you posted here...

jamal-mouline, 2024-10-01:
Update regarding evaluation of nRF 2.7.0 / Zephyr 3.6.99

I got this link error from u_port_os.c

C:/GitAV/AutoVentive/external/modules/lib/ubxlib/port/platform/zephyr/src/u_port_os.c:172:30: warning: implicit declaration of function 'Z_KERNEL_STACK_LEN'; did you mean 'K_KERNEL_STACK_LEN'? [-Wimplicit-function-declaration]
  172 |             stackAllocSize = Z_KERNEL_STACK_LEN(stackSizeBytes);
      |                              ^~~~~~~~~~~~~~~~~~
      |                              K_KERNEL_STACK_LEN

the associated code is copy/paste below

#if KERNEL_VERSION_NUMBER >= ZEPHYR_VERSION(3,7,0)
            stackAllocSize = K_KERNEL_STACK_LEN(stackSizeBytes);
#else
            stackAllocSize = Z_KERNEL_STACK_LEN(stackSizeBytes);
#endif

@RobMeades
is there a known fix/patch to resolve this?
thank you

RobMeades, 2024-10-01:
That's a bit strange.  `Z_KERNEL_STACK_LEN` became `K_KERNEL_STACK_LEN` at Zephyr release tag 3.7.0, you can see that here:

https://github.com/zephyrproject-rtos/zephyr/commit/f05111dea048814ef312eb664aaa87e9eda3844c

So, if you have Zephyr version 3.6.99, `Z_KERNEL_STACK_LEN` must surely be the thing?

Can you look through the nRFConnect SDK code you have ended up with in version 2.7.0 and verify (a) what `zephyr/‎include/zephyr/kernel/thread_stack.h` has as the definition for `THING_KERNEL_STACK_LEN` and (b) what `zephyr/version.h` contains?

jamal-mouline, 2024-10-02:
@RobMeades 
to your first questions

zephyr/version.h has been modified by Nordic and contains the following
VERSION_MAJOR = 3
VERSION_MINOR = 6
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =

zephyr/‎include/zephyr/kernel/thread_stack.h has a K definition K_KERNEL_STACK_LEN as you had mentioned and not a Z
#define K_KERNEL_STACK_LEN(size) \
	ROUND_UP(Z_KERNEL_STACK_SIZE_ADJUST(size), Z_KERNEL_STACK_OBJ_ALIGN)

jamal-mouline, 2024-10-02:
@RobMeades 

Taking your hint account, I made the following patch into u_port_os.c
and it sure builds now
What would be the proper way to move forward with this?
Thank you

#if KERNEL_VERSION_NUMBER >= ZEPHYR_VERSION(3,6,99)
            stackAllocSize = K_KERNEL_STACK_LEN(stackSizeBytes);
#else
            stackAllocSize = Z_KERNEL_STACK_LEN(stackSizeBytes);
#endif

RobMeades, 2024-10-02:
Good stuff; I will change the >=  3.7.0 Zephyr version check to be >= 3.6.99 instead.

jamal-mouline, 2024-10-02:
Now that the build completes error free with nRF Connect 2.7.0

I was hopeful the previous error related to 
_uPortGpioSet() returning error code -5 ... Nordic's GPIO driver is not initializing the port_pin_mask ... etc
would be resolved

Unfortunately, the error is still there, therefore I am forced to go back to safety 2.5.0

Going forward,
what would be the plan to resolve the issue with Nordic in order to be able to use their latest available nRF Connect SDK

Is there a roadmap for ubxlib to test/validate with the later nRF Connect SDKs?
Thank you

RobMeades, 2024-10-02:
Understood, I thought that might be the case.  I doubt it will be a complicated problem to resolve but I must leave the answer to @Daniel-Herlestam-ublox, @m-abubakar and @sakrubx to resolve as I retire from u-blox tomorrow.  You can, of course, always find me on the internet, but they are now "the guys" (and they do very much know what they are doing) :-).

I will assign this issue over to @m-abubakar to avoid confusion.

Good to talk to you @jamal-mouline!

jamal-mouline, 2024-10-02:
Hello @RobMeades 
It was a pleasure working with you over the ~ past year or so
Best wishes for your future endeavors
Jamal

RobMeades, 2024-10-03:
@jamal-mouline: you can find the change to Zephyr version conditional compilation in commit 556a048469cd52b9f91f6f916de69111c57f0a15.

jamal-mouline, 2024-10-03:
General question I had in the back of my mind --non urgent

Does ubxlib maintain some sort of versioning numbers that the application could query?
Would be useful when reporting issues... I could say I am using ubxlib version MM.mm.pp
Thank you

jamal-mouline, 2024-10-07:
@m-abubakar Hi,
Kindly checking on updates for this ticket

Plus this additional question regarding the UART power saving feature
There is #define associated for this feature, but it seems it has a reverse logic. I could be wrong though!
What is your recommendation to disable UART power saving
U_CFG_CELL_DISABLE_UART_POWER_SAVING=1
or
U_CFG_CELL_DISABLE_UART_POWER_SAVING=0

From my experiments, it seems to work better with the latter >> AT+UPSV=0

Please advise...

jamal-mouline, 2024-10-08:
Hello @Daniel-Herlestam-ublox and @sakrubx

Just checking if this ticket is still being supported...

Thank you

m-abubakar, 2024-10-08:
Hi @jamal-mouline,
  Sorry for the delay. We are working on your queries and will respond shortly. Thank you for understanding.

m-abubakar, 2024-10-09:
Hi @jamal-mouline,
Thank you so much for your patience. Yes, it involves a bit of reverse logic. Setting `U_CFG_CELL_DISABLE_UART_POWER_SAVING=1` effectively means you're opting not to use the power-saving feature—in other words, you are disabling UART power saving (i.e., `AT+UPSV=0`).

To enable power saving, it is recommended not to define `U_CFG_CELL_DISABLE_UART_POWER_SAVING`. Additionally, [E-DRX](https://github.com/u-blox/ubxlib/blob/4561c455b5ab9e701ef14b13e8ad96482c65ba7a/cell/api/u_cell_pwr.h#L765) and [3GPP](https://github.com/u-blox/ubxlib/blob/4561c455b5ab9e701ef14b13e8ad96482c65ba7a/cell/api/u_cell_pwr.h#L572) power-saving features should work as expected if you don't define this configuration.

Thank you.

jamal-mouline, 2024-10-09:
Hi @m-abubakar 
Appreciate the feedback
It does make sense that defining
**U_CFG_CELL_DISABLE_UART_POWER_SAVING=1**
disables power saving >> AT+UPSV=0

What looked counter-intuitive was that
**U_CFG_CELL_DISABLE_UART_POWER_SAVING=0**
had the same exact effect.

My conclusion is that all it matters is whether the feature U_CFG_CELL_DISABLE_UART_POWER_SAVING is defined or not, regardless of its assigned value 0 or 1

This issue is better understood now.
So we could move on to the previous questions:

1-Versioning of ubxlib commit being used?
2-Roadmap for compatible ubxlib version with latest nRF Connect 2.7.0 / Zephyr 3.6.99

Thank you

m-abubakar, 2024-10-10:
Hi @jamal-mouline,

Regarding the version information, if I understood your query correctly, you're asking if there's an API call that returns this information. Unfortunately, at this time, ubxlib does not provide an API to query the version of the library in use.

I am currently working on the second matter and will keep you updated as soon as I make any progress.

Thank you.

jamal-mouline, 2024-10-10:
Hello @m-abubakar 
Thank you for the update

jamal-mouline, 2024-10-16:
Hi @m-abubakar 
Checking whether you have any high level information about
https://github.com/u-blox/ubxlib/issues/285
and whether it is assigned to somebody in the ubxlib team?
Thank you

closed 2024-11-11 by manemannen