u-blox/ubxlib #279: IoT Hub's MQTT messages lost

dmandai, 2024-09-19:
Hello,

I am using this library in a project that needs the LARA-R6001 to be connected to the Microsoft IoT Hub (through MQTT).

When the LARA-R6001 excecutes the connection using the function:
`int32_t` uMqttClientConnect(uMqttClientContext_t *pContext, const uMqttClientConnection_t `*pConnection);`

But doesn't execute the Topic subscription.

The message Queue, if not empty, of the IoT Hub is cleared and no messages are available.
From the IoT Hub stand point the device has received all the messages.

We have already verified that this doesn't happen with other devices.
In fact, after the connection, the other device shows the messages received.

Any suggestion?

Thank you,
Davide.

m-abubakar, 2024-09-20:
Hi @dmandai,

   Sorry to hear that you are facing issues using the library. could you please share how you're using the uMqttClientConnect and uMqttClientSubscribe functions in your code?

You might find it helpful to refer to the [MQTT example](https://github.com/u-blox/ubxlib/blob/master/example/mqtt_client/mqtt_main.c) for comparison with your implementation.

Thank you.

dmandai, 2024-09-20:
Hi @m-abubakar,

Here the code used to execute the connection:
```
// Set MQTT connection options.
inst->connection->pBrokerNameStr = inst->url;
inst->connection->pClientIdStr = inst->client_id; // OPTIONAL
inst->connection->pUserNameStr = inst->user_name;
inst->connection->pPasswordStr = "";
inst->connection->keepAlive = true;
inst->connection->retain = false;
inst->connection->inactivityTimeoutSeconds = AZ_IOT_DEFAULT_MQTT_CONNECT_KEEPALIVE_SECONDS;
inst->connection->mqttSn = false; // DEFAULT
// Connect MQTT client to the Azure IoT Hub.
return uMqttClientConnect(inst->pContext, inst->connection);
```

Where the TLS SETTINGS are the following:
```
inst->tlsSettings->tlsVersionMin = U_SECURITY_TLS_VERSION_1_2;
inst->tlsSettings->pRootCaCertificateName = rootCertificateName;
inst->tlsSettings->pClientCertificateName = certificateName;
inst->tlsSettings->pClientPrivateKeyName = privateKeyName;
inst->tlsSettings->certificateCheck = U_SECURITY_TLS_CERTIFICATE_CHECK_ROOT_CA_URL;
inst->tlsSettings->pClientPrivateKeyPassword = NULL;
inst->tlsSettings->pExpectedServerUrl = iniCfg.mqttUrl;
```

Here the code used to execute the subscription:
```
ret = uMqttClientSubscribe(inst->pContext, topic, U_MQTT_QOS_AT_LEAST_ONCE); //U_MQTT_QOS_EXACTLY_ONCE);
```
Where:
```
Topic =  "devices/+/messages/devicebound/#"
```

Thanks,
Davide.

closed 2024-11-11 by manemannen