u-blox/ubxlib #281: support for RUST language

jozeblackblox, 2024-09-23:
Hello!

I hope this is still in scope...

I want to use ubxlib in must std RUST project - I started from plain [template ](https://github.com/esp-rs/esp-idf-template).

How to build and use ubxlib?

AgafonovVadim, 2024-09-25:
So, 
**### clone and Build ubxlib**
You'll need to clone the ubxlib C library:

`git clone https://github.com/u-blox/ubxlib.git`

Now, build the library using your system's C/C++ build toolchain.

```
cd ubxlib
make   # Adjust this depending on the build system ubxlib uses
```

_Make sure the library builds successfully and locate the generated library files (`.a`, `.so`, or `.dll` depending on your platform)._

 **### Use bindgen to Create Rust Bindings**
 
Open your Cargo.toml and add 

```
[dependencies]
bindgen = "0.63.0"

[build-dependencies]
cc = "1.0"   # For compiling C code if needed
```

And it's done)

Example
```
use std::env;
use std::path::PathBuf;
use bindgen;

fn main() {
    // Specify the path to the ubxlib library.
    let ubxlib_dir = "../ubxlib"; // Adjust this to your actual path

    // Tell Cargo to tell rustc to link the system ubxlib
    println!("cargo:rustc-link-search=native={}", ubxlib_dir);
    println!("cargo:rustc-link-lib=static=ubxlib");

    // Generate bindings to ubxlib
    let bindings = bindgen::Builder::default()
        .header(format!("{}/include/ubxlib.h", ubxlib_dir))  // Path to the ubxlib header
        .generate()
        .expect("Unable to generate bindings");

    // Write the bindings to the $OUT_DIR/bindings.rs file.
    let out_path = PathBuf::from(env::var("OUT_DIR").unwrap());
    bindings
        .write_to_file(out_path.join("bindings.rs"))
        .expect("Couldn't write bindings!");
}
```

jozeblackblox, 2024-09-26:
Hello!

Thanks you for answer!

I still have problems creating  .a file to include it in my Rust project.

Since I use ESP32, I tried with [CMakeLists from ESP32 Runner](https://github.com/u-blox/ubxlib/blob/master/port/platform/esp-idf/mcu/esp32/runner/CMakeLists.txt), but without success. I tried with Windows and WSL2.

Does UBXLIB rely on esp-idf or can I make standalone build?

Can you specify build process with few more details?

Thanks!

AgafonovVadim, 2024-09-26:
Good afternoon, @jozeblackblox!

Let's start from the beginning.

> Does UBXLIB rely on esp-idf or can I make standalone build?
- **Yes**, `ubxlib `typically relies on `esp-idf` when targeting ESP32 devices. This is because the ESP32 is usually programmed using the ESP-IDF (Espressif IoT Development Framework), which provides the necessary SDK, drivers, and build environment for ESP32.

> Can you specify build process with few more details?

- To build ubxlib for ESP32 and generate the .a static library, you need to set up esp-idf and configure the build process accordingly.

Ok, let me try to explain it step by step.

### Step 1:

1. **Install ESP-IDF:** Follow the official guide to install ESP-IDF on your machine. For Windows, it's recommended to use WSL2 for better compatibility. (I will tell you about the example of WSL 2, so you can use Ubuntu)

**I. Install ESP-IDF**
``` bash

sudo apt update
sudo apt install git wget flex bison gperf python3 python3-pip cmake ninja-build ccache libffi-dev libssl-dev dfu-util
```

**II. Clone ESP-IDF repo**
``` bash
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
./install.sh
```

**III. Set environment variables**
``` bash
. ./export.sh
```

2. **Set Up Toolchain: Install the Xtensa**
``` bash
./install.sh
```

### Step 2: Configure `ubxlib`

1. **Clone ubxlib (like you did)**

``` bash
git clone https://github.com/u-blox/ubxlib.git
cd ubxlib
```

2. **Configure for ESP32:**

**Navigate** to the `port/platform/esp-idf `directory:
``` bash
cd port/platform/esp-idf
```
You'll find a CMakeLists.txt file here that is configured to work with ESP-IDF. This CMake file ensures that ubxlib is built using ESP-IDF's environment.

### Step 3: Build the Static Library

1. **Configure the Project:** 

Inside the` ubxlib/port/platform/esp-idf` directory, run the CMake configuration to generate the build files. You can specify where the static library (.a) should be output.

```
idf.py set-target esp32
idf.py menuconfig
```
2. **Build**
```
idf.py build
```

This will generate the necessary **.a** file under the `build` directory. Look for the static library, likely named something like libubxlib.a.

I'm happy to help! Please try the instructions I sent, and feel free to reach out anytime. I'll be glad to hear if everything works out!

Good luck! 🍀🍀🍀

AgafonovVadim, 2024-10-10:
Hi there!!! 👋

I'm glad to hear that I could help you🤗. I wish you the best of luck with your projects! 

> I hope you will someday add support for the RUST language or maybe create a working example with an external C library 🙂 🙏

We’ll have to wait a bit, I’m still studying:)

I’d like to ask you a couple of questions about your experience. If you don’t mind, let me know how I can reach you, or you can message me wherever it’s convenient for you—all my links are in my profile🙃. Thanks!

Have a good day! 🍀☀️😎

closed 2024-11-11 by manemannen