u-blox/ubxlib #282: Antenna Tuning

jamal-mouline, 2024-09-30:
Hi @hkro @ZillKhan @steve.brown

As discussed on the CD / uBlox call from last Wed Sep 9/25
Opening this ticket as a placeholder to provide a method to configure the "Antenna Tuning" feature via ubxlib
Either a new API using the device handle obtained from uDeviceOpen() API
or an alternative way to directly setup the required AT command

Thank you

hkro, 2024-10-01:
Hi @jamal-mouline, while we could provide a generic `ubxlib` API for this, any factory/production oriented operation, which is likely time-sensitive and product/module specific, you would be better off with an AT command crafted specifically to your needs using the `ubxlib` [AT Client](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/at_client/api/u_at_client.h#L44) directly. Since you are using `ubxlib`, just obtain the AT handle with [uCellAtClientHandleGet()](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/cell/api/u_cell.h#L252) and write only the code you need for your specific production case.

jamal-mouline, 2024-10-01:
Hi @hkro 
Thank you for the feedback.
We can certainly make direct use of the AT Client to configure the antenna tuner

To that end, I wanted to double check the correct sequence of commands to accomplish that
Below is a copy paste from the AT Manual for the SARA-R5 / LEXI-R5 series
On page 285, section 15.5.5, I found this example
Could you please confirm these are the proper steps?

Enabling antenna tuner control
AT+CFUN=0 OK Disable the protocol stack.
AT+UTEST=1 OK Enter the test mode.
AT+UTEST=4,1 OK The module enables the antenna dynamic tuner feature.
AT+UTEST=0 OK Enter the normal mode.
AT+CFUN=1 OK Restore full module functionality.

Also, according to the documentation for +UTEST=4, section 15.5
"
The antenna dynamic tuner control setting is stored in the NVM, and its configuration is effective after exiting the test mode.
"
I wanted to confirm that this has to be done only one time --likely at the very first modem power up, since the setting is stored in non volatile memory.

Thank you

jamal-mouline, 2024-10-02:
Hello @ZillKhan 

Assuming the "antenna tuner control" enabling is performed through the 5 steps documented in my earlier note --pending your confirmation, I have couple additional questions:

1-Please confirm that AT+CFUN=0 and AT+CFUN=1 have already support APIs in place
I was able to find:
void uCellPrivateCFunMode(uCellPrivateInstance_t *pInstance, int32_t mode);
int32_t uCellPrivateCFunOne(uCellPrivateInstance_t *pInstance);
to support them respectively

However they both use an "instance pointer" instead of an AT handler

2-Would it make sense to make use of
uCellPrivateInstance_t *pUCellPrivateGetInstance(uDeviceHandle_t cellHandle);
to retrieve the "instance pointer" from the cell handler obtained from uDeviceOpen()

Thank you

ZillKhan, 2024-10-03:
> Hi @hkro Thank you for the feedback. We can certainly make direct use of the AT Client to configure the antenna tuner
>
> To that end, I wanted to double check the correct sequence of commands to [...]

Hello @jamal-mouline 
Yes, these are the correct steps for enabling Antenna Tuner control as mentioned in the AT manual.
The Antenna Tunning setting is stored in NVM, You only need to set the Antenna Tuning once, likely at the first power up of the modem.

ZillKhan, 2024-10-03:
> Hello @ZillKhan
>
> Assuming the "antenna tuner control" enabling is performed through the 5 steps documented in my earlier note --pending your confirmation, I have couple additional questions: [...]

@jamal-mouline 
1. Yes, ubxlib already support +CFUN=0,1 API and the mentioned APIs are correct.
2. It make sense to use `pUCellPrivateGetInstance` to get instance pointer. You will also require AT client handler for this purpose for that you can use this API `uCellAtClientHandleGet`.

For reference we have a AT handler use case in `fota_https` application here: https://github.com/u-blox/cell_ucpu_app/blob/main/apps/fota_https/fota_https.c#L197

jamal-mouline, 2024-10-03:
@ZillKhan 
OK, thank you for the confirmations and additional data points
Will give it a try and let you know if any questions...

jamal-mouline, 2024-10-04:
@ZillKhan 
I put together a tentative API to support this feature --see attached
Would appreciate any review comments
Thank you

[AntennaTuner.c.txt](https://github.com/user-attachments/files/17259181/AntennaTuner.c.txt)

ZillKhan, 2024-10-07:
> @ZillKhan I put together a tentative API to support this feature --see attached Would appreciate any review comments Thank you
> 
> [AntennaTuner.c.txt](https://github.com/user-attachments/files/17259181/AntennaTuner.c.txt)

Hello @jamal-mouline,
I had discussion with ubxlib team and uCPU team, we cannot use pUCellPrivateGetInstance() directly as this API is private to ubxlib scope only. That means we can't use `uCellPrivateCFunMode()` `uCellPrivateCFunOne` as both of these APIs uses `uCellPrivateInstance_t ` which has scope within ubxlib only.

Considering this in mind, I have updated `CellModemAntennaTunner()` that includes +CFUN=0 and +CFUN=1 AT commands directly using AT handler.

I have tested this chunk in uCPU framework and its working as expected. Attaching the updated code. 
[AntennaTunerUpdated.c.txt](https://github.com/user-attachments/files/17277478/AntennaTunerUpdated.c.txt)

Let me know if you require any further support is this regards.

Thanks.

jamal-mouline, 2024-10-07:
@ZillKhan 
Your comments make sense to me as well --not making use of the "private" APIs.
I will work with the updated framework, and test it against the hardware target when it becomes available
No further questions at this time
Thank you

jamal-mouline, 2024-10-24:
Hi @ZillKhan 

Additional question please:
The SARA-R5 / LEXI-R5 series - AT commands manual
section 15.5.2
page 283
Antenna dynamic tuner write configuration table

The example takes **12** mask arguments (beyond the =4,2)
AT+UTEST=4,2,0,64,0,134232064,1048576,0,34472080,0,0,16777231,2,0
OK

Whereas the syntax description lists **13** masks
AT+UTEST=4,2,<atun_0_lte_bitmask0>,<atun_0_lte_bitmask1>,<atun_0_gsm_umts_bitmask0>,<atun_1_lte_bitmask0>,<atun_1_lte_bitmask1>,<atun_1_gsm_umts_bitmask0>,<atun_2_lte_bitmask0>,<atun_2_lte_bitmask1>,<atun_2_gsm_umts_bitmask0>,<atun_1_gsm_umts_bitmask0>,<atun_2_lte_bitmask0>,<atun_2_lte_bitmask1>,<atun_2_gsm_umts_bitmask0>

Could you please help us clarify which format is correct?

Playing with a SARA-R520M10 devkit, it seems that the syntax description is INCORRECT, listing one extra mask
the question is which one?

at+utest=4,3
+UTEST: 1,0,64,0,134232064,1048576,0,34472080,0,0,16777231,2,0

OK
at+utest=4,2,0,64,0,134232064,1048576,0,34472080,0,0,16777231,2,0
OK

jamal-mouline, 2024-10-24:
@ZillKhan 
I run programmatically the procedure for enabling antenna tuner control that we discussed earlier
Somehow AT+UTEST=4,1 returns an ERROR
Any idea for the underlying reason?
Hints on getting more details for the ERROR root cause?

uart:~$ modem_ctrl antenna_tuner

AT+CFUN=0

<inf> CellModem: Antenna Tuner

OK
AT+UTEST=1

OK
AT+UTEST=4,1

ERROR
**<err> CellModem: The module does NOT enable the antenna dynamic tuner feature**
AT+UTEST=0

OK
AT+CFUN=1

OK

PS
Here is the associated piece of code supporting AT+UTEST=4,1

      uAtClientLock(atHandle);
      uAtClientCommandStart(atHandle, "AT+UTEST=");
      uAtClientWriteInt(atHandle, 4);
      uAtClientWriteInt(atHandle, 1);
      uAtClientCommandStop(atHandle);

      // Grab the response.
      uAtClientCommandStopReadResponse(atHandle);
      errorCode = uAtClientUnlock(atHandle);

      if (errorCode != (int32_t)U_ERROR_COMMON_SUCCESS)
      {
         LOG_ERR("The module does NOT enable the antenna dynamic tuner feature");
      }

jamal-mouline, 2024-10-28:
Hi @ZillKhan

Quick follow on the last comment above:
Setting AT+UTEST=4,1 programmatically through ubxlib is returning an ERROR systematically
This is a blocking issue for us!

Please provide a partial API just for doing that specific portion
Please attach a log showing success

PS
The other 4 steps are working fine (1. AT+CFUN=0 / 2. AT+UTEST=1 / 4. AT+UTEST=0 / 5. AT+CFUN=1)

ZillKhan, 2024-10-29:
> Hi @ZillKhan
>
> Quick follow on the last comment above: Setting AT+UTEST=4,1 programmatically through ubxlib is returning an ERROR systematically **This is a blocking issue for us!**
>
> Please provide a [...]

@jamal-mouline now @rebatoma will further resolve/support you for your queries.

jamal-mouline, 2024-10-29:
@rebatoma Hi Mattia,
Looking forward to your input/support on this one
Best regards,
Jamal