u-blox/ubxlib #284: FOAT Support for Cellular Modules

michaelboeding, 2024-10-07:
It would be nice to have the ability to perform an FOAT update from the host MCU to the cellular module built into ubxlib. Or the ability to use the AT client directly to make my own implementation of this. Unless im missing this and it's already a built in feature?

m-abubakar, 2024-10-08:
Hi @michaelboeding,

Thank you for getting in touch! We do already support monitoring-only functionality in ubxlib. However, adding FOAT update support is indeed a valuable enhancement, and we’ll definitely discuss it internally.

In the meantime, you can use the AT client directly. Please refer to the [ATClient](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/common/at_client/api/u_at_client.h#L44) documentation for more details. You can also obtain the AT Client Handle using [uCellAtClientHandleGet](https://github.com/u-blox/ubxlib/blob/2896142106fe596fb289748c68ba4afd8c1e717b/cell/api/u_cell.h#L252).

Let me know if using the AT client directly along with the current monitoring support works for you.

Thank you.

michaelboeding, 2024-10-11:
Hey @m-abubakar I have started implementing this in my code and am using the AT client you listed above - thanks! So far I have gotten to where the module is ready to receive the data over UART so I start transferring the data OTA and then to the module over UART as received using the Xmodem protocol. But I never seem to get a packet to send. See below on some of my logs. Note the weird bytes that the module seems to be emitting...does that mean anything? I also noticed in the documentation it says that " Once the +UFWUPD AT command is ready to transfer binary data, it will not accept any AT
commands for approximately 109 s" should I not be using the AT client to send the data over UART or is this fine even though it says this? 

E (187915) SARAR4: Timeout or error while waiting for ACK for packet 1
E (187915) SARAR4: Failed to send packet 1
E (187915) ExternalModuleUpdater: Firmware update send to SARAR422S over UART via AT Client Failed
I (187925) SARAR4: Sending firmware data using AT client...
I (187925) SARAR4: Sending Xmodem packet 1 with 128 bytes
AT
AT

OK
[01][01][fe][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][08][1f][fe]?[13][00][00][00]T[04][fe]?T[04][fe]?[00][00][00][00][80][1b][fe]?F[01][00][00][d4][99][fc]?[04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[14][9f][b8][c2][04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[14][9f][b8][c2][04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[1E (187915) SARAR4: Timeout or error while waiting for ACK for packet 1
E (187915) SARAR4: Failed to send packet 1
E (187915) ExternalModuleUpdater: Firmware update send to SARAR422S over UART via AT Client Failed
I (187925) SARAR4: Sending firmware data using AT client...
I (187925) SARAR4: Sending Xmodem packet 1 with 128 bytes
AT
AT

OK
[01][01][fe][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][08][1f][fe]?[13][00][00][00]T[04][fe]?T[04][fe]?[00][00][00][00][80][1b][fe]?F[01][00][00][d4][99][fc]?[04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[14][9f][b8][c2][04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[14][9f][b8][c2][04][b1][fe][90][c2]![[00]90@?j[bc][1b]B[1

michaelboeding, 2024-10-17:
Any update on this? Am I doing something wrong based on that output?

closed 2024-11-11 by manemannen